cncjs/cncjs #164: Zero Out Work Offsets dont work with TinyG

tomauga, 2017-06-09:
Using Windows 7 with CNCJS Desktop version  1.9.5 connected to TinyG, the XYZ->Zero Out Work Offsets does not appear to do anything. I would expect after clicking this that the displayed coordinates would zero and clicking Goto Workspace Zero would go to the set coordinate. Goto Workspace always goes to Machine Zero.

SirGeekALot, 2017-06-09:
I noticed this problem, too.  Here is my workaround to this bug:

https://plus.google.com/u/0/communities/103429207810600659899

cheton, 2017-06-10:
Not sure why you had this issue. Here is my test result running g2core (fb: 100.19, fv: 0.99) on an Arduino Due.

1. Jog to X10 Y10
    [image]

2. Zero Out Work Offsets (G10 L20 P1 X0 Y0 Z0)
    [image]

3. After that, the work position will be set to X0 Y0 Z0
    [image]

dirtdiver9, 2017-06-10:
These steps work perfectly for me, but then after the job is finished the bit returns to the start point but the work position is never X0, Y0 ( I understand Z being elevated ).  If X and Y would end up being and showing 0,0 then I could very quickly get started on the next tool path after probing Z.  

Right now I have to home, probe, and start again for the second tool path which adds a lot of time.

Thanks,
Patrick

cheton, 2017-06-10:
@dirtdiver9 
Do you mean that `G0 X0 Y0` will not move to the expected start point after a job is finished?
Can you show a minified gcode file that can describe your situation. A recorded video will be good for  troubleshooting.

tomauga, 2017-06-10:
@cheton I'm not sure how that is working for you since you are setting 'offsets' of 0,0,0. My understanding is that you should be sending:

   G10 L20 Pn X{xOffset} Y{yOffset} Z{zOffset)

Where:
  n = active workspace coordinate system
  x|y|zOffset are the axes offsets from the machine zero.

Setting to X0 Y0 Z0 clears Pn offset right? I'm new to G-Code so am probably wront.

Here is screenshot after going to machine zero, moving X 3mm clicking then clicking Zero Out Work Offsets:
[image]

When I issue gcode in console: G10 L20 P1 X3 Y0 Z0   the offset is moved in X 3mm and then the UI Work Position updated automatically and is set to zero as shown below:
[image]

Hope this helps, (first time attempting to upload images).

-Tom

neilferreri, 2017-06-10:
@tomauga "G10 L20 P1 X0Y0Z0" will set the current position as 0,0,0 in the G54 coordinate system. It's not an offset at all, and I think that may be the source of confusion. If you want to set a true coordinate offset, you'd use G10 L2, bit using it the way @cheton said is very simple. Jog to where you want the work origin to be, set it to 0,0 and it will persist until you want to change it.

tomauga, 2017-06-11:
@neilferreri  True for Grbl but not TinyG. See TinyG documentation:

> G10 Set Parameters (Offsets)
> 
> Currently the only G10 command supported is G10 L2, which is used to set coordinate offsets. Use Pn to select coordinate system 1-6 (G54 - G59, respectively), and one or more axis value to set the offset for that axis.

https://github.com/synthetos/TinyG/wiki/Gcode-Support#g10-set-parameters-offsets

You will find on TinyG that "G10 L20" == "G10 L2"

cheton, 2017-06-11:
It seems that g2core has added support for `G10 L20`, but not available in TinyG. 

TinyG
https://github.com/synthetos/TinyG/wiki/Gcode-Support#g10-set-parameters-offsets

g2core
https://github.com/synthetos/g2/wiki/Gcode-Support#g10-l2-set-parameters-offsets

This should explain why I can use "G2 L20" to modify the value with g2core, but it didn't work as expected with TinyG.
[image]

neilferreri, 2017-06-11:
@tomauga Forgot you were using TinyG and I should've read the posts more clearly.  I'm with you. Your understanding of the the "G10 L2" is correct, and I'm not sure how @cheton had that work exactly like it does on a grbl controlled machine. Is TinyG under active development? Looks like the L20 functionality was added as a feature request a few years ago.

cheton, 2017-06-11:
@neilferreri @tomauga 

The wiki explains the naming differences of TinyG and g2core:
https://github.com/synthetos/g2/wiki/What-is-g2core#some-clarification-on-naming

I also found this issue that was posted several years ago:
- <b>Possibility of adding G10 L20 command</b> https://github.com/synthetos/TinyG/issues/100

Hi @aldenhart
Is there any updates for `G10 L20` support in TinyG? Thank you.

cheton, 2017-06-18:
According to the readme of TinyG firmware version 0.97, the latest edge branch (firmware build 449.01) has added `G10 L20` support. See below:
https://github.com/synthetos/TinyG/tree/edge-0.97#summary-of-changes

You can go to the download page and make a try:

**Download page**
http://synthetos.github.io/

**TinyG Edge firmware v0.97 / build 449.01**
http://synthetos.github.io/binaries/tinyg-edge-449.01.hex

cheton, 2017-06-18:
https://github.com/synthetos/TinyG/tree/edge-0.97#branches

### BRANCHES

* MASTER is the current production code. This is not updated very frequently.
* EDGE branches are work-in-process for the next Master releases. Depending on the velocity of changes they will be somewhere between alpha and beta. An attempt is made to keep everything working and usable, but there are no guarantees for edge branches. Generally speaking, edge branches should not be used for production uses - use the master branch instead.
* Edge-0.97 is designed to be (mostly) backwards compatible with Master 440.xx. Changes are new features and additions that do not break current processing. However, there are a few changes that affect communication and other behaviors. These are listed below as Changes Affecting Migration from Master 440.xx to Edge-0.97.
* Edge-0.98 ports-in functionality available in the g2 code base. There are many changes that affect migration from Master 440.xx or edge-0.98. Please refer the the edge-0.98 readme or wiki for details.

tomauga, 2017-06-19:
@cheton You are correct, this is available in the Edge branch which hasn't been updated in a very long time. Not sure I want to experiment with it at this point.

cheton, 2017-12-14:
Closes this issue. For people want to use  `G10 L20` command, please upgrade TinyG firmware to edge-0.97 branch (firmware build 449.01), or use g2 code base.

closed 2017-12-14 by cheton