cncjs/cncjs #166: How to energize motors?

tomauga, 2017-06-09:
I need to be able to run multiple files on the same work piece at the same coordinates and change tools between. (Using F-Engrave)

I haven't found a way to energize/lock motors so this can be done without accidentally changing an axis. Is there UI for this?

I haven't tried tool changes in a gcode file yet. Is this supported? If so are motors locked? How to resume etc.

dirtdiver9, 2017-06-09:
I do this all the time where I have one path for a large tool, and a second for the vbit.
I load the first file, Home, position the bit, and hit play.  When it is done, I home, switch the tool, position the bit, and play the second file.

When the first file is done the bit returns to what looks like work X0, Y0 and with Z elevated, but the display doesn’t show X0, Y0 so I haven’t trusted it which I why I re-home, and then position the bit with my 3 Axis block.

—Patrick

tomauga, 2017-06-09:
Thanks Patrick. How are you setting the home position? And which controller are you using?

In ChiliPeppr there is a button to energize the motors so you dont accidentally change the x/y.

-Tom

dirtdiver9, 2017-06-09:
I’m using the grbl controller and I have a known size aluminum block.  http://triquetra-cnc.com/product/triquetra-p3-auto-zero-touch-plate-with-g-code-generator-copy/ <http://triquetra-cnc.com/product/triquetra-p3-auto-zero-touch-plate-with-g-code-generator-copy/>
Can’t say enough about Charlie and his stuff.
It came with a macro that lets you probe all 3 axis and then I have another macro that returns to X0,Y0,Z0 so I can verify the bit before I hit play.

When my machine is sitting there the motors have enough holding strength to let me change the bit.

tomauga, 2017-06-09:
Which grbl controller are you using?

dirtdiver9, 2017-06-09:
The older X-carve one.  Daughter card to the Arduino.

neilferreri, 2017-06-09:
I leave my steppers enabled all the time when connected using grbl. That said, if I change tools, I think it's good practice to home the machine afterwards. Are all of your cuts starting at 0,0 or do you have them already arranged in your workspace?

tomauga, 2017-06-09:
My cuts all start at 0,0. I was just hoping Cheton could add a button that would energize the steppers. I can always use the $md/$me commands for TinyG. Just not very friendly.

Another problem (which I reported) is not all the software brings the spindle back to 0,0 when the file is finished and cncjs hasnt been working for me to set workspace 0,0.

cheton, 2017-06-10:
Sure! I will add <kbd>Motor Enable ($me)</kbd> and <kbd>Motor Disable ($md)</kbd> buttons in the next release.

rgallus, 2017-06-13:
Thanks cheton, I was going to make this same request.

cheton, 2017-06-18:
@tomauga @rgallus 

May I confirm if you're able to use `{me:0}` and `{md:0}` commands from your side as well as sending `$me` and `$md` commands? I'd like to send commands in JSON format, just want to make sure if can work for TinyG firmware v0.97.

rgallus, 2017-06-18:
Thank you for your response.  I had forgotten about the $md and $me
commands.  I reviewed the TinyG information and found them.  I very much
like cncjs and want to thank you for its development. Looking forward to
the next update.

Rick

tomauga, 2017-06-19:
@cheton Yes, I have access to both json and $ versions of the command. Please note that they are only active for the timeout period. ChiliPeppr gets around this by continually sending me until you press the un-energize button at which point he sends md.

cheton, 2017-06-20:
@tomauga 

Thank you for reminding me. I will follow the same way for energizing motors:
https://github.com/chilipeppr/widget-tinyg/blob/master/widget.js#L674

rgallus, 2017-06-20:
I only have the ability to use $.  For some reason I can't energize Jason.

cheton, 2017-06-20:
@rgallus 
Could you type `$sys` command to show your settings? You can try setting `{ej:1}`to enforce JSON mode.

rgallus, 2017-06-20:
Both systems now work.  Did not know about using the bracket when writing
from the console.

Thank you

Rick

cheton, 2017-06-21:
Added in the commit a48de34fb178e17e4f9747974862c2de1ab23459. You can see a preview below. I will publish a new release in the next few days if no issues found.

[image]

tomauga, 2017-06-21:
Great job, thanks Cheton!

cheton, 2017-07-01:
Added in 1.9.6

closed 2017-07-01 by cheton