cncjs/cncjs #169: 1.9.5 The server suddenly crashes with keypad jogging

cheton, 2017-06-10:
The Node.js server of CNCjs 1.9.5 may suddenly crash with an unexpected abort error caused by Socket.IO 2.0, especially when doing keypad jogging. Rolling back Socket.IO to 1.7.x will resolve this issue. I will provide a bug fix in the next few days.

In this moment, please downgrade CNCjs to 1.9.4 if you encountered this issue.

Related issue: https://github.com/cncjs/cncjs/issues/161

AustinSaintAubin, 2017-06-10:
NOTE: Recommended Downgrade Process for Raspberry Pi, just to be helpful to anyone that might need it.

```
# Stop current CNCjs version
pm2 stop cncjs

# Setting: Make Backup
mv ~/.cncrc ~/.cncrc_bak

# Remove current CNCjs version
sudo npm uninstall -g cncjs

# Install and run older version of CNCjs
sudo npm install -g cncjs@v1.8.15 --unsafe-perm

# Settings: Restore Backup
#rm ~/.cncrc
mv ~/.cncrc_bak ~/.cncrc

# Start CNCjs
pm2 start cncjs
```

cheton, 2017-07-01:
Fixed in 1.9.6

closed 2017-07-01 by cheton

fragmuffin, 2017-07-01:
Thanks @AustinSaintAubin, I've used that method a few times now while debugging an issue on my Raspberry Pi... very useful.