cncjs/cncjs #170: Arduino is soft reset once a job is completed

erpalma, 2017-06-13:
I'm not sure if this is a bug or just my issue: when a job is finised the arduino is soft resetted, losing work coordinates.

cheton, 2017-06-13:
Once a job has finished, a soft reset (^x) will be sent to Grbl make sure all modal groups `[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
` will restore to defaults. For example, if you have run a gcode program measured in inches (G20), the units mode will be changed to mm (G21) after a soft reset. It will also clear TLO as well.

```js
 'gcode:stop': () => {
    this.event.trigger('gcode:stop');

    this.workflow.stop();

    const activeState = _.get(this.state, 'status.activeState', '');
    const delay = 500; // 500ms
    if (activeState === GRBL_ACTIVE_STATE_RUN) {
        this.write(socket, '!'); // hold
    }

    setTimeout(() => {
        this.write(socket, '\x18'); // ctrl-x
    }, delay);
}
```

A good practice is to probe each time before you start a job. If you want to save the work coordinate offsets, use `G10 L20` command as shown below in the Axes widget.

[image]

cheton, 2017-06-13:
Just to let you know I will add some controller specific settings to the UI in a future version, this will allow users to decide whether to send a soft reset after a job has finished.

erpalma, 2017-06-14:
Thanks for the answer! I understand the reason of the soft reset, but when doing pcb I have to perform multiple "passes" to the board (etch, drill, mill) and each of them must have the same work coordinates. I do probe at each pass since I change the tool, but I just probe the Z axis. I'll try to save work coord offsets as you suggested!

BTW this is something that I really hate wrt Mach3. I know that GRBL can't save coords in eeprom due to reduced number of erase cycles, but this is quite annoying. If something goes wrong and grbl reset (eg EMI) I usually can't resume the job and I have to throw away the pcb board I was working on. Of course this is not a cncjs issue ;)

Thanks again, this software is great!

neilferreri, 2017-06-14:
@erpalma You can store the X and Y values separately for exactly your situation.  Just use the serial communication to send G10L20P1X0Y0. Your XY work origin will be persistent a you can reprobe to set Z for each tool. If you repeat the same type of job (PCBs), I'd suggest you name a jig to make sort of jig to make placing the workpiece consistent.

dirtdiver9, 2017-06-14:
Hey Cheton,

You asked me a bit ago for examples of what happens after I finish a job and how the work coordinates do not end up at X0, Y0.  The bit returns to X/Y of where I believe is originally X0,Y0 but the coordinates do not show X0,Y0 on the screen.  I have some pictures, etc if you are interested.  After following the thread I believe it is the Reset that is done after the job that is causing the issue so I’ll wait for your code change to allow that to be turned on and off and see how that goes.

I don’t know if it is helpful for anyone else but I have a macro I run after probing that runs:

G90 G0 X0 Y0 Z0 
G92 X0 Y0 Z0
G10 L20 P1 X0 Y0 Z0

erpalma, 2017-06-14:
Hmmm, probably I didn't understand this morning but I always used G10L20P1X0Y0. The problem is that when the job is finished the machine is at (x1,y1) wrt to origin (0,0) in work coords, after reset the machine is still at (x1,y1) but work coords says (x2,y2). So there is no way to come back at the original (0,0). The only way I think is to add G0 X0 Y0 before the end of the gcode file.

neilferreri, 2017-06-15:
@dirtdiver9 I'm confused by your macro.  
G0 X0 Y0 Z0 --> Go to 0,0,0
G92 X0 Y0 Z0 --> Temporarily set current position (0,0,0) as 0,0,0
G10 L20 P1 X0 Y0 Z0 --> Save current position (already 0,0,0 and temporarily 0,0,0) to 0,0,0

I'm not sure what your probing routine is as there are many ways to do that.  I think sending a G92 before the G10L20 could be causing the issues with origin not remaining after a reset.  Using a temporary offset before the G10 can cause issues as the G92 is cleared when grbl is reset.

dirtdiver9, 2017-06-15:
Hey @neilferreri,
  The probe macros I have set 0,0,0 and then move away and stop.  I put my dust shoe on and then run the macro below and verify it is really at the 0,0,0 of the board.

I appreciate the tip about G92 and will move it around in the macro and see how it behaves then.

Thanks again,
Patrick

neilferreri, 2017-06-15:
@dirtdiver9
How does your probe set 0,0,0? I'd leave G92 out of it if you're using the G10. A way I use G92 is to change my Z when I change end mills.

cheton, 2017-08-14:
I just remembered why I sent a reset on hold, it's only useful when you want to stop a running G-code program immediately from the UI, and then returned to the Idle state. However, for a normal workflow, it's not necessary to send Grbl reset on program end. I will change the code like below:
```js
 'gcode:stop': () => {
     // Whether or not to force stop a G-code program
     const { force = false } = { ...args[0] };

     this.event.trigger('gcode:stop');

     this.workflow.stop();

     // "Hold and Reset" only when the user manually stop the workflow from the U
     if (force) {
         const activeState = _.get(this.state, 'status.activeState', '');
         if (activeState === GRBL_ACTIVE_STATE_RUN) {
             this.write(socket, '!'); // hold
             setTimeout(() => {
                 this.write(socket, '\x18'); // ctrl-x
             }, 500);
        }
     }
 }
```

cheton, 2017-08-25:
Changed the behavior in 1.9.8
https://github.com/cncjs/cncjs/releases/tag/v1.9.8

Now it won't send a soft reset on job completion, unless you manually stop a running G-code program from the UI.

closed 2017-08-25 by cheton