cncjs/cncjs #172: npm v5.0.3 installation integrity checksum failure

pluggee, 2017-06-15:
I'm trying to install cnc.js on CHIP Pro, ARM based debian. I'm getting an integrity checksum failure, 
I didn't see that when installing on my x64 ubuntu
I've attached the log file for more details
I've seen a couple of posts showing the same failure with npm v5 here

https://github.com/npm/npm/issues/17146
https://github.com/npm/npm/issues/16861

did anyone face that problem?

[2017-06-15T02_34_59_195Z-debug.log.txt](https://github.com/cncjs/cncjs/files/1076446/2017-06-15T02_34_59_195Z-debug.log.txt)

pluggee, 2017-06-15:
I got the same exact error when installing on a vanilla raspberry pi, running raspbian

[log_rpi.txt](https://github.com/cncjs/cncjs/files/1076843/log_rpi.txt)

cheton, 2017-06-15:
Try `npm cache verify` or `npm cache --force clean` to make sure everything is consistent.

nomykk, 2017-06-15:
Getting the same error with current npm version on Windows 10.

pluggee, 2017-06-15:
tried ~~~~npm cache --force clean~~~~ on both RPi and CHIP, no luck
I also verified that the cache was clean before trying to install

olange, 2017-06-15:
@pluggee @nomykk Have you tried `npm install --update-binary --no-shrinkwrap`? I had luck with that command, after doing `npm cache clean --force`. And I regenerated and versionned a fresh `npm-shrinkwrap.json` after these operations.

pluggee, 2017-06-15:
no luck, I tried `npm install --update-binary --no-shrinkwrap cncjs` on rpi

[rpi2.log.txt](https://github.com/cncjs/cncjs/files/1079104/rpi2.log.txt)

pluggee, 2017-06-17:
@olange @nomykk 
any idea what could be the offending package?

olange, 2017-06-21:
@pluggee I am not knowledgeable of the internals of NPM, it is as hard for me to point out the offending package, as for you probably.

Looking at your log file and searching for the SHA1 `sha1-UIwZDsSVggcyOw0+TvzkJ2t65qk=` that fails, I found these lines:

```
6198 silly pacote trying cncjs@https://registry.npmjs.org/cncjs/-/cncjs-1.9.5.tgz by hash: sha1-UIwZDsSVggcyOw0+TvzkJ2t65qk=
6199 silly extract webappengine@1.1.3
6200 silly pacote trying webappengine@https://registry.npmjs.org/webappengine/-/webappengine-1.1.3.tgz by hash: sha1-/OuwAQBhqKSDzgYdtVRPLRDxskI=
6201 silly pacote data for sha1-UIwZDsSVggcyOw0+TvzkJ2t65qk= not present. Using manifest.
6202 silly pacote data for sha1-/OuwAQBhqKSDzgYdtVRPLRDxskI= not present. Using manifest.
```

It is however unclear to me what to start with these lines.

cheton, 2017-06-22:
No sure what's going wrong. Try manually removing `~/.npm` and previously installed cncjs if you have, and then install again:
```
$ rm -rf ~/.npm
$ npm uninstall -g cncjs
$ npm cache verify
$ npm install -g cncjs@latest
```

nomykk, 2017-06-22:
@pluggee  @olange 
Deleting the cache manually from path "C:\Users\user-name\AppData\Roaming" (directories: npm , npm-cache) worked for me. After deleting directories npm install did the rest accurately.

nuintun, 2017-06-22:
same problem, no ```package-lock.json```
 
[image]

pluggee, 2017-06-22:
hey @cheton , I tried removing `~/.npm` and following your instructions, still the same problem
I have really no knowledge of npm (other that following instructions), did anyone try to install cncjs on a vanilla raspbian (released 2017-04-10)?
I followed instructions on a tiny intel box running ubuntu 16.04 and it went through, I only faced this problem with 2 systems running ARM cores, Raspberry Pi running Raspbian and C.H.I.P. Pro running debian

pluggee, 2017-06-28:
any other ideas??

anosikeosifo, 2017-07-03:
The following solved it for me: 

rm -rf ~/.npm
npm install --update-binary --no-shrinkwrap

*O.S: MacOS 10.12.5
 NPM 5.0.4

SumeethPrabhakar, 2017-07-19:
I deleted the npm cache (folders npm , npm-cache) manually from path "C:\Users\user-name\AppData\Roaming". worked for me.

taylorstine, 2017-07-19:
I fixed this by running:

```
gpg --keyserver pool.sks-keyservers.net --recv-keys 94AE36675C464D64BAFA68DD7434390BDBE9B9C5
gpg --keyserver pool.sks-keyservers.net --recv-keys FD3A5288F042B6850C66B31F09FE44734EB7990E
gpg --keyserver pool.sks-keyservers.net --recv-keys 71DCFD284A79C3B38668286BC97EC7A07EDE3FC1
gpg --keyserver pool.sks-keyservers.net --recv-keys DD8F2338BAE7501E3DD5AC78C273792F7D83545D
gpg --keyserver pool.sks-keyservers.net --recv-keys C4F0DFFF4E8C1A8236409D08E73BC641CC11F4C8
gpg --keyserver pool.sks-keyservers.net --recv-keys B9AE9905FFD7803F25714661B63B535A4C206CA9
gpg --keyserver pool.sks-keyservers.net --recv-keys 56730D5401028683275BD23C23EFEFE93C4CFFFE
curl -O https://nodejs.org/dist/v8.1.4/SHASUMS256.txt.asc
gpg --verify SHASUMS256.txt.asc
```

I'll be honest I'm not totally sure what it does (something with key signing obviously).  I got this from the README included with the [node binary downloads](https://nodejs.org/dist/v8.1.4/node-v8.1.4-linux-x64.tar.xz).

Ankit14190Pinal, 2017-07-23:
I got error when I installing angular "npm install -g @angular/cli"

I removed @angular and @angular-cli from npm-cache 'C:\Users\Engraved\AppData\Roaming\npm-cache'
now it's working for me.

StavroMueller, 2017-07-24:
Was getting this error, but in my case it was fixed by first running `npm cache clean`, even though the output scolded me a bit.

liliom55, 2017-08-03:
Still get error with npm version 5.3.0
`npm ERR! code EINTEGRITY
npm ERR! sha1-mnHEh0chjrylHlGmbaaCA4zct78= integrity checksum failed when using sha1: wanted sha1-mnHEh0chjrylHlGmbaaCA4zc
t78= but got sha1-BdAljdQ4f3upWYe2eNqgFyKwnWE=. (4849289 bytes)`

hezhiwebworld, 2017-08-08:
npm ERR! sha512-vOk6uEMctu0vQrvuSqFdJyqj1Q0S5VTDL79qtjo+DhRr+1mmaD+tluFSCZqhvi/J
UhXSzoZN2BhtstaPEeE8cw== integrity checksum failed when using sha512: wanted sha
512-vOk6uEMctu0vQrvuSqFdJyqj1Q0S5VTDL79qtjo+DhRr+1mmaD+tluFSCZqhvi/JUhXSzoZN2Bht
staPEeE8cw== but got sha512-pd+s9UiGqI1iFyuo9HnBrpGNNOE1o0VThHvIcRX0ECoMjQkINlVo
PG4MNzp4eyTVcvRieTaIEt35aB5PCU+NuA==. (2233 bytes)

npm  更新之后就是这么的坑爹

pluggee, 2017-08-09:
@taylorstine I tried adding and verifying the keys, but still no luck, I'm getting the same error

AustinSaintAubin, 2017-08-11:
what is your `node -v` and `npm -v` ?

pluggee, 2017-08-13:
@AustinSaintAubin 
npm -v = 5.0.3
there's an update available to 5.3.0
node -v = v4.8.3

AustinSaintAubin, 2017-08-13:
Try updating to node v6.

cheton, 2017-08-14:
@pluggee 

Try `npm i -g npm@latest` to update NPM to the latest version (5.3.0), and clean npm cache again before installation. I'm not sure if the checksum error can be resolved with the latest NPM, hope it works for you.

Enamex, 2017-08-17:
[**@cheton**](https://github.com/cheton) commented on [Aug 14, 2017, 4:36 AM GMT+2](https://github.com/cncjs/cncjs/issues/172#issuecomment-322087486 "2017-08-14T02:36:02Z - Replied by Github Reply Comments"):
> [@pluggee](https://github.com/pluggee)
> 
> Try `npm i -g npm@latest` to update NPM to the latest version (5.3.0), and clean npm cache again before installation. I'm not sure if the checksum error can be resolved with the latest NPM, hope it works for you.

Doing this was the first time I noticed this error, actually.

So I can't update npm. On 5.0.3 right now.

onuar, 2017-08-17:
@cheton npm 5.3.0 still has the same problem.

gezzeg, 2017-08-27:
also have same problem here on fresh installation.

Try "npm cache --force clean" but no luck.

Then i remove project folder and reinstall. Luckily some how it works!.

[image]

OrionStark, 2017-08-30:
BTW, I try to downgrade the npm version to npm@4.6.1 and it's doing well.

cheton, 2017-08-31:
NPM v5 is still unstable and existed some [known issues](https://github.com/npm/npm/issues/16991), for anyone who encountered installation problems with npm v5, try to downgrade the npm version to npm@4.6.1, clean npm cache, and continue the installation:

```sh
npm i -g npm@4.6.1
rm -rf ~/.npm
```

Each time you do the installation, you can run `npm cache verify` to verify the npm cache:
```sh
npm cache verify
npm install -g cncjs@latest # or 'sudo npm install --unsafe-perm -g cncjs@latest' for root account
```

Note: Do not install cncjs in a project folder containing a package.json file, Install cncjs globally with the `-g` option, or it may cause unexpected result.

aliir74, 2017-09-04:
My problem solved with @taylorstine solution.
node version: 8.4.0
npm version: 5.4.0

AbhilashDawar, 2017-09-04:
Updated to the latest versions as mentioned by @aliir74 i.e.
node version: 8.4.0
npm version: 5.4.0
Then ran command prompt as administrator. It worked.

DeltaCap019, 2017-09-05:
@cheton solution worked for me.. make sure you **Quit** the terminal in mac first, before implementing this solution

ParnianKamran, 2017-09-11:
I had the same issue and I used all the above solutions but they did not work for me. Please try Uninstalling node and npm and delete npm and npm-cache manually and again install node (Latest LTS Version: v6.11.3 (includes npm 3.10.10)) and npm and I hope that it works for you too.

onuar, 2017-09-11:
Since I've upgraded npm to '5.4.1' version, I don't encounter this error.

ParnianKamran, 2017-09-11:
@onuar I also upgraded npm to 5.4.1 version but it did not work, maybe my solution could be the last one if the above solutions did not work

cheton, 2017-09-15:
Seems to be resolved in NPM 5.4.0. Does anyone still have failure with NPM 5.4.0?

andy-ex, 2017-09-16:
Had the same problem, upgraded to NPM 5.4.0 and installation was successful. Thanks @cheton !

cheton, 2017-09-16:
Can someone help create a new section in the [Troubleshooting](https://github.com/cncjs/cncjs/wiki/Troubleshooting) page for how to resolve the installation problem? That will save a lot of trouble and time for people who also encountered similar errors with NPM installation. 

https://github.com/cncjs/cncjs/wiki/Troubleshooting

srsgores, 2017-09-25:
For me, I uninstalled nodejs and then reinstalled.  Problem solved, even with `npm@5.3.0`

pluggee, 2017-10-12:
@cheton , alright, seems to be fixed now if I follow vanilla instructions, when running `npm install npm@latest -g` it installs npm 5.4.2 and that solves the issue. I've successfully installed cncjs on a RPi 3 and RPi zero W
thanks for your support!

closed 2017-10-12 by pluggee

Arafullahi, 2017-10-30:
try `npm i -g npm `
and `npm cache verify`

tangjia0902, 2017-11-10:
@hezhiwebworld 
Seems I met same error as you when run "npm install":
npm ERR! sha512-N9OVsMBspboNvYaLAQnLEhb2eQ96lavogMR5LoH5k8nb1PvBZHSBFhzhsq2LNzGTBBOtBviOc1GiSu+wlM/pGw== integrity checksum failed when using sha512: wanted sha512-N9OVsMBspboNvYaLAQnLEhb2eQ96lavogMR5LoH5k8nb1PvBZHSBFhzhsq2LNzGTBBOtBviOc1GiSu+wlM/pGw== but got sha512-vXVO4mBnzw63SlWWii6j7K54a+HFrqc+fA4svOVWqVBTUaw8+rys3JClRchoD8EPCBAq0D9kJpEb+gRndZbhxQ==. (5518 bytes)

Did you fix the error?

cheton, 2017-11-10:
@tangjia0902 

What's your npm version?

tangjia0902, 2017-11-10:
@cheton 
npm version: 
npm: '5.3.0'

cheton, 2017-11-10:
Try the following:

```sh
npm i -g npm
rm -rf ~/.npm
npm cache verify  # optional
```

zibraque, 2017-11-14:
@tangjia0902  me too . did you fixed it yet?
@cheton  it's not working at all , npm --version 5.3.0 || windows10 wamp
```
################> npm i -g npm
npm ERR! code EINTEGRITY
npm ERR! sha512-M3aO8EjHebaCw6uur4C86SZqkypnoaEVo5R63FEEU0dw9wLxf/JlwWtJItShYVyQS2WDxG2It10GEe5GmVEM2Q== integrity checksum failed when using sha512: wanted sha512-M3aO8EjHebaCw6uur4C86SZqkypnoaEVo5R63FEEU0dw9wLxf/JlwWtJItShYVyQS2WDxG2It10GEe5GmVEM2Q== but got sha512-E0Xl+HGfFiPpkiHI05tsCo2vgRcqTsM2Gh3Ul1uHScGzR7PggUChr2i6Ql3z571LBKzs3eTm00fEoORJIETrLw==. (1703936 bytes)

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\admin\AppData\Roaming\npm-cache\_logs\2017-11-14T01_45_09_893Z-debug.log
```

cheton, 2017-11-15:
Have you tried removing the npm cache folder "C:\Users\admin\AppData\Roaming\npm-cache\" on Windows?

zibraque, 2017-11-15:
@cheton  I solved it by reloaded node.js . Tks anyway

tangjia0902, 2017-11-15:
@cheton @zibraque 
Finally, I solved it by config .npmrc to "registry = https://registry.npm.taobao.org"
Thanks all.

satishGangineni, 2017-11-17:
deleting the npm and npm cache folders from C:\Users\<user>\AppData\Roaming the path works for me!

BhuvaneswariVajravel, 2017-12-05:
Hi i got the same issue in windows
my versions are bellow
node -v
8.9.1
npm -v
5.4.0

i removed npm and npm-caches from AppData/Roaming

but when i try to >npm install -g @angular/cli@latest

i got the following EINTEGRITY error

> 
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.1.3 (node_modules\@an
> gular\cli\node_modules\fsevents):
> npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: Cannot cd into 'C:\Users\b
> huvaneswari_v\AppData\Roaming\npm\node_modules\.staging\fsevents-a025fdf0'
> 
> npm ERR! code EINTEGRITY
> npm ERR! sha1-PFtv1/beCRQmkCfwPAlGdY92c6Q= integrity checksum failed when using
> sha1: wanted sha1-PFtv1/beCRQmkCfwPAlGdY92c6Q= but got sha1-JsYlG6Gk8MTk+61T/R9R
> DoCk/QU=. (1024615 bytes)
> npm WARN tar ENOENT: no such file or directory, open 'C:\Users\bhuvaneswari_v\Ap
> pData\Roaming\npm\node_modules\.staging\rxjs-87a55af1\_esm2015\testing\TestSched
> uler.js.map'
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     C:\Users\bhuvaneswari_v\AppData\Roaming\npm-cache\_logs\2017-12-05T
> 07_52_25_588Z-debug.log
> 
> C:\>npm -v
> 5.4.0
> 
> C:\>node-v
> 'node-v' is not recognized as an internal or external command,
> operable program or batch file.
> 

please anyone help me

johnlin0207, 2017-12-25:
I updated my npm from 5.0.3 to 5.6.0,it seems worked!

AlexanderMac, 2017-12-29:
Completely removed node (I use nvm). And reinstalled it.

jonnycraze, 2018-01-20:
After going through much of what worked for others above I ended up removing my package-lock file, and that fixed it for me.

rodrigoengelberg, 2018-01-29:
The following solved it for me:
rm -rf node_modules
rm package-lock.json
rm -rf ~/.npm
npm install --update-binary --no-shrinkwrap

samfrach, 2018-02-24:
nothing worked for me on windows 10... but when I run "cmd" as Administrator... everything worked :)

bjmckenz, 2018-03-02:
> "npm cache verify" fixed the issue for me. Output says "Corrupted
> content removed: 23"

Unsubscribe

KEMBL, 2018-05-28:
"integrity checksum failure"
it happened with me when I changed my own npm server version from Sinopia to Verdaccio and copied packages to the new repo and then tried to use them. So be aware of such a case.

ptusch, 2018-06-20:
@KEMBL Did you actually fix this? Stuck at the same thing right now. :s

[7 more replies]