cncjs/cncjs #174: Running commands after boot in CNC.js on a raspberry pi fails unless user first logs on using remote desktop.

fk47, 2017-06-30:
Hi all,

I am currently trying to setup some commands in CNC.js, so that CNC.js can run and stop a python GUI script. If I decide to reboot with a command in CNC.js, the raspberry reboots with CNC.js working great, however when I try to run a python GUI script using a command from CNC.js, it will not work unless the user first logs on to the raspberry pi using remote desktop. Does anyone have a clue on how to bypass this? I would like to keep the security of having the remote connection password protected, but still getting access to file paths and running programs in CNC.js from boot.

For example, running:
`cd /pathtoscript
python script.py`

does not work unless the user first logs in to the desktop environment using XRDP in this case.

Thankful for any help regarding this issue.

fk47, 2017-06-30:
The error number I am receiving is errno=1 or errno=127 in the command drop down menu.

cheton, 2017-07-01:
The error number 1 stands for "Operation not permitted" or "Not super-user". What's the owner  permissions of you file? Please also attach your console output like below:

[image]

fk47, 2017-07-03:
Thank you for your answer. I managed to find error logs from pm2 in the .pm2 directory. Apparently the issue is on the python scripts side. Since it is a GUI it expects a display to output to and since I am running the raspberry headlessly and a "virtual display" has not yet been created with remote desktop it can not start. In other words, the problem is not really related to CNC.js so sorry for wasting your time... :man_shrugging: :

cheton, 2017-07-03:
Would you mind updating the wiki for your troubleshooting procedure (e.g. finding error logs from pm2)? This can be useful when troubleshooting commands running on a Raspberry Pi. 😃 

https://github.com/cncjs/cncjs/wiki/Troubleshooting

closed 2017-07-03 by cheton