cncjs/cncjs #175: Option to HALT if error is reported from machine interface

fragmuffin, 2017-07-02:
I've recently encountered an issue with my CAM software creating G2 (arc) commands that are outside the tolerance of Grbl.
This results in the error being reported by Grbl, and is not processed by the machine... which is fine.

The issue is that the next GCode block is sent to the machine and it continues.

In this event, could there be an option to stop gcode streaming and run a macro?

dirtdiver9, 2017-07-02:
Hi Peter,
  What CAM software were you using?  Was there something special you were doing with it?  Just curious so I don’t hit the same issue.

Having the machine halt would be nice for me as well.  I was getting errors and if it would have stopped I could have saved the work piece.

Thanks!

—Patrick

fragmuffin, 2017-07-03:
Hi @dirtdiver9 
I was using heekscad (heekscam) (more details in [this comment](https://github.com/gnea/grbl/issues/184#issuecomment-312433484) )

Because I'm paranoid about my machine breaking itself (because I don't have any money to replace it ;) ) I'm writing a gcode python library so I can write a post-processor to clean up  & validate any gcode I send to cncjs.
Currently in development (so it's currently empty), but if you're interested: https://github.com/fragmuffin/pygcode
I'm coding as much of it as I can today

fragmuffin, 2017-07-07:
Could a "gcode interpreter error" be added as a possible _Event_, then we could simply add an event to trigger `M0`

fragmuffin, 2017-08-13:
to quote `grbl` documentation:

> It's highly recommended to do what all professional CNC controllers do when they detect an error in the G-code program, halt. Don't do anything further until the user has modified the G-code and fixed the error in their program. Otherwise, bad things could happen.

source: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#g-code-error-handling

cheton, 2017-08-13:
It is doable to pause sending G-code when detecting an error, and user can then choose to continue or stop the job.

cheton, 2017-08-14:
I will add an option (radio button) to the <b>Settings > Controllers</b> page in the next release, it might look like below:

<b>G-code Error Handling</b>
- [x] Pause on error (default)
- [ ] Skip errors and continue to execute the program

cheton, 2017-10-06:
Added in the commit (d16264a4f53b7b2d9cf19441cea44176ddfbe134). Defaults to pause execution when the CNC milling controller detects an error in the G-code program.

fragmuffin, 2017-10-07:
nice one @cheton !

cheton, 2017-10-13:
https://github.com/cncjs/cncjs/releases/tag/v1.9.11

Now the behavior will be changed to pause execution on error

closed 2017-10-13 by cheton