cncjs/cncjs #176: Some lines of gcode not being skipped on first run after file load

JonathanGoldnauVogt, 2017-07-04:
Hello, I'm having an issue where the first run of a job messes up with various gcode not making it to the machine. I am using an arduino based GRBL board  at 115200 baud on a relatively powerful i5 laptop. This issue happened on multiple computers.  cncjs version 1.9.6

The missing gcode causes erratic behavior such as the bit not moving down to the workpiece on the initial plunge. Running the job again works normally.

Anyone have this happen?

cheton, 2017-07-10:
Sorry for the late reply. Did you see any errors reported from the Console widget? If you installed cncjs using NPM, you can run `cnc -vvv` to enable verbose output. If your cncjs were downloaded from the [releases page](https://github.com/cncjs/cncjs/releases), check out [Troubleshoot Electron app on Windows](https://github.com/cncjs/cncjs/wiki/Troubleshooting#troubleshoot-electron-app-on-windows).

If you have a recorded video that will be helpful to troubleshoot.

erpalma, 2017-08-04:
It happened also to me. I thought that it was some sort of issue (EMI?) in my setup, but maybe that's not the case.

JonathanGoldnauVogt, 2017-08-07:
@cheton @erpalma I am keeping an eye on this, when I encounter the issue again I will try to get the logs.

cheton, 2017-08-15:
This issue might be similar to https://github.com/cncjs/cncjs/issues/186#issuecomment-322367804

It will happen on 1.9.x when querying parser state ($G). If the query timer has timed out after 10 seconds with a response, a parser state query will send again to Grbl, and therefore it will consume all available receive buffer on Grbl after a certain period. This will be easily to reproduce when decreasing the feed rate for motion commands, like `G1X100Y100F50`.

https://github.com/cncjs/cncjs/blob/master/src/app/controllers/Grbl/GrblController.js#L465-L479

```js
const queryParserState = _.throttle(() => {
    const now = new Date().getTime();
    const lastQueryTime = this.actionTime.queryParserState;

    if (lastQueryTime > 0) {
        const timespan = Math.abs(now - lastQueryTime);
        const toleranceTime = 10000; // 10 seconds

        // Check if it has not been updated for a long time
        if (timespan >= toleranceTime) {
            log.debug(`Continue parser state query: timespan=${timespan}ms`);
            this.actionMask.queryParserState.state = false;
            this.actionMask.queryParserState.reply = false;
        }
    }

    // :  :  :
}
```

I should already know where is the root cause to this issue. I will let you know once the issue is resolved.

nsfilho, 2017-08-21:
Cheton, the GRBL BLOCK_BUFFER_SIZE is 15 or 16. Looking in your code, I see you are using 8 bytes to avoid overflow buffer and 5 secs to retry sending....

What did you think about change blocksize (15 or 16 -- depending of planner block size) and use a little more time (around 30 secs), because some commands (i.e. F35) use a lot of time to conclude.

Probably this isn't the root cause of the error, because it is important to check when receive buffer have slot to send, but help to reduce the errors. No?

cheton, 2017-08-22:
@nsfilho 

The deduction of 8 bytes is used to determine the available buffer size to use on the sender, it's not relevant to Grbl's BLOCK_BUFFER_SIZE. You can see my comment at https://github.com/cncjs/cncjs/issues/186#issuecomment-323911628

cheton, 2017-08-25:
Release Notes of CNCjs 1.9.8
https://github.com/cncjs/cncjs/releases/tag/v1.9.8

closed 2017-08-25 by cheton