cncjs/cncjs #177: Inch mode does not correspond to the machine? (smoothieware)

itman496, 2017-07-09:
Hello, I have only just started using CNCJS today, and I have to say, I love it.

I primarily work in inch mode because.. thats what I do.  The issue I have is when I switch to inch mode, the smoothieboard works fine, gcode works fine, but the view of the machine position (the digits next to the jog arrows) do not change properly.  When I move one inch, they only change to +0.0384 or something like that.  The 3D viewer in the center has the same problem, it moves a minuscule amount.  The machine however is moving properly, which is odd.

MM units setting seems to work fine, however.  Units change properly above the jog arrows and the 3D view seems to work properly.  I am no longer at my workshop so I can't take any video, but if need be I can tomorrow.

Setup is Smoothieboard on the latest CNC build, and the latest build of CNCjs running native on windows 10 x64.

Thanks for the great software!

cheton, 2017-07-10:
If you have a recorded video that will be great. You can check below questions before you can record a video in your workshop.

1. What's your firmware build / version?
2.  Did you use Smoothieware in `grbl_mode`? If it's not enabled, you can turn it on by setting `grbl_mode true` in your config.
    http://smoothieware.org/grbl-mode
3. There are 3 ways to change display units for the UI. What's the way you used to change the units mode?
    * Type 'G20' or 'G21' command on the Console widget
    * Change the units from the units dropdown of the 3D view 
    * When running a G-code file, any of G20 and G21 commands in your gcode will change current display units accordingly.

itman496, 2017-07-10:
Hello,

My firmware build is the latest CNC build from this page: https://github.com/Smoothieware/Smoothieware/tree/edge/FirmwareBin  I can confirm it uploaded properly.

I have put GRBL mode in the config even though its enabled by default for the cnc build, and it was there when this problem was occurring.

The way I changed the units was the mm/in button above the gcode preview window on the top rightish of the screen.  I can see the G20/G21 command go out on the console, and the machine responds properly.  The units would change to inches in the 3D preview window and on the machine position panel on the top right.  

But when I hit the 'jog 1" button' the machine would move 1" but the DRO would show 0.0394" or something very close to that number for every inch I moved.  1 divided by 25.4 is 0.03937 so that leads me to believe the problem is that while the machine and interface have entered inch mode, the part of the programming that calculates the number to display on the DRO or where to move the virtual spindle pointer on the 3D view is stuck thinking the smoothieboard is reporting back in MM, when its actually reporting back its position in inches.

itman496, 2017-07-11:
Hello, I have taken video:

https://www.youtube.com/watch?v=R_J_5d0_R5Q

cheton, 2017-07-11:
Thank you for your video. I will have a chance by today or tomorrow to make a test with my Smoothie board. I will let you know once I figured it out what probably went wrong.

itman496, 2017-07-11:
No problem, let me know if you need me to test anything else!

cheton, 2017-07-11:
Thank you for reporting this issue. It's confirmed a bug of cncjs when handling inch units for Smoothieware. You can type <b>?</b> to query status reports as shown below:

[image]

Here is the difference among Grbl, TinyG, and Smoothieware controllers:
- Grbl: mpos and wpos are always reported in mm units
- Smoothieware: mpos and wpos are reported in current active units 
- TinyG: mpos are reported in mm units, but wpos are reported in current active units

I used to think the status report of Grbl and Smoothiware can use the same way for parsing machine and work position, but now I realize that differences exist. I will fix this issue in the next few days.

itman496, 2017-07-11:
Wow, thats the first time I've ever contributed something meaningful in terms of a bug report before!  Thank you for responding so quickly.

I'm a bit new to this, may I ask how I can incorporate the fix you appear to have made in the commit into what I use? Or do I just stand by until a new version is released?

cheton, 2017-07-12:
If you're downloading a Windows installer from the releases page, you can also go to https://github.com/cncjs/cncjs/releases/tag/master-latest to get the latest changes of the master branch.

I will publish a new version (1.9.7) including this bug fix later, but might need a couple of days to include other improvements as well.

cheton, 2017-07-12:
BTW, please allow some days of delay because I have a flu and need to get plenty of rest. 😷
I will go back to work when I feel better.

AustinSaintAubin, 2017-07-12:
Get to feeling better.

itman496, 2017-07-12:
Yeah, feel better, no rush! I can deal with this issue for now no problem, the machine still functions perfectly!  Hope for a speedy recovery.

cheton, 2017-07-21:
Fixed in 1.9.7

closed 2017-07-21 by cheton