cncjs/cncjs #179: Automating cncjs

jandolina, 2017-07-18:
Is it possible to send gcode straight to cncjs from node, browser or command line?  Alternatively, can I tell cncjs to open files and execute them right away?

Thanks!

cheton, 2017-07-19:
I just made some changes to the [cncjs-pendant-boilerplate](https://github.com/cncjs/cncjs-pendant-boilerplate) project. You can send commands (e.g. `load`, `unload`, `start`, `stop`, `pause`, `resume`, `status`) to control the server, or modify the source code to support other commands.

```bash
18:42 $ ./bin/cncjs-pendant-boilerplate
? Specify which port you want to use? /dev/cu.wchusbserialfd120
Connected to ws://localhost:8000?token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpZCI6IiIsIm5hbWUiOiJjbmNqcy1wZW5kYW50IiwiaWF0IjoxNTAwNDYwOTgyLCJleHAiOjE1MDMwNTI5ODJ9.ToMMplzPvtKkeLMJuu-TfC1V5AJEz8YXLTnTUBuw4KU
Connected to port "/dev/cu.wchusbserialfd120" (Baud rate: 115200)

Type "help" to display supported options:
pendant$ help

  Commands:

    help [command...]     Provides help for a given command.
    exit                  Exits application.
    command               Enters the command-line mode.
    status                Displays sender status
    controller <command>  Available commands: state, settings
    load <file>
    unload
    start
    stop
    pause
    resume

pendant$ load circle.nc
ok
pendant$ start
ok
pendant$ status
{ sp: 1,
  hold: true,
  name: 'circle.nc',
  context:
   { xmin: 0,
     xmax: 0,
     ymin: 0,
     ymax: 0,
     zmin: 0,
     zmax: 0,
     mposx: 4.012,
     mposy: -2.988,
     mposz: 0,
     mposa: 0,
     mposb: 0,
     mposc: 0,
     posx: 4.012,
     posy: -2.988,
     posz: 0,
     posa: 0,
     posb: 0,
     posc: 0 },
  size: 153,
  total: 8,
  sent: 7,
  received: 1,
  startTime: 1500460993993,
  finishTime: 0,
  elapsedTime: 0,
  remainingTime: 0 }
pendant$ ok
Grbl 1.1f ['$' for help]
```

cheton, 2017-07-19:
If you want to add more commands, you can check out the list of supported commands below: https://github.com/cncjs/cncjs/blob/master/src/app/controllers/Grbl/GrblController.js#L747

jandolina, 2017-07-19:
Perfect!
This is just what I was looking for. Thank you.

closed 2017-07-19 by jandolina