cncjs/cncjs #181: Making "Move To Work Zero" and "Move To Machine Zero" buttons configurable

cheton, 2017-08-01:
Two motion buttons on the axes widget, <kbd>Move to Work Zero</kbd> and <kbd>Move To Machine Zero</kbd>, now default to "G0 X0 Y0 Z0" and "G53 G0 X0 Y0 Z0", respectively. For some safety reasons, one may want to go to Z0 first, and then X0Y0 to ensure machine safety. Others may want to go to X0Y0 and then Z0 for different purposes.

To fit all use cases, I'm considering providing textarea input controls in the widget settings page for configuring some motion buttons with custom commands, for examples:

#### Go To Work Zero (defaults to "G0 X0 Y0 Z0")
Example 1:
```
G0 Z0  ; Safe Z height
G0 X0Y0
```
Example 2:
````
G0 Z50  ; Safe Z height
G0 X0Y0
G0 Z0
````

#### Go To Machine Zero (defaults to "G53 G0 X0 Y0 Z0")
```
G53 G0 Z0
G53 G0 X0 Y0
```

amx1, 2017-08-02:
Hi Cheton,

perfect idea. I hard coded this already for my router with first lift the tool head G0 Z0 and then G0 X0 Y0. 

- Axel

cheton, 2017-08-25:
Release Notes of CNCjs 1.9.8
https://github.com/cncjs/cncjs/releases/tag/v1.9.8

closed 2017-08-25 by cheton