cncjs/cncjs #185: Losing Saved Settings

GeerGuy, 2017-08-13:
Sometimes I lose the widgets that I previously selected and the other settings, like the Probe values and Jog settings.  They are reset to defaults when the program is started again.  Using Windows 7.

cheton, 2017-08-15:
If you're running the app built with Electron, try removing the <b>%AppData%\cncjs</b> directory (e.g. <b>c:\Users\Username\AppData\Roaming\cncjs</b>) and try again. I guessed there might be a file permission issue with your folder settings.

If you're using a web browser, make sure your browser is not in "Incognito Mode" (i.e. private browsing), or the localStorage settings won't persist after closing the browser.

GeerGuy, 2017-08-15:
I will try that, thanks.

cheton, 2017-08-16:
I had ever encountered issue that cncjs cannot store system settings to `c:\Users\<Username>\.cncjs` on a Windows desktop due to incorrect permissions. Everything works fine after removing this file.

cheton, 2017-08-27:
Sometimes the localStorage won't persist after refreshing or closing the window.

Related issues: https://github.com/electron/electron/issues/1731, https://github.com/cncjs/cncjs/issues/41

cheton, 2017-08-27:
Found something strange in CNCjs 1.9.8, sometimes it will throw an error on app start when parsing the `c:\Users\<Username>\AppData\Roaming\cncjs\cnc.json` file, this file was used to persist local storage settings to disk (issue #41).

[image]

cheton, 2017-08-28:
Fixed the bug using a debounced function and synchronous file writing (e.g. `fs.writeFileSync`), that will definitely fix the issue that caused a broken JSON output (https://github.com/cncjs/cncjs/commit/3ddb776fb7784a1eb266fe4afcf0968624251bdd)

[image]

cheton, 2017-08-28:
Added a commit that will prompt the user for the corrupted workspace settings on program startup

[image]

cheton, 2017-08-31:
Fixed in 1.9.9

See CNCjs 1.9.9 release notes at https://github.com/cncjs/cncjs/releases/tag/v1.9.9

closed 2017-08-31 by cheton