cncjs/cncjs #187: Installation on linux failed

NervenCid, 2017-08-22:
For Linux system i got this error:

npm ERR! code EINTEGRITY
npm ERR! sha512-FJjCmponv4KCYLkZGYqd7L2cO4mmfjTgo6nlFO7G/V/Z9iP+fDgJvuA0/daALQGJ18aFGyduS4Ep1JkusQC62A== integrity checksum failed when using sha512: wanted sha512-FJjCmponv4KCYLkZGYqd7L2cO4mmfjTgo6nlFO7G/V/Z9iP+fDgJvuA0/daALQGJ18aFGyduS4Ep1JkusQC62A== but got sha512-MPigyuJsnbli1XJurgkdfYQn7u3OU1hwth69yA2abJfixUk9mbsrknT0JvI/RmDEnC+Jt/B80M8tHyGFZNuRNQ==. (442997 bytes)

nsfilho, 2017-08-22:
I use cncjs in Linux without problem. If you could, please, send a little more information, i.e. whats version of your npm? linux version? full command-line?

cheton, 2017-08-22:
Similar to https://github.com/cncjs/cncjs/issues/172

NervenCid, 2017-08-23:
@nsfilho 

I'm using Raspbian  STRETCH  with a raspberry pi 3.

I tryed using 

>npm install -g cncjs

and 

>sudonpm install -g cncjs

But both failed

cheton, 2017-08-23:
@NervenCid 

You can find a complete setup guide for RPi 3 at https://github.com/cncjs/cncjs/wiki/Raspberry-Pi-Setup-Guide#install-cncjs

For my personal preference, I recommend using [Node Version Manager](https://github.com/creationix/nvm) to install Node.js, it allows you to install NPM packages under user home, so you don't need to deal with system directory permissions.

If you're using the system default Node.js, you need to append `--unsafe-perm` during installation, like so: 

```
sudo npm install -g cncjs@latest --unsafe-perm
```

cheton, 2017-10-13:
Run `npm install npm@latest -g` to install npm 5.4.2 will fix this issue.

https://github.com/cncjs/cncjs/issues/172#issuecomment-336237332

closed 2017-10-13 by cheton