cncjs/cncjs #189: "Go To" should move Z axis first/last

erpalma, 2017-08-24:
I think that when you press Go To Work Zero, for example, the machine should move the Z axis independently from the others. If targetZ < curZ, then X/Y should move first since it might happen that the curZ position is set to avoid obstacles (holding screws?). Vice versa, when targetZ >  curZ, this axis should move first in order to raise the bit from the surface (drill bit in a hole?). 

Is this crazy? :)

cheton, 2017-08-24:
Similar to issue #181

With a customizable button, both cases can be covered by moving to a safe Z height before going to the zero position.

```
G0 Z50  ; Safe Z height
G0 X0Y0
G0 Z0
```

This will be available in 1.9.8 soon.

cheton, 2017-08-25:
Release Notes of CNCjs 1.9.8
https://github.com/cncjs/cncjs/releases/tag/v1.9.8

You can configure "Go To Work Zero" and "Go To Machine Zero" commands under Axes settings

closed 2017-08-25 by cheton