cncjs/cncjs #190: Which g2core versions are supported/preferred?

mennodegraaf, 2017-08-24:
Recently I discovered CNCjs after having earlier experience with mainly Chilipeppr and LaserWeb. I do have a couple of CNC machines with different controllers: GRBL 1.1f, TinyG and g2core on an ArduinoDue with my custom design g2core Shield.

I am using CNCjs 1.9.7, Node.js 4 and NPM 5.3.0 installed on a Linux64 machine, which works great with the GRBL 1.1f controller. I did some preliminary tests with g2core and it turned out that the old g2core master branch was working properly, but with the recent g2core edge branch (I used 100.26) I could never properly complete a gcode job. It always stopped processing somewhere halfway.

Unfortunately, I could not find out which versions of g2core are supported in the documentation or which version is preferred?

Also, please let me know, how I can get more detailed logging information of the issue?

ril3y, 2017-08-25:
> It's most likely an incompatible firmware running on your Arduino Due. May
> I know where did you get the firmware of g2core edge branch, or what
> parameters you're using to compile the source code from the g2core edge
> branch?

edge branch 168 is going to be pushed to master shortly fyi.

I can build you a Due bin if you want @cheton.
https://github.com/synthetos/g2/tree/dev-168-gquintic

cheton, 2017-08-25:
@ril3y Great! That will be helpful. You can put it somewhere or attach file to comments, I will give it a try later.

amx1, 2017-08-25:
I am using 100.26 without any problems. But Cheton is right, you have to be careful to take the right configuration parameters and the right motate branch! I checked also the quintic 168 branch. It is runnning, but there are a lot of changes ongoing and if you read the forum, some feature like the new math for the motion planner are not yet tested to the end. BTW. the 168 bin is 20k bigger in my case. If you don't need the new marlin features  I recommend to stay with the edge version and wait for a merge into edge.

What are the last commands and messages of cncjs before the job stops? Do you get an error message or what happens? You can increase the verbosity of cncjs by starting it with -vvv.  

Best

mennodegraaf, 2017-08-25:
@amx1 Good to know that using 100.26 is in principle possible.

I am pretty sure I am using the right Motate as I checked it out with all dependencies using:

git clone https://github.com/synthetos/g2.git -b edge --recursive

Then copied settings_shapeoko2.h to settings_eShapeoko.h and adjusted it for my machine. After that, I built it using

make BOARD=gShield SETTINGS_FILE=settings_eShapeoko.h

I will try to get more more output by starting cnc js with -vvv

mennodegraaf, 2017-08-25:
For some weird reason it now works properly after checking out a fresh g2core edge branch and copying my settings file to it. Note: also the gShield-pinout.h had to be copied, because that was deleted from edge.

closed 2017-08-30 by cheton