cncjs/cncjs #191: Probing widget on g2core does not set coordinate offset properly

mennodegraaf, 2017-08-25:
After playing around with a cheap chinese probe, I noticed that the CNCjs 1.9.7 probe widget does not set the coordinate offset properly. Probing works correctly, after that the coordinate offset is set with:

; Set the active WCS Z0
G10 L20 P0 Z20

This seems to be accepted by GRBL, but not by g2core 100.26. According to https://github.com/synthetos/g2/wiki/Gcode-Coordinate-Offsets#g10-ln-set-parameters the supported P codes are P1-P6 for G54-G59 respectively. By manually using P1 it is accepted by g2core.

cheton, 2017-08-25:
`G10 L20 P- axes` (P number is in the range 0 to 9) is a EMC2 (linuxcnc.org) standard. However, P0 is currently not supported by TinyG and g2core, I will change the P number to fit the corresponding G54-G59 work coordinate system in the 1.9.9 release.

cheton, 2017-08-25:
BTW, you can upgrade CNCjs to 1.9.8, it fixed some critical bugs especially for Grbl. Check out latest release notes at https://github.com/cncjs/cncjs/releases

mennodegraaf, 2017-08-25:
Great, thanks

aldenhart, 2017-08-25:
TinyG and G2core follow the convention that P1 is coordinate system G54 -- going to P6 = G59. 
This is how it's defined for Tormach, Haas, Fanuc, and most of the "big boys":
https://www.tormach.com/g10.html
http://diy.haascnc.com/node/30342#gsc.tab=0
http://www.cnccookbook.com/CCCNCGCodeG10SettingToolWorkOffsetsInGCode.html

By convention P0 refers to the active coordinate system. P0 is not currently implemented in TinyG and G2core.

cheton, 2017-08-31:
See CNCjs 1.9.9 release notes at https://github.com/cncjs/cncjs/releases/tag/v1.9.9

closed 2017-08-31 by cheton