cncjs/cncjs #192: CNC started by itself

nkings10, 2017-08-26:
#### Description

I have been unable determine the exact cause of this issue. I'm reporting it just incase anyone else in the future has a similar glitch.

I had left home with a cut running and it had finished whilst i was out, as i come home it began cutting the same cut again.

I woke my laptop from sleep and refreshed the web interface and could see that it was about 3 mins into the cut. The machine was running fine and simply going over what it had already cut. It was almost like someone had walked in and pressed play as i had arrived home. The TinyWeb pendant was still in recent apps on my phone and the only thing i can boil it down to was that when my phone reconnected to WiFi it caused a glitch and started the cut again.

The other strange thing is that when pressed pause then stop on the laptop the machine started heading for the back of the y-axis and then hit the back of the machine. After looking closer it had completely reset its machine position. However the 3 mins of cutting it done right before i pressed pause/stop was exactly in the right place.

#### Versions

CNCjs: 1.9.6
Node.js: 7.10.1
NPM: 5.3.0

#### How Do You Install CNCjs?

NPM

#### CNC Controller

3DTek GRBL AIO

#### Hardware

Raspberry Pi

#### Operating System

Android / MacOS

cheton, 2017-08-26:
Not sure what is the exactly reason that caused the second cut. For security considerations, I'd recommend users enable the account management to prevent someone from doing somethig on the web interface.

Also, I see you were runing CNCjs 1.9.6. I suggest you upgrade to 1.9.8, which resolves a critical buffer overrun issue especially for Grbl, and it will no longer trigger a soft reset (^x) on job completion unless you manually stop a running G-code program on the web interface.

nkings10, 2017-12-04:
Hey Chenton, have not had this issue happen again. Couldn't replicate it either and that was before upgrading to the latest version. Thanks for the help.

closed 2017-12-04 by nkings10