cncjs/cncjs #193: Fails to start after 1.9.8 update

metanurb21, 2017-08-27:
#### Description
Been using 1.9.7 for a while and it's mint. Got the message to update to 1.9.8, so I did per instructions.
Now it won't start: error: uncaughtException: Module version mismatch. Expected 46, got 51

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.8
- Node.js: 4.8.4
- NPM: 2.15.11

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [x ] Linux

metanurb21, 2017-08-27:
Wrong node version. :(

closed 2017-08-27 by metanurb21

cheton, 2017-08-27:
To support Node.js v8, I have to upgrade the node-serialport module from v4 to v5 (https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/5.0.0). However, node-serialport v5 will only support LTS Node.js platforms, and it has dropped support for Node.js v0.10, v0.12, v5, and v7. This is a notable breaking change, I will upgrade node-serialport to v5 in CNCjs v2.0 as a major release.

For now, you can use Node.js v4 and v6 if you need stability, or use Node.js v7 if you want to play with the latest Node.js features.

cheton, 2017-08-27:
Added a section of supported Node.js versions to the README
https://github.com/cncjs/cncjs/tree/dev#supported-nodejs-versions