cncjs/cncjs #194: Support homing on g2core

mennodegraaf, 2017-08-28:
Using CNCjs 1.9.8 on g2core 100.26, the top right homing button does not start the homing procedure for some reason. If I manually request the homing procedure with:

G28.2 X0 Y0 Z0

it does work properly, so the g2core side is ok. If I connect to GRBL homing does work, but obviously it requires a different homing command ($H) than the g2core.

cheton, 2017-08-28:
The homing command that was currently used for TinyG and G2core controller is 
```
{home:1}
```

After reading the documentation (https://github.com/synthetos/g2/wiki/Homing-g2core), it looks that I was using the wrong command. `{home:1}` is not a homing command, it's used to mark the machine as homed.

I will change it to `G28.2 X0 Y0 Z0` or `G28.2 X0 Y0 Z0 A0` in the upcoming 1.9.9 release.

mennodegraaf, 2017-08-28:
@cheton that would do the trick for now, but maybe towards the future a more robust solution would be needed as the g2core can disable homing for each axis if the homing input is configured at 0. I am not sure if it would give an error message if the sequence would contain an axis for which homing was disabled.

mennodegraaf, 2017-08-28:
@cheton related to this another useful feature is the ability to home individual axis. For example Chilipeppr does this by adding this to the dropdown menu for each axis. Perhaps you can add this as well for g2core/TinyG (for GRBL only the complete machine can be homed)

cheton, 2017-08-29:
https://github.com/synthetos/g2/wiki/Homing-g2core#homing-cycles

> The homing sequence is fixed and always starts with the Z axis (if requested). The sequence runs ZXYABC, but skipping all axes that are not specified in the G28.2 command. Attempting to home an axis that is not configured will result in an error.

It will give an error message for misconfigured axis.

cheton, 2017-08-31:
See CNCjs 1.9.9 release notes at https://github.com/cncjs/cncjs/releases/tag/v1.9.9

closed 2017-08-31 by cheton