cncjs/cncjs #195: Q: Possible to share a single session among multiple browsers

kikolobo, 2017-08-29:
Is it possible to share a single session among multiple browsers. Currently if I join on different browser the server attempts to open a new serial port. Seems like a fair behavior, but today I wanted to stop a job using my phone while my laptop was farther way and couldn't. So I was wondering it this is something that can be setup?

cheton, 2017-08-29:
CNCjs is designed to share a single serial connection among multiple browsers. If you open a port on a browser, and open the same port again on another browser, both will share the same serial connection.

For your case, if you have loaded a G-code program on a browser (e.g. laptop), but didn't see it on another browser (e.g. mobile phone), you can try to refresh the browser on your phone, it will load data from CNCjs server and display the G-code program.

Manual refresh is just not user-friendly, I can try to improve this part, so you won't have to perform a manual refresh on another device to display the currently loaded G-code program.

kikolobo, 2017-08-29:
Got it.. I thought I Could not open the same serial port twice on different sessions.

cheton, 2017-08-29:
Note: Sp far only serial connection, accounts, commands, events, and macros will share among different browsers. The workspace settings (e.g. container, widget, etc.) shown below are currently not shareable, because data are persisted to browser's localStorage, not on the server.

[image]

cheton, 2017-08-31:
I made some improvements in 1.9.9, now you will see concurrent display among multiple browsers without the need of manually refresh.

See CNCjs 1.9.9 release notes at https://github.com/cncjs/cncjs/releases/tag/v1.9.9

closed 2017-08-31 by cheton