cncjs/cncjs #196: 2nd browser/machine connection problems.

digipigeon, 2017-09-05:
#### Description

Whilst connecting from any other machine or browser apart from the initial browser that I used to set things up no active connection can be established.

Tried 2 other browsers and multiple other devices to access. Upgraded from 1.9.0-alpha.5 to 1.9.9 and still have this same problem on 2x separate identical hardware/software.

Chrome on first machine used works 100%.

Any other browser on the first machine or any browser including chrome on any other machine does not. The page loads up web socket connects, etc. But the connection box and everything else remains disabled.

#### Versions

- CNCjs: 1.9.9
- Node.js: 8.4.0
- NPM: 5.3.0

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi 3
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [X] Linux (rasbian-jessie)

cheton, 2017-09-06:
@digipigeon 

You can try the following troubleshooting steps to find errors:
1. Launch a browser and append the query string parameter `?log_level=debug` to turn on debug output. 
2. Open <b>Developer Tools</b> and go to browser console, and check if there are JavaScript errors logging in the browser.

[image]

digipigeon, 2017-09-09:
Hi, unfortunately no errors

[image]

cheton, 2017-09-10:
In your screenshot, your port is not connected on the second browser. Have you ever tried selecting the same port again and then click on [Open] button on the connection widget?

SteveHogg, 2017-09-10:
Possibly a dumb response here since I've only used the windows app so far, though I plan to switch to using an RPi so I'm interested to know. Is there a login needed for cncjs? I know on my 3d printer the UI looks fine but most of not all the controls are disabled until I log in to octoprint. Reason I'm wondering is I see a little person image on the main cncjs webpage (not close to my machine right now) but you screenshot does not have one in the top right corner. If login is needed then it makes sense for it all to be disabled.

cheton, 2017-09-10:
@SteveHogg 
Connecting to a port is a required action, you need to select a port to connect a client to the server to enable UI controls for serial communication.

The default settings will not enable secure web access. If you have enabled account access in <b>Settings &gt; My Account</b>, you will be redirected to a login page as below, it requires you to sign in before accessing the page.
[image]

digipigeon, 2017-09-10:
@cheton there is no option to select the ports form the UI. All this is greyed out / contain no selectable ports.

Note in the screenshot, log timestamp 22:14:26.717 contains an empty array for the ports available.

cheton, 2017-09-11:
The message shown at the timestamp 22:14:26.717 is a misleading information, both the ports and baud rates are user-defined values in the `~/.cncrc` file, not a list of detected ports.
See https://github.com/cncjs/cncjs#file-format
```json
{
  "ports": [
     {
       "comName": "/dev/ttyAMA0",
       "manufacturer": ""
     }
  ],
  "baudrates": [250000]
}
```

From your screenshot, the "serialport:list" command did return an array containing two ports.
[image]

May I know if you can see above message when clicking on the refresh port button?
[image]