cncjs/cncjs #197: Cant connect to arduino

saxa, 2017-09-10:
#### Description

Hi I have installed and configured correctly cncjs and all went well and smoothly. Thanks for such a nice program. Problem is that I can not connect to the grbl on my Arduino Uno. 
I have successfully uploaded grbl to my arduino with the arduino ide, but cncjs can not connect to it.
Definetly i]this is an permissions issue. I can connect to my arduino from minicom on the /dev/ttyACM0
port and query the commands and so on.
It seems that cncjs have a hardcoded ttyS2 in it as I can see the following lines on my terminal:

bash-4.3$ cnc   
2017-09-10T21:27:33.720Z - info: init Loading configuration from "/home/sasa/.cncrc"
2017-09-10T21:27:33.723Z - info: init Watching "/home/sasa/cncjs/watchdir" for file changes.
2017-09-10T21:27:33.827Z - info: init Starting the server at http://127.0.0.1:8000
2017-09-10T21:27:47.895Z - error: service:cncengine { Error: Command failed: udevadm info --query=property -p $(udevadm info -q path -n /dev/ttyS2)
/bin/sh: udevadm: command not found
/bin/sh: udevadm: command not found

    at ChildProcess.exithandler (child_process.js:270:12)
    at emitTwo (events.js:125:13)
    at ChildProcess.emit (events.js:213:7)
    at maybeClose (internal/child_process.js:927:16)
    at Socket.stream.socket.on (internal/child_process.js:348:11)
    at emitOne (events.js:115:13)
    at Socket.emit (events.js:210:7)
    at Pipe._handle.close [as _onclose] (net.js:545:12)
  killed: false,
  code: 127,
  signal: null,
  cmd: 'udevadm info --query=property -p $(udevadm info -q path -n /dev/ttyS2)' }
2017-09-10T21:28:13.155Z - error: service:cncengine { Error: Command failed: udevadm info --query=property -p $(udevadm info -q path -n /dev/ttyS2)
/bin/sh: udevadm: command not found
/bin/sh: udevadm: command not found

    at ChildProcess.exithandler (child_process.js:270:12)
    at emitTwo (events.js:125:13)
    at ChildProcess.emit (events.js:213:7)
    at maybeClose (internal/child_process.js:927:16)
    at Socket.stream.socket.on (internal/child_process.js:348:11)
    at emitOne (events.js:115:13)
    at Socket.emit (events.js:210:7)
    at Pipe._handle.close [as _onclose] (net.js:545:12)
  killed: false,
  code: 127,
  signal: null,
  cmd: 'udevadm info --query=property -p $(udevadm info -q path -n /dev/ttyS2)' }
^C

I am running this as a normal user and I am using Slackware Linux 64bit.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [ ] Windows
- [ ] Mac
- [x] Linux

saxa, 2017-09-10:
By the way as root user I get the following:

root@vivo64:~# udevadm info -q path -n /dev/ttyS2 
/devices/platform/serial8250/tty/ttyS2
root@vivo64:~# udevadm info -q path -n /dev/ttyACM0
/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/tty/ttyACM0
root@vivo64:~# udevadm info --query=property -p /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/tty/ttyACM0
DEVLINKS=/dev/serial/by-id/usb-Arduino_Srl_Arduino_Uno_95437313035351F0E121-if00 /dev/serial/by-path/pci-0000:00:14.0-usb-0:2:1.0
DEVNAME=/dev/ttyACM0
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/tty/ttyACM0
ID_BUS=usb
ID_MM_CANDIDATE=1
ID_MODEL=Arduino_Uno
ID_MODEL_ENC=Arduino\x20Uno
ID_MODEL_FROM_DATABASE=Arduino Uno Rev3
ID_MODEL_ID=0043
ID_PATH=pci-0000:00:14.0-usb-0:2:1.0
ID_PATH_TAG=pci-0000_00_14_0-usb-0_2_1_0
ID_PCI_CLASS_FROM_DATABASE=Serial bus controller
ID_PCI_INTERFACE_FROM_DATABASE=XHCI
ID_PCI_SUBCLASS_FROM_DATABASE=USB controller
ID_REVISION=0001
ID_SERIAL=Arduino_Srl_Arduino_Uno_95437313035351F0E121
ID_SERIAL_SHORT=95437313035351F0E121
ID_TYPE=generic
ID_USB_CLASS_FROM_DATABASE=Communications
ID_USB_DRIVER=cdc_acm
ID_USB_INTERFACES=:020201:0a0000:
ID_USB_INTERFACE_NUM=00
ID_VENDOR=Arduino_Srl
ID_VENDOR_ENC=Arduino\x20Srl\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20
ID_VENDOR_FROM_DATABASE=dog hunter AG
ID_VENDOR_ID=2a03
MAJOR=166
MINOR=0
SUBSYSTEM=tty
USEC_INITIALIZED=32691912552
root@vivo64:~#

cheton, 2017-09-11:
You can specify the device path `/dev/ttyACM0` in the `~/.cncrc` file, check out FAQ at https://github.com/cncjs/cncjs/wiki/FAQ#raspberry-pi-error-opening-serial-port-ttyama0

saxa, 2017-09-11:
Hi , thanks for the suggestion, I will try that tonight at my home, but I can confirm that I have the /dev/ttyACM0 in the .cncrc file.

saxa, 2017-09-12:
Ok here is few files which I think they are relevant:

my user is part of group dialout:x:16:uucp,sasa

`bash-4.3$ cat /home/sasa/.cncrc
{
    "ports": [
        {
            "comName": "/dev/ttyACM0",
            "manufacturer": "ArduinoUno"
        }
    ],
    "baudrates": [
        115200,
        250000
    ],
    "watchDirectory": "/home/sasa/cncjs/watchdir",
    "accessTokenLifetime": "30d",
    "allowRemoteAccess": false,
    "controller": "",
    "state": {
        "checkForUpdates": true
    },
    "commands": [
        {
            "title": "Update (root user)",
            "commands": "sudo npm install -g cncjs@latest --unsafe-perm; pkill -a -f cnc",
            "id": "787080af-154d-4056-a4fd-c3e62923a641",
            "enabled": false,
            "mtime": 1505078927041
        },
        {
            "title": "Update (non-root user)",
            "commands": "npm install -g cncjs@latest; pkill -a -f cnc",
            "id": "c64285d4-e2f6-4163-92dc-dc5d02a91d5a",
            "enabled": true
        },
        {
            "title": "Reboot",
            "commands": "sudo /sbin/reboot",
            "id": "14dc3d78-f199-4834-b379-734ed11afd2f",
            "enabled": true
        },
        {
            "title": "Shutdown",
            "commands": "sudo /sbin/shutdown",
            "id": "2833ea0b-f2a9-4d9d-b814-a33cea575e0d",
            "enabled": true
        }
    ],
    "events": [],
    "macros": [],
    "users": [
        {
            "id": "7304a00a-2eb5-4d49-bb88-54b4434c16d8",
            "mtime": 1505078915654,
            "enabled": true,
            "name": "admin",
            "password": "$2a$10$l8mXnmYFCFBHc/FvS7F5F.1ZeIyrzTP35iFYNbDM1yZ4dpBQ.OgM."
        }
    ],
    "secret": "$2a$10$/iBdminjcY1tEGsxr7dUsu"
}bash-4.3$ 
`

`root@vivo64:~# tail -f /var/log/messages                                                                     
Sep 12 18:05:34 vivo64 blueman-mechanism: loading Ppp  
Sep 12 18:05:34 vivo64 blueman-mechanism: loading RfKill  
Sep 12 18:05:34 vivo64 blueman-mechanism: loading Rfcomm  
Sep 12 18:06:06 vivo64 blueman-mechanism: Exiting  
[... 32 lines cut ...]
ep-lF6DUctQnHGc_b2UeU_TNJjYP8VNF POST /api/signin 401 11.129 ms
ep-lF6DUctQnHGc_b2UeU_TNJjYP8VNF POST /api/signin 200 237.267 ms
2017-09-12T23:23:48.814Z - debug: service:cncengine New connection from 127.0.0.1: id=SOx70hhXCQBDYwu-AAAA, token.id=7304a00a-2eb5-4d49-bb88-54b4434c16d8, token.name=admin     at Namespace.<anonymous> (/home/sasa/cncjs/nvm/versions/node/v8.4.0/lib/node_modules/cncjs/dist/cnc/app/index.js:13677:17)
2017-09-12T23:23:49.290Z - debug: service:cncengine socket.list(): id=SOx70hhXCQBDYwu-AAAA     at Socket.<anonymous> (/home/sasa/cncjs/nvm/versions/node/v8.4.0/lib/node_modules/cncjs/dist/cnc/app/index.js:13706:21)
2017-09-12T23:23:49.328Z - error: service:cncengine { Error: Command failed: udevadm info --query=property -p $(udevadm info -q path -n /dev/ttyS2)
/bin/sh: udevadm: command not found
/bin/sh: udevadm: command not found

    at ChildProcess.exithandler (child_process.js:270:12)
    at emitTwo (events.js:125:13)
    at ChildProcess.emit (events.js:213:7)
    at maybeClose (internal/child_process.js:927:16)
    at Socket.stream.socket.on (internal/child_process.js:348:11)
    at emitOne (events.js:115:13)
    at Socket.emit (events.js:210:7)
    at Pipe._handle.close [as _onclose] (net.js:545:12)
  killed: false,
  code: 127,
  signal: null,
  cmd: 'udevadm info --query=property -p $(udevadm info -q path -n /dev/ttyS2)' } '    at /home/sasa/cncjs/nvm/versions/node/v8.4.0/lib/node_modules/cncjs/dist/cnc/app/index.js:13710:29'
ep-lF6DUctQnHGc_b2UeU_TNJjYP8VNF GET /api/state?_=1505258629265 200 6.859 ms
ep-lF6DUctQnHGc_b2UeU_TNJjYP8VNF GET /api/commands?_=1505258629267&paging=false 200 9.459 ms
ep-lF6DUctQnHGc_b2UeU_TNJjYP8VNF GET /api/macros?_=1505258629860&paging=false 200 4.661 ms
ep-lF6DUctQnHGc_b2UeU_TNJjYP8VNF GET /api/version/latest?_=1505258630019 200 962.706 ms
`

saxa, 2017-09-12:
Here is a screenshot of the browser where you can see that there is no connection at all, also no port is listed when you open the dropdown list of ports in the Connection widget.
[image]

saxa, 2017-09-12:
Just as a test from same user I can easily connect with minicom -D /dev/ttyACM0 
[image]

cheton, 2017-09-13:
According to your logs, I found the error was thrown by the node-serialport module in the following lines of code:
https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/4.0.7/lib/list-unix.js#L88-L91
```js
function lookupPort(file) {
  var udevadm = 'udevadm info --query=property -p $(udevadm info -q path -n ' + file + ')';
  return execAsync(udevadm).then(udevParser);
}
```

It looks like <b>udevadm</b> is not on the default path. You can see a related issue here: https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1085.

Can you check if you can resolve the absolute path of `udevadm` (e.g. `which udevadm`) with your user account?

saxa, 2017-09-13:
Dear Cheton, many thanks for your time. Yes , this is exactly what I was checking up yesterday on my machine. I found that on Slackware we do have "udevadm" command in /sbin/ but only root can use that command. I was looking also for a way on how to make that available to node, but I have not found a way around it. One thing I was thinking was to install noge systemwide as a slackware package and see if that solves it. Other option is in some way to hardcode the /dev/ttyACM0 into the 
cncjs as for some reason what is in the .cncrc config it does not get picked up correctly or it changes during runtime someway.

But this bug report you showed me its probably the solution, I do not have much experience with nvm/node and all that js stuff.

saxa, 2017-09-13:
Ok, I will try to symlink the udevadm in /usr/bin and see if that solves my problem.

saxa, 2017-09-17:
Ok, I just made a symlink as root # ln -s /sbin/udevadm /usr/bin/udevadm and it worked out.
As it is stated on that bug report this is not a sollution but its a problem of npm serial port. I will see if Slackware solves this in some other way. For now it seems all to work ok. 

Many thanks Cheton !

closed 2017-09-17 by saxa

cheton, 2017-09-17:
Would like mind to write down your resolution of issue in the https://github.com/cncjs/cncjs/wiki/Troubleshooting page? This information will help save time when someone encountered similar problem. Thank you 😄

saxa, 2017-09-18:
Sure, dont worry, I will do that but firstly I want to see if I find a good solution rather than a hack.