cncjs/cncjs #198: Set all Axis's to Zero button on the main screen interface instead of hidden in sub menu

markcr, 2017-09-11:
#### Description

Feature Request - It would be great to bring the Button to zero all axis's to the front gui as its a high use button that seems hidden away. Nested into the Go to Work Zero buttons would be the most ideal spot.

Fantastic work in cncjs its the perfect solution to work with my new xcarve.

#### Versions

- CNCjs: 1.9.9
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [] NPM
- [x ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x ] Windows
- [ ] Mac
- [ ] Linux

kikolobo, 2017-10-09:
+!

cheton, 2018-02-03:
Now you can configure custom MDI buttons with CNCjs 1.9.13.

#### Custom Commands

Go to **Axes Settings** to add custom commands and enable manual data input (MDI) support.

[image]

Toggle MDI mode

[image]

closed 2018-02-03 by cheton