cncjs/cncjs #199: Tinyweb pendant plugin for windows desktop version?

markcr, 2017-09-11:
#### Description

Is it possible to access the tinyweb pendant web (https://github.com/cncjs/cncjs-pendant-tinyweb/releases) plugin using the electron windows desktop version? - I could'nt find a way to install it similar to the raspberry pi instructions?

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

cheton, 2017-09-11:
It's doable, but it requires you to open the Developer Tools and check the server listen port (e.g. 12345) from the Network tab, then you can access to the pendant UI via http://127.0.0.1:12345/pendant.

I will add shortcut support to the top menu that allows you to open a new browser window for the mounted folder, that will be much easier to access the pendant UI.

markcr, 2017-09-11:
Fantastic - sounds great look forward to seeing it :)

cheton, 2017-09-22:
[1.9.10 release notes](https://github.com/cncjs/cncjs/releases/tag/v1.9.10)

[image]

closed 2017-09-22 by cheton