cncjs/cncjs #200: Spanish contributions

yeyeto2788, 2017-09-13:
#### Spanish Contributions

Hey guys! 

Just to upload some files translated into Spanish. As the [README](https://github.com/cncjs/cncjs/README.md) says I can create a ticket an upload the files here right?

I had to change the extension of the file so I could uploaded it. I'll be working on the other files as soon as someone reply to this, just to make sure I'm doing it in the right way.

[controller.txt](https://github.com/cncjs/cncjs/files/1300125/controller.txt)
**NOTE:** This should be the controller.json file on [https://github.com/cncjs/cncjs/tree/master/src/web/i18n/es](https://github.com/cncjs/cncjs/tree/master/src/web/i18n/es)

Regards

cheton, 2017-09-15:
I added a localization section to the contribution guide for learning how to contribute new translations. You can follow the instructions at https://github.com/cncjs/cncjs/blob/master/CONTRIBUTING.md#localization and create a pull request. This is a common practice on GitHub to merge commits.

If you still have problems using git, you can upload files to here, I will make updates accordingly.

yeyeto2788, 2017-09-15:
Hey @cheton,

I have added the translations via pull request.

#201 I hope I did it right.

Best regards.

cheton, 2017-09-15:
Thank you for your PR, I've merged it.

closed 2017-09-15 by cheton

yeyeto2788, 2017-09-18:
No problem,

Anything else I can help with?

I'm currently finishing the last file to have all 100% translated. I guess at this point we're at about 83% translated. Just in case you want to add it to the documentation.

Regards.

cheton, 2017-09-18:
Current translations will be released with CNCjs 1.9.10. You can always submit a PR if you have any updates to the resource files. Thank you!