cncjs/cncjs #202: Raspberry Pi CNCjs Pre-built Image (is their a need)

AustinSaintAubin, 2017-09-15:
#### Description

I am deciding if I should spend time to build a CNCjs Raspberry Pi image that can be installed on an SD card and then booted. 
- Is their a want for this? ( Please thumbs up this, or comment )
- What should be included? (See: [Raspberry Pi Setup Guide](https://github.com/cncjs/cncjs/wiki/Raspberry-Pi-Setup-Guide))
  - CNCjs w/ pm2 autostart
  - PM2
  - Mjpegstreamer
  - FFMpeg (camera recorder)
  - [cncjs-pendant-raspi-gpio](https://github.com/cncjs/cncjs-pendant-raspi-gpio)
- How often should their be an update?
  - Every major release. 1.0.0, 2.0.0, 3.0.0
  - Every major + middle: 1.5.0, 2.0.0, 2.5.0, 3.0.0
  - Every release: 1.9.8, 2.0.0, 2.0.1, 2.0.2

#### Hardware

- [x] Raspberry Pi

markcr, 2017-09-16:
Or alternatively maybe python setup script to automate the procedure which would save you building images every update ?

mayhem2408, 2017-09-16:
I have a couple of simple scripts to install cncjs on the raspberry pi.
----- install-cncjs.sh -----
echo "prefix=~/.npm" >> ~/.npmrc
cat ~/.npmrc
npm config set user 0
npm config set unsafe-perm true

npm install -g socket.io
npm install -g socket.io-parser
npm install -g cncjs
export PATH=$PATH:~/.npm/bin/
-------- EOF -------

and a desktop shortcut 

----- cncjs.Desktop -----
[Desktop Entry]
Name=Start CNC.JS(Run once, wait for terminal window to open)
Version=1.0
Comment=Launches application
Exec=/home/pi/.npm/bin/cnc
Icon=/usr/share/pixmaps/openbox.xpm
Terminal=true
Type=Application
Categories=Education
----- EOF -----

These are for the rasbian OS and works great for me.

fk47, 2017-10-10:
I think this is a great idea. Would be nice if the image would also include the prerequisites for using the PS3-joystick. I can not get the bluetooth to work at all with the ttyS0-port instead of the ama0 port (or vice versa). I am a bit confused and do not have much time to fix it. So setting up the correct ports so that cncjs and bluetooth controllers can be used simultaneously would be very helpful.

MitchBradley, 2017-10-18:
My best guess is that trying to maintain a set of rPi disk images would turn into a version nightmare.

Speaking of which - https://github.com/cncjs/cncjs/wiki/Raspberry-Pi-Setup-Guide refers to NodeJS v8 in the "via Package Manager" section and v4 in the "via NVM" section.  As a complete n00b to the Node ecosystem, this ended up sending me down some rather deep ratholes.

The background is that I am using cncjs (successfully, finally) on a Pi Zero W.  It's armv6 so the "via Package Manager" instructions don't work, as the armv6 architecture is not supported by that script.  When I followed the "via NVM" instructions and dutifully chose v4 - which resulted in a downgrade from the v6 LTS that I had somehow managed to install - all sorts of hell broke loose, such as breaking npm.

Somehow I managed to get cncjs working, despite my ignorance, with Node v6.

Now that I understand things marginally better, I managed to get Node v8 going on the Pi Zero, using the v8.5.0 install script at https://github.com/sdesalas/node-pi-zero .  Of course that resulted in a serialport module version mismatch, but I finally figured out the recipe
```
cd cncjs/node_modules/serialport
npm --build-from-source install
```
Oh - another version thing - the "Auto Start Options" section contains references to v4.5.0 in a PATH entry that might not be relevant anymore.

AustinSaintAubin, 2017-10-22:
I will work to update the [Raspberry Pi Setup Guide](https://github.com/cncjs/cncjs/wiki/Raspberry-Pi-Setup-Guide).

MitchBradley, 2017-10-23:
Cool, let me know if I can help.

AustinSaintAubin, 2017-10-30:
If you want to help with [documentation](https://github.com/cncjs/cncjs/wiki/Raspberry-Pi-Setup-Guide) in any way, then by all means have at it. 
I need to take some time and update the documentation, but I have not had time for CNC work... new baby & stuff at work is a handful.

Feel free to update the [wiki](https://github.com/cncjs/cncjs/wiki/Raspberry-Pi-Setup-Guide) as needed. or redesign for better organisation. Just test before posting.
Anything to help the community would be appreciated.

I have a fresh Pi 3 I run everything on to verify & test.
I plan to also test on a Pi Zero in the future... when I have time.

MitchBradley, 2017-10-30:
Fair enough.  I wasn't sure how you felt about other people editing your page.  I will be careful.

I have a Pi2 in addition to the ZeroW.  Just FYI, your instructions worked quite well on the 2.

AustinSaintAubin, 2017-11-01:
I welcome improvements.
Its the CNCjs community wiki, I just contribute to it.
If I didn't want it edited I would have put it on a personal blog.
Also, The wiki is version controlled, so submissions can be tracked and undone if needed.

Glad instructions worked on the pi2.
Looks like I need to work on the Pi Zero though...

MitchBradley, 2017-11-01:
Cool.  As a newcomer to this community, I just wanted to tread lightly until relationships are established.

Agreed about the need for separate pages.  Organizing information is always tricky.

cheton, 2017-11-02:
@MitchBradley 

Feel free to edit the wiki, everyone can help improve it. You can create a new page to organize the content if needed.

AustinSaintAubin, 2017-11-12:
Going to focus on keeping the Wiki up to date.

closed 2017-11-12 by AustinSaintAubin

MitchBradley, 2018-02-12:
Welcome to the Linux world, where there are so many different configurations and so much version churn that getting anything to work is a struggle and keeping it working over time is a never-ending battle.  I have been using Unix derivatives continuously since the '70s and I still have trouble.