cncjs/cncjs #204: Incorrect work offset setting in inch mode

MitchBradley, 2017-09-16:
#### Description

When in inch mode, e.g. after typing G21 in the serial console, changing a work offset by typing a new value into the XYZ widget multiplies the value by 25.4, even though the units are shown in inches and the controller is interpreting coordinates in inches.

For example, suppose that machine X and work X are both 0.  If you type 1 into the X work offset and hit enter, the Work DRO says 25.4 instead of 1, and the command "G10 L20 P1 X25.4" is issued to the controller.

I believe this is incorrect, and that it can be fixed by deleting three occurrences of 
```
                                        if (units === IMPERIAL_UNITS) {
                                            value = in2mm(value);
                                        }
```

in src/web/widgets/Axes/DisplayPanel.jsx

#### Versions

- CNCjs: 1.9.9
- Node.js: 4.5
- NPM: 5.4

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [ ] Smoothieware
- [x ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi (server)
- [x] Desktop or Laptop (client)
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2017-09-17:
Thank you for reporting this bug. The bug fix will be included in 1.9.10.

cheton, 2017-09-22:
Fixed in 1.9.10

closed 2017-09-22 by cheton