cncjs/cncjs #206: Grbl unit tracking

MitchBradley, 2017-09-18:
#### Description

The units Grbl uses for position reports depends on the $13 sewing, independent of the G20/21 parser state.  Cncjs currently only works correctly if $13=0.  One way to fix this would be to request a settings report upon connection to the controller to determine the initial $13 seeing, then track changes that might occur via manual writes through the serial connection. When$13 is changed manually, the saved position state should be adjusted, since a $13 write does not automatically trigger a new position report

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2017-09-18:
S/

MitchBradley, 2017-09-18:
Gag - autocompletion fail!. I meant setting not sewing above :-)

cheton, 2017-09-26:
Fixed in the commit (https://github.com/cncjs/cncjs/commit/6f11c9ab6840b557df95be226fc7b41232b8e038). It will be published with the 1.9.11 release.

I also changed the order for emitting "controller:settings" and "controller:state" events, this can prevent workarounds used in cncjs-pendant-tinyweb.

cheton, 2017-10-13:
https://github.com/cncjs/cncjs/releases/tag/v1.9.11

Fixed in 1.9.11

closed 2017-10-13 by cheton