cncjs/cncjs #208: Build problems on Cloud9

MitchBradley, 2017-09-20:
#### Description

I'm having problems building cncjs on Cloud9 (https://c9.io).

#### Background

Cloud9 is a development environment service that gives you a cloud-hosted Ubuntu instance in a Docker container.  (My development computer is getting overwhelmed with IDEs and toolchains so I'm trying a workflow of using a separate cloud instance for each project.)

#### What I did

I created a Cloud 9 workspace, from the Node.js template (in which a recent version node is preloaded in the instance), from my cncjs fork (github.com/MitchBradley/cncjs) which is unmodified from upstream. That has the effect of creating a Ubuntu Ubuntu 14.04.5 LTS instance with Node preinstalled, and "git clone"ing cncjs into ~/workspace . 

#### First Problem

 I then ran the following commands, following the instructions in CONTRIBUTING.md:

```
mitchbradley:~/workspace (master) $ npm --version
3.10.10
mitchbradley:~/workspace (master) $ git remote add upstream git://github.com/cncjs/cncjs.git
mitchbradley:~/workspace (master) $ git checkout -b mitch -t origin/master
Branch mitch set up to track remote branch master from origin.
Switched to a new branch 'mitch'
mitchbradley:~/workspace (mitch) $ npm install -g npm
```
[npm-install-output.txt](https://github.com/cncjs/cncjs/files/1319457/npm-install-output.txt)
```
mitchbradley:~/workspace (mitch) $ git config --global user.name "MitchBradley"
mitchbradley:~/workspace (mitch) $ git config --global user.email "wmb@firmworks.com"
mitchbradley:~/workspace (mitch) $ npm install
module.js:471
    throw err;
    ^

Error: Cannot find module 'process-nextick-args'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/ubuntu/.nvm/versions/node/v6.11.2/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_readable.js:26:23)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
```
#### Second Try

I then downgraded the Node version to 4 and tried again:

```
mitchbradley:~/workspace (mitch) $ nvm use 4
Now using node v4.8.4 (npm v2.15.11)
mitchbradley:~/workspace (mitch) $ npm install -g npm
/home/ubuntu/.nvm/versions/node/v4.8.4/bin/npx -> /home/ubuntu/.nvm/versions/node/v4.8.4/lib/node_modules/npm/bin/npx-cli.js
/home/ubuntu/.nvm/versions/node/v4.8.4/bin/npm -> /home/ubuntu/.nvm/versions/node/v4.8.4/lib/node_modules/npm/bin/npm-cli.js
npm@5.4.2 /home/ubuntu/.nvm/versions/node/v4.8.4/lib/node_modules/npm
mitchbradley:~/workspace (mitch) $ npm install >npm.install.log
```
[npm-install-log-stderr.txt](https://github.com/cncjs/cncjs/files/1319471/npm-install-log-stderr.txt)
[npm.install.log.txt](https://github.com/cncjs/cncjs/files/1319475/npm.install.log.txt)
[npm-debug.log.txt](https://github.com/cncjs/cncjs/files/1319474/npm-debug.log.txt)

```
mitchbradley:~/workspace (mitch) $ node --version
v4.8.4
mitchbradley:~/workspace (mitch) $ nvm ls
        v0.8.28
       v0.10.48
       v0.12.18
->       v4.8.4
        v6.11.2
         v8.4.0
         system
default -> 6 (-> v6.11.2)
node -> stable (-> v8.4.0) (default)
stable -> 8.4 (-> v8.4.0) (default)
iojs -> N/A (default)
```
#### How Should I Proceed?

At this point I am stuck.  I tried to go back to Node 6 but that failed:

```
mitchbradley:~/workspace (mitch) $ nvm use 6
module.js:471
    throw err;
    ^

Error: Cannot find module 'process-nextick-args'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/ubuntu/.nvm/versions/node/v6.11.2/lib/node_modules/npm/node_modules/readable-stream/lib/_stream_readable.js:26:23)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
nvm is not compatible with the npm config "prefix" option: currently set to ""
Run `npm config delete prefix` or `nvm use --delete-prefix v6.11.2` to unset it.
```

The "npm config delete prefix" suggestion did not help; the same error occurred afterward.

#### Versions

- CNCjs: 1.9.9
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2017-09-21:
Okay, I now have a working Cloud9 cncjs build in production mode.  Starting from a fresh Cloud9 workspace, created from the Node.js template and a github cncjs repo:

      npm install npm
      npm install -g process-nextick-args
      npm install
      npm run build

Add the following line to ~/.cncrc .  (To edit that file, which is in a directory above the Cloud9 workspace directory, you can run "c9 open ../.cncrc" from the bash window.  It's not possible to open it from the workspace directory browser on the left side of the IDE.)

    "allowRemoteAccess": true,

It's necessary because Cloud 9 runs its preview browser on a different network than the Node server.

Make this patch to package.json : 

```
diff --git a/package.json b/package.json
index 0ddc29db..c9d618a1 100644
--- a/package.json
+++ b/package.json
@@ -48,6 +48,6 @@
     "build:win-x64": "bash -c 'scripts/electron-builder.sh --win --x64'",
-    "start": "./bin/cnc",
+    "start": "./bin/cnc --port 8080",
     "start-electron": "electron ./dist/cnc/main",
     "watch-dev": "webpack --watch --config webpack.appconfig.development.js",
-    "start-dev": "NODE_ENV=development ./bin/cnc -vv -p 8000 -m /static:~+/static -m /widget:~+/widget",
+    "start-dev": "NODE_ENV=development ./bin/cnc -vv -p 8080 -m /static:~+/static -m /widget:~+/widget",
     "dev": "npm run build-dev && npm run start-dev",

```
It's necessary because Cloud 9's preview browser requires port 8080.

Finally, to get it going:

npm run start

Then choose the Cloud 9 menu item "Preview > Preview Running Application" to run a browser connected to the cncjs server.

To be supplied: instructions for running grbl-sim on Cloud 9.

MitchBradley, 2017-09-25:
Running grbl-sim on Cloud 9:

The instructions at https://cnc.js.org/docs/faq/#testing-without-arduino-board are mostly correct. The one problem is that the socat package is not installed by default on Cloud 9.  Here are the complete Cloud 9 instructions,  assuming that ~/workspace already contains the cncjs repository:

Open a fresh bash window with either the Window>New Terminal menu item or Alt-T.  In that window:

```
$ cd ~
$ sudo apt-get update
$ sudo apt-get install socat
$ git clone git@github.com:grbl/grbl.git
$ cd grbl/grbl
$ git clone git@github.com:grbl/grbl-sim.git
$ cd grbl-sim
$ make new
$ ~/workspace/examples/grbl-sim/simport.sh
```

You will need to follow instructions 5 and 6 from the faq link above to configure .cncrc for the fake serial port.

In some cases, I have had problems whereby cncjs connected to /dev/ttyFAKE but the jog buttons were grayed out.  That can often be fixed by clicking on the red Reset button and then moving the mouse to the serial console window and typing the Enter key a few times.

cheton, 2017-09-26:
The `grbl-sim` is for experimental use and quite unstable, it may occasionally hang or freeze when running a G-code program. Connecting to a Arduino board is recommended, but I'm not sure if it's possible to make a serial port proxy for Cloud9.

MitchBradley, 2017-10-19:
Closing issue as it has been mostly resolved and is not longer actionable.

closed 2017-10-19 by MitchBradley