cncjs/cncjs #209: CNCjs hangs with files bigger than 30bmb

dutchpatriot, 2017-09-23:
Hi,
CNCjs works perfect until I have big files with multiple passes. Above 30mb and than the webserver crashes. I've noticed this happening on my desktop computer and also on my raspberry pi 3. I thought it was a memory issue but since my desktop computer has the same issue I figured would it be possible to implement a option in CNCjs where you can tell the software to just repeat the process x times? This way the files can be smaller. The only thing one has to change on the code is remove G28 at the end and than change the (in my instance, Z-axis) G90 commands in G91.. This way one file with one pass can act as a full multipass file, this way you could have bigger cnc work done because the one file is smaller and there would be no need to restart after every pass...

Just an idea... Love your program! Thanks for the nice work
Dutchpatriot
 
#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2017-09-23:
The issue is related to https://github.com/cncjs/cncjs/issues/116. The CNCjs server can load a large file over 100MB or more, but the Chrome browser may run out of memory while rendering a G-code program which is larger than 20~30MB.

Multi-pass is a good solution for handling large files. I will add support for splitting the large file into small parts (e.g. 10MB chunk), so the UI will render only the active part and therefore consumes less memory. The drawback is that you cannot see a full 3D preview after splitting, but only the active one will be shown on the UI. Otherwise, implemeting a job queue might be another solution to it. User can append multiple files into the job queue and run tasks in sequence. When a task is completed, the next task in sequence will be automatically started.

I will try to evaluate and seek the best way of large file handling. Any suggestion is welcome for this issue.

dutchpatriot, 2017-09-23:
Indeed chrome is the problem. Firefox handles files with multiple millions of lines better than chrome does. Now running a 2 million+ file and no problem. 

> Multi-pass is a good solution for handling large files. I will add support for splitting the large file into small parts (e.g. 10MB chunk), so the UI will render only the active part and therefore [...]

Would be a nice solution since it's only a chrome problem I would suggest to make a muli-pass option with a splitting check box. The check box will than split the file, not checked will just load the file. A fly over description would be, when browser crashes check this to split file... This way chrome users will have the ability to split when it doesn't load, and other users will have the multi-pass option and still have the 3d rendered preview.

Your other solution would be nice to but when one queue's multiple jobs and one job would fail than grbl could crash with laser on and what not. This would be out ruled when every job would be simulated, and I think that would be like a gezillion more code to write and I think not worth the trouble, although a job queue would be supper cool too, if that is implemented I would create a material change option on my laser engrave so that after a job my material would be moved X amount so that without my intervening it would just keep on going...

Anyways I've just upgraded to 1.9.10 and love the new option to add your own widgets! 

I want to contribute whatever I can! I am Dutch so if you want I could translate the UI in to Dutch so you have another added language support...

Thanks again for your work and the reply
Dutchpatriot

cheton, 2017-09-24:
@dutchpatriot 

I pushed a commit with preliminary Dutch language support: https://github.com/cncjs/cncjs/commit/59509ce883e1d68e56122d76b62e5746cdfc81df

You can fork the repository and create the pull request to submit your changes. For details, see the contribution guide at https://github.com/cncjs/cncjs/blob/master/CONTRIBUTING.md#localization

The resource files: https://github.com/cncjs/cncjs/tree/master/src/web/i18n/nl

dutchpatriot, 2017-10-04:
Well the files are translated...

Tried to do a pull request or multiple and can't get it merged, my text editor changed some " in to  “ or ”, which was hard to see... I noticed the RED in github so that made me paste it in another editor and so I found the problem... This is my first time to do this forking and pulling on github so shoot me for like screwing up. I think I've got it right now but I have a whole list of commits now.