cncjs/cncjs #210: change/feature request move machine zero and move work zero

amx1, 2017-09-23:
Dear Cheton,

there are two things I would like to propose to change. The first one is quite simple. TinyG2 M100 gcode uses a comment ({..}) as argument. This will always be deleted before sending in sender.js.

The second one is a bit more a question of implementation strategy. I like the function to configure the machine home and work home gcode sequences. The actual implementation has two drawbacks. The config is stored on client side. You have to configure each client individually and reconfigure after a cache delete of the browser. The second one is that you have to hard code the sequence in additional modules like a pendant (I am working on  a custom pendant). My proposal is to implement these functions as commands of the controller. I tested this for TinyG  and introduced a "machine_home" command which then sends the same sequence for all modules. The configuration should be saved in .cncrc    

#### Description
Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

feature request: sender.js - TinyG2 M100 ({..}) commands, arguments will be removed
delete function: Remove anything inside the parentheses '()'
see also: https://github.com/synthetos/g2/wiki/Mcodes M100, M100.1, M101

feature request: 
1) instead of hard coding in gui, implement "move machine zero", "move work zero", ... as functions of controller
2) store custom commands for 1) in .cncrc and not on client side

#### Versions

- CNCjs: 1.9.10
- Node.js: 8.4.0
- NPM: 5.4.2

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [] Grbl
- [ ] Smoothieware
- [ x] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [ ] Windows
- [ ] Mac
- [ x] Linux

cheton, 2017-10-13:
https://github.com/cncjs/cncjs/releases/tag/v1.9.11

Fixed in 1.9.11

cheton, 2017-10-13:
Reopen this case since the second item (i.e. move machine zero & move work zero) was not implemented in 1.9.11

cheton, 2018-02-03:
Now you can configure custom MDI buttons with CNCjs 1.9.13.

#### Custom Commands

Go to **Axes Settings** to add custom commands and enable manual data input (MDI) support.

[image]

Toggle MDI mode

[image]

closed 2018-02-03 by cheton

amx1, 2018-02-06:
Dear Cheton 

perfect! Thank you very much.

Axel

justinclift, 2018-05-21:
Oh wow.  Hadn't realised custom commands are possible.  That's excellent.

Definitely something I'll be trying out too. :smile:

amx1, 2018-05-21:
Dear Justin,

 

I saw that you are working on g2core too. I checked out the latest g2core edge version, because of step polarity. My custom board needs an active low step signal and up to now,  I changed the stepper routine accordingly. The edge version runs fine and I don´t have to patch the stepper files anymore.

 

There is one bug maybe. I am not able to change the output of the status report. CNCJS sends during the initialization phase a long {“sr”:{“line”:t, …}} command, but this is ignored by g2core/edge. In my case, I did not compile reporting of posx, posy, posz as default option and {“sr”:{“posx”:t,…}} is not working. Therefor the work position is not shown in CNCJS and the tool is not moving. Do you have the same behaviour of g2core? 

 

Best

Axel

justinclift, 2018-05-21:
@amx1 Heh Heh Heh.  Already saw your issue report about it on the g2 repo.  Replied there a few minutes after you created it. :wink:

I haven't had a chance to look into it yet.  A major issue came up with a different project which I'm having to sort out now.