cncjs/cncjs #212: Smoothieware Tool Change

TobyGWilliams, 2017-09-27:
#### Description

We are trying to implement CNCjs with Smoothieware for our CNC machine upgrade project. All is working really well however we are drawing a blank at tool change time.

Smoothieware uses the 'stop streaming gCode' approach to pausing the machine, this enables jogging the machine around and probing commands by sending new commands. It won't stop on an M0 command. Unless I'm missing something, I can't see how to implement this philosophy in CNCjs. Can somebody point me in the right direction or could we make this a feature request?

If it's not already implemented, my idea of how this could work in CNCjs: The event system could be extended to include a tool change trigger. This could then allow the user to insert a set of commands to execute the tool change (e.g. spindle stop, coolant stop, move to tool change location, etc.) including a stop stream system command (I believe %wait lets the planner queue empty?)

#### Versions

- CNCjs: 1.9.9

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [x] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [x] Linux

cheton, 2017-09-27:
M0/M1 (Program Pause) and M2/M30 (Program End) are currently not handled by CNCjs, I will add two event triggers (i.e. `M0,M1 Program Pause` and `M2,M30 Program End`) and implement the expected behaviors (https://github.com/grbl/grbl/issues/681#issuecomment-101047305) in the 1.9 releases:

#### M0, M1 Program Pause
- Pause streaming data
- Trigger `M0, M1 Program Pause` and `G-code: Pause` events
- Click the <kbd>▶</kbd> button to resume

#### M2, M30 Program End
- Stop streaming data. Any g-code commands after the M2 or M30 will be ignored.
- Trigger `M2, M30 Program End` and `G-code: Stop` events

TobyGWilliams, 2017-09-27:
Adding a M6/tool change trigger would be cool as well.

cheton, 2017-09-28:
Sure. The support for M6 tool change should be the first priority.

cheton, 2017-10-01:
The preliminary support for pausing G-code execution with M0, M1, M2, M6, and M30 commands is now available in the master branch. I will make a thorough test in the next few days.

TobyGWilliams, 2017-10-03:
Sadly, I've not been able to install directly from github to do testing. Unless I think of something, I'll have to wait for it to be pushed out via npm/windows installer.

T

TobyGWilliams, 2017-10-03:
Just did a test, works great on smoothie.

cheton, 2017-10-07:
#### Remaining Todo Items
* [ ] Restore work position after completion
   (1) Move to Z-safe height defined by user
   (2) Move to the previous position along the X and Y axis
   (3) Move to the previous position along the Z axis

* [ ] Restore modal state after completion

    Modal Group | Member Words
    :----- | :-----
    Coordinate System Select | G54, G55, G56, G57, G58, G59
    Plane Select | G17, G18, G19
    Distance Mode | G90, G91
    Feed Rate Mode | G93, G94
    Units Mode | G20, G21
    Spindle State | M3, M4, M5
    Coolant State | M7, M8, M9

* [ ] Update M6 Tool Change instructions on wiki

cheton, 2017-10-13:
https://github.com/cncjs/cncjs/releases/tag/v1.9.11

CNCjs 1.9.11 is now available with M6 tool change support. You can upgrade to 1.9.11 and read the wiki page at https://github.com/cncjs/cncjs/wiki/Tool-Change to learn how to add a tool change macro.

closed 2017-10-13 by cheton