cncjs/cncjs #215: CNC and fusion

nkp1215, 2017-10-10:
#### Description

Hey so I'm new to CNC and GCode and Everything else. So my work place built and openbuilds CNC. And we are using your software on a raspberry pi. I'm trying to get a file I made on Fusion 360 to run on my CNC. Every time I try to run the G-code it either says soft limit reached or error code. Which post processor should I use to help me. I just need help

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

AustinSaintAubin, 2017-10-10:
I think you have your GRBL controller mis-configured.
I might be about to work with you to get this configured.
Upload a video of the behavior and run `$$` and post the output.

PM me of you would like to arrange a 1-1. (my time is limited, but I might be able to spare some time)

You might also try setup with: http://x-carve-instructions.inventables.com/1000mm/step10/
Its compatible with most GRBL controllers.

cheton, 2017-12-05:
@nkp1215 

You can also raise your question at https://github.com/gnea/grbl/issues if you still cannot resolve the soft limit issue.

closed 2017-12-05 by cheton

nkp1215, 2018-01-16:
@AustinSaintAubin 

I'm sorry I've been out of the office and I never seen you response. I hope your still open to helping me as I'm still having trouble.
This is what '$$" gives me
$0=10 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=3 (Step direction invert, mask)
$4=1 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=3 (Status report options, mask)
$11=0.020 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=1 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=1 (Homing direction invert, mask)
$24=25.000 (Homing locate feed rate, mm/min)
$25=500.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=3.000 (Homing switch pull-off distance, millimeters)
$30=128368 (Maximum spindle speed, RPM)
$31=0 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=26.667 (X-axis travel resolution, step/mm)
$101=26.667 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$110=5000.000 (X-axis maximum rate, mm/min)
$111=5000.000 (Y-axis maximum rate, mm/min)
$112=1000.000 (Z-axis maximum rate, mm/min)
$120=150.000 (X-axis acceleration, mm/sec^2)
$121=150.000 (Y-axis acceleration, mm/sec^2)
$122=100.000 (Z-axis acceleration, mm/sec^2)
$130=780.000 (X-axis maximum travel, millimeters)
$131=730.000 (Y-axis maximum travel, millimeters)
$132=85.000 (Z-axis maximum travel, millimeters)
ok

This is a screenshot of my page currently.
[image]

And the video wouldn't upload but literally nothing happens. The toolhead turns on for 2 seconds and shuts off.

MitchBradley, 2018-01-16:
I think the problem is in the Z axis range.  Your GRBL setup has the valid Z range at -85 to 0.  ($23=1, $132=85).  The GCode Z range is -10 to +16 in work coordinates.  Your work coordinate system has Work Z0 at Machine Z-10, so the GCode Z range is -20 to +6 in machine coordinates.  +6 is outside the range -85 to 0.

That GCode program wants 16 mm of headroom above the Z=0 point in the program, and your machine setup only has 10 mm of headroom.  Without knowing the details of your part, stock, machine, and fixturing, I can't tell you exactly how to fix it, but the bottom line is that you have to do something to prevent the Z coordinate from going above the top of the machine.