cncjs/cncjs #217: Upgrade to Serialport 6x

reconbot, 2017-10-13:
#### Description

Serialport 6x has been released. It's got some vast improvements since version 4. That includes memory leak fixes, and major performance improvements.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [x] Mac
- [x] Linux

cheton, 2017-10-13:
Thank you for the information. I will upgrade to serialport@6 in the next release.

whimsycwd, 2017-10-17:
I come across a similar issue describe in 
https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/587

When I use cncjs in Win10, connecting to serialport, The app crash when I unplug the cable every time.

https://github.com/whimsycwd/snapjs/commit/699e5faabab21aca492d42d723b5be3e01304aa4
Upgrade serialport to 6.0.0, this issue fixed.  haven't found out other problem so far.

cheton, 2017-12-07:
Upgraded to serialport 6.x in [1.9.12](https://github.com/cncjs/cncjs/releases/tag/v1.9.12)

closed 2017-12-07 by cheton