cncjs/cncjs #218: WebGL not supported in some video card.

whimsycwd, 2017-10-17:
#### Description

Describe the issue or feature request in detail. Take a screenshot or record a video will be helpful.

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core
- [x] Marlin

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

In a rare case, some test computer(2/20 maybe),   prompt that it doesn't support WebGL.  **These two computers are modern ones, purchased in one year". I guess the root cause is the video card.
However when I use Chrome Browser on the same computer to access application instead of using electron chromium. It did render. 

After doing some research, chromium has a gpu blacklist. 
*  https://groups.google.com/a/chromium.org/forum/#!topic/chromium-dev/AQNCH7plJe4
*  https://www.khronos.org/webgl/wiki/BlacklistsAndWhitelists#Chrome_on_Windows

```
app.commandLine.appendSwitch('ignore-gpu-blacklist', 'true');
```
Adding above flag can workaround the issue, but I do not fully understand the mechanism behind the scene. And the risk of opening the flag.

vilo, 2018-09-26:
Sorry, where does the line:
app.commandLine.appendSwitch('ignore-gpu-blacklist', 'true');
supposed to go?

I am running standalone cncjs on mac. WebGL is working in both Chrome and Safari browsers, but when I open the cncjs app I get the graphic card error.

whimsycwd, 2018-10-06:
https://github.com/cncjs/cncjs/blob/d4fd37e5f19f744d7fe60c0772dac0b0a4e32523/src/main.js#L59
Add that line here. It works for me. @vilo