cncjs/cncjs #219: Announcing cncjs-shopfloor-tablet - plus a question

MitchBradley, 2017-10-17:
I have created a cncjs UI variant optimized for shop floor use on a tablet computer.  

https://github.com/MitchBradley/cncjs-shopfloor-tablet

It is sort of a "super pendant" (derived from cncjs-pendant-tinyweb) that adds GCode loading/display, better control of work coordinates, and MDI.  The button layout reduces operator errors.  I am currently using it to control my small production mill.

The question:  In order to list the cncjs server's watch directory files, I had to move the 'api/watch/files' API to the "public access" section.  There must be a better way, but I don't understand the authorization logic in the cncjs web app well enough to duplicate it in the simpler "pendant" environment.  Any suggestions, pointers, or code snippets would be much appreciated.

cheton, 2017-10-18:
Hi @MitchBradley 

Here is the mechanism that checks for access token, it will try to get the token using query parameter or from a json data in the request body:
https://github.com/cncjs/cncjs/blob/master/src/app/app.js#L225
```js
// Check if the provided credentials are correct
const token = get(req, 'query.token') || get(req, 'body.token');
bypass = bypass || (token && verifyToken(token));
```

To get API access, you can use `GET /api/watch/files?token=xxxxxxxx` or `POST /api/watch/files` with a json data `{ "token": "......" }` in the request body.

You can find the access token at https://github.com/MitchBradley/cncjs-shopfloor-tablet/blob/master/src/controller.js#L18-L35.

cheton, 2017-10-18:
Would it help if I add a "watchdir:list" command on the server for convenience?

For example:
```js
// List the files in the root of the watch directory
controller.command('watchdir:list', function(files) {
});

// List the files in a specified path of the watch directory
controller.command('watchdir:list', path, function(files) {
});
```

MitchBradley, 2017-10-18:
Thanks, I will try the token thing.  Regarding the proposed watchdir:list command, how would it return the results?  Some new event?

The API method has the nice property that the requester and the response handler are located together in the same JQuery.get() invocation.  I think that a watchdir:list formulation would result in separated request and response code, potentially less clear than keeping them together.

MitchBradley, 2017-10-18:
Oh, sorry, I misunderstood.  I see now that "function(files) { ... }" in controller.command(...) is the response handler.

That would indeed be helpful, but perhaps not hugely different than the API formulation.

MitchBradley, 2017-10-18:
Adding the token to the request solved the problem.

`jQuery.get(""../api/watch/files"", function(data) {`
became
`jQuery.get(""../api/watch/files"", {token: cnc.token}, function(data) {`

There are some advantages to the proposed watchdir:list command:
1. It uses the websocket connection so no need to create a new connection (minor advantage since it is infrequently used)
2. An event that reports the file list could be triggered any time a file changes in the watch directory, thus  eliminating the need for the **Rfrsh** button in my tablet UI.  This is a Would Be Nice thing, but not a big problem.

closed 2017-10-18 by MitchBradley

cheton, 2017-10-22:
Hi @MitchBradley,

I've invited you to the cncjs organization. If you don't mind to make your repo public here, you can go to GitHub's <b>Settings > Danger Zone</b> to transfer your repository to this organization (i.e. https://github.com/cncjs/cncjs-shopfloor-tablet). That will get more people to see your work.

Thank you for your contribution.

MitchBradley, 2017-10-23:
> your repo public here, you can go to GitHub's *Settings > Danger
> Zone*to transfer your repository to this organization (i.e.

And voila, cncjs-shopfloor-tablet is now under cncjs.

cheton, 2017-11-30:
Hi @MitchBradley 

I have setup Travis CI for your cncjs-shopfloor-tablet repo. Each time you finish the changes, you can increase the version in package.json, and create a new release tag that matches the version.

With each release, the CI build process will create a ZIP file, and upload the file to GitHub release. 
https://travis-ci.org/cncjs/cncjs-shopfloor-tablet