cncjs/cncjs #220: Jog Controls deactive/greyed out

HendrikRoth, 2017-10-18:
#### Description

The jog controls are greyed out for me. I can connect to my grbl board (protoneer) and i can move via manuell gcode commands. But the jog controls widget is still greyed out. Any ideas? Thanks.
[image]

#### Versions

- CNCjs: 1.9.4
- Node.js: 6.9.4
- NPM: 3.10.10

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2017-10-18:
I don't know what the underlying problem is, but from the point of view of the jog widget, the controls are probably grayed out because the Grbl device has not responded with a signon message.  Here is what it is supposed to look like immediately after connection to Grbl.

[image]

When you tested moving with manual gcode commands, were you doing that from another terminal program like screen or minicom, or were you issuing the commands from the cncjs console widget?

If the serial port is open by two different applications at the same time, characters sent from that port will often go to only one of the applications.  I suppose it is possible that the characters from /dev/ttyAMA0 were being "stolen" by another program.  Usually, when cncjs tries to open a port that another app is already using, the open will fail, but it is certainly possible to double-open a port under some circumstances.  For example, if cncjs has the port already open, you can open it again with
```
screen /dev/ttyAMA0 115200
```
and then screen will "steal" all the characters.

Another thing to try is clicking on the red Reset button at the upper right.  That should result in a Grbl signon message like
```
Grbl 1.1e ['$' for help]
```
I suppose it is also possible that your Pi serial port is not setup correctly.  See
https://forum.protoneer.co.nz/viewtopic.php?t=143
especially the part about "RPi2 Serial Config"

MitchBradley, 2017-10-18:
I meant "RPi3 Serial Config", not RPI2...

HendrikRoth, 2017-10-19:
The Reset Button did the try, i am so sorry!

closed 2017-10-19 by HendrikRoth

cheton, 2017-10-20:
@HendrikRoth This issue might be similar to #188, @edalger replaced his laptop with a new one to drive xPro v3, and this issue seems resolved.

I'm not sure whether your grbl board (protoneer) have an Arduino boot loader, the bootloader will trigger a reset to Grbl upon a connection. CNCjs relies on the Grbl welcome message to determine if Grbl has been successfully initialized. So far, if you cannot find the root cause, you can click the reset button as a workaround solution.

cheton, 2017-10-24:
Hi @HendrikRoth 

Just to let you know you should upgrade CNCjs 1.9.4 to 1.9.11. CNCjs 1.9.3 to 1.9.7 have a critical issue that may consume all available receive buffer on Grbl especially when running a G-code program with slow feed rate. See the release notes at https://github.com/cncjs/cncjs/releases/tag/v1.9.8