cncjs/cncjs #222: Error: Generated checksum for "electron-v1.7.9-linux-arm.zip" did not match expected checksum.

erpalma, 2017-10-21:
#### Description

When cross compiling for arm (arm-linux-gnueabihf) I get:

/root/tmp/cncjs/node_modules/electron/install.js:48
  throw err
  ^

Error: Generated checksum for "electron-v1.7.9-linux-arm.zip" did not match expected checksum.
    at ChecksumMismatchError.Error (native)
    at ChecksumMismatchError.ErrorWithFilename (/root/tmp/cncjs/node_modules/electron/node_modules/sumchecker/build.js:41:124)
    at new ChecksumMismatchError (/root/tmp/cncjs/node_modules/electron/node_modules/sumchecker/build.js:56:133)
    at Hash.<anonymous> (/root/tmp/cncjs/node_modules/electron/node_modules/sumchecker/build.js:203:22)
    at emitNone (events.js:86:13)
    at Hash.emit (events.js:185:7)
    at emitReadable_ (_stream_readable.js:432:10)
    at emitReadable (_stream_readable.js:426:7)
    at readableAddChunk (_stream_readable.js:187:13)
    at Hash.Readable.push (_stream_readable.js:134:10)

#### Versions

- CNCjs: master
- Node.js: 6.11
- NPM: 5.5.1

MitchBradley, 2017-10-21:
I'm rather new to the Node.js/npm world, but this post seems relevant:

https://github.com/electron-userland/electron-prebuilt/issues/264

Based on that, there may a version problem with the "sumchecker" package.  You could try:
```
npm list sumchecker
```
to find out which version you currently have, then
```
npm install sumchecker@2.02
```
and then re-cross-compile.  Or maybe
```
npm install sumchecker@2.01
```
Depending on how you are doing the build, you might need to add the '-g' flag to the npm commands above.

erpalma, 2017-10-23:
Nice, it worked!

Thanks

closed 2017-10-24 by cheton

gosongs, 2017-11-15:
@MitchBradley 

it is `npm install sumchecker@2.0.2`