cncjs/cncjs #223: cncjs won't run on new Raspberry Pi 3 Model B

ENGR-mike, 2017-10-21:
I've successfully run cncjs on a Windows PC and loved it.  I then proceeded to install it on a new Raspberry Pi and have had no success getting it to run.  Being that I'm just a gearhead, nuts&bolts mechanical engineer just wanting to do CNC at home, the command line installation was intimidating.  With that being said, the following are specifics about my installation:
- CNCjs: 1.9.11
- Node.js: 8.7.0
- NPM: 5.4.2
I installed it via NPM.  I am using an xPRO v3 controller, but that is irrelevant at this point since I cannot get cncjs to run on the pi.  My pi is a brand new Raspberry Pi 3 model B running Raspbian (i.e Debian) Stretch (I think Stretch is version 9).

I followed the Raspberry Pi Setup Guide explicitly, recognizing that the lines starting with # were comments.  At the end of the setup where it says:
"Reboot to test
sudo reboot
You're Done, STOP HERE!!!"
I was baffled, because there were no further instructions on how to even start cncjs.  I went back to the NPM website and noticed it had a line saying:
"Run cnc to start the server, and visit http://yourhostname:8000/ to view the web interface. Pass --help to cnc for more options."

So, I typed cnc at a command prompt and the first time I received an EEADRINUSE error.  I was able to contact Cheton directly and he suggested changing to port 8888.  I get it, but there were a number of steps in the Raspberry Pi Setup Guide dealing with setting up and redirecting to port 8000.  In any event, I typed "cnc -p 8888" and no longer received an EEADRINUSE error.  It appeared to be proceeding but stopped after the line "Starting the server at http://127.0.1.1:8888" and never did anything else.

Hey guys, I may not be a programmer, coder or have any working knowledge of linux, but I've never been afraid of trying something new like this.  However, I have to say I am very frustrated with making no progress.  Please help!

Thanks,
Michael Dwyer

ENGR-mike, 2017-10-21:
By some strange luck, I think I figured it out. Although I installed everything from a CLI, I switched to the GUI, typed http://:8888/ in chromium and was able to start CNCjs and make a connection to my controller board. I shut it down and returned to the CLI to stop PM2 and reset it to start on port 8888 (port 8000 was not working for me). I also reset the iptables to redirect port 80 to port 8888. I rebooted my pi and went directly to chromium and typed http://:8888/ again. I was able to start CNCjs and open a connection to my controller without any problems.

closed 2017-10-21 by ENGR-mike

MitchBradley, 2017-10-21:
I'm glad you figured it out because I was in the middle of typing in a very long-winded explanation of how cncjs is split into two parts - a server and a browser-based UI - digressing into deep explanations of web technology.

Note that, if your Pi is connected to a network, it is possible to run the cncjs UI on other machines on that network, so you could monitor jobs from, say, a smartphone on the same WiFi as the Pi.  You can have multiple UI's running at the same time; that works surprisingly well.

I often set up jobs from the PC in my office and then run/control them from a $100 Android tablet in the shop. The tablet makes a nice control surface for the shop; cheaper and easier to mount than a monitor + keyboard, and no issues with dust and chips getting into the keyboard.

ENGR-mike, 2017-10-21:
Sounds like you have a nice setup. If you don’t mind, I might contact you in the future about doing something similar.

Thanks,
Michael

From: Mitch Bradley
Sent: Saturday, October 21, 2017 12:31 PM
To: cncjs/cncjs
Cc: ENGR-mike; State change
Subject: Re: [cncjs/cncjs] cncjs won't run on new Raspberry Pi 3 Model B(#223)

I'm glad you figured it out because I was in the middle of typing in a very long-winded explanation of how cncjs is split into two parts - a server and a browser-based UI - digressing into deep explanations of web technology.
Note that, if your Pi is connected to a network, it is possible to run the cncjs UI on other machines on that network, so you could monitor jobs from, say, a smartphone on the same WiFi as the Pi. You can have multiple UI's running at the same time; that works surprisingly well.
I often set up jobs from the PC in my office and then run/control them from a $100 Android tablet in the shop. The tablet makes a nice control surface for the shop; cheaper and easier to mount than a monitor + keyboard, and no issues with dust and chips getting into the keyboard.

ENGR-mike, 2017-10-21:
Mitch,

Everything seems to be working fine except for the visualizer widget.  I can see code, but if I turn on 3D view, I cannot see a tool path.  Is there a setting I should change on the Raspberry Pi?

Thanks,

From: Mitch Bradley
Sent: Saturday, October 21, 2017 12:31 PM
To: cncjs/cncjs
Cc: ENGR-mike; State change
Subject: Re: [cncjs/cncjs] cncjs won't run on new Raspberry Pi 3 Model B(#223)

I'm glad you figured it out because I was in the middle of typing in a very long-winded explanation of how cncjs is split into two parts - a server and a browser-based UI - digressing into deep explanations of web technology.
Note that, if your Pi is connected to a network, it is possible to run the cncjs UI on other machines on that network, so you could monitor jobs from, say, a smartphone on the same WiFi as the Pi. You can have multiple UI's running at the same time; that works surprisingly well.
I often set up jobs from the PC in my office and then run/control them from a $100 Android tablet in the shop. The tablet makes a nice control surface for the shop; cheaper and easier to mount than a monitor + keyboard, and no issues with dust and chips getting into the keyboard.

MitchBradley, 2017-10-21:
I would be happy to answer questions about my setup.  You can see my DIY
controller here:

https://plus.google.com/108078671773315527328/posts/eoQT6tYGpkx

Regarding the visualizer problem, I suspect that WebGL is not enabled in
your browser.  See, for example,

https://raspberrypi.stackexchange.com/questions/49690/chromium-50-webgl-support-on-jessie

and search for "raspberry pi 3 webgl" for other options.

This looks particularly interesting:

https://forums.resin.io/t/fullscreen-webkit-browser-with-hw-accelerated-css-webgl-and-html5-video-on-the-raspberrypi-3/741

I haven't tried it so I don't know what sorts of problems might arise,
but I really like the idea of running cncjs fullscreen.  When you are
controlling a machine, any extraneous control areas, such as browser
tabs and menus, can result in expensive machine crashes.  That's why I
developed and use a very simple control screen for production in my shop
- see https://github.com/MitchBradley/cncjs-shopfloor-tablet .

ENGR-mike, 2017-10-22:
Actually, webGL is enabled, so I’m at a loss.

From: Mitch Bradley
Sent: Saturday, October 21, 2017 6:00 PM
To: cncjs/cncjs
Cc: ENGR-mike; State change
Subject: Re: [cncjs/cncjs] cncjs won't run on new Raspberry Pi 3 Model B(#223)

I would be happy to answer questions about my setup.  You can see my DIY
controller here:

https://plus.google.com/108078671773315527328/posts/eoQT6tYGpkx

Regarding the visualizer problem, I suspect that WebGL is not enabled in
your browser.  See, for example,

https://raspberrypi.stackexchange.com/questions/49690/chromium-50-webgl-support-on-jessie

and search for "raspberry pi 3 webgl" for other options.

This looks particularly interesting:

https://forums.resin.io/t/fullscreen-webkit-browser-with-hw-accelerated-css-webgl-and-html5-video-on-the-raspberrypi-3/741

I haven't tried it so I don't know what sorts of problems might arise,
but I really like the idea of running cncjs fullscreen.  When you are
controlling a machine, any extraneous control areas, such as browser
tabs and menus, can result in expensive machine crashes.  That's why I
developed and use a very simple control screen for production in my shop
- see https://github.com/MitchBradley/cncjs-shopfloor-tablet .

MitchBradley, 2017-10-22:
I suppose you could try a different browser or a different GCode program or connect to cncjs from another computer to see if any of those change the symptoms.

cheton, 2017-10-22:
Hi @ENGR-mike 

Could you do the following steps and attach your screenshots?
1. Turn on debug logging on your browser with an additional query parameter (`?log_level=debug`) follows the url:
    e.g. http://xxx.xxx.xxx.xxx:8888/?log_level=debug
2. Open <b>Developer Tools</b> and go to <b>Console</b>, then check if there are any errors reported.
3. Take screenshots for the browser's window and the output in the Console.

ENGR-mike, 2017-10-24:
I’ll try to do this tomorrow evening.  It’s hard to accomplish anything during the work week.

Thanks,
Michael

From: Cheton Wu
Sent: Saturday, October 21, 2017 8:08 PM
To: cncjs/cncjs
Cc: ENGR-mike; Mention
Subject: Re: [cncjs/cncjs] cncjs won't run on new Raspberry Pi 3 Model B(#223)

Hi @ENGR-mike
Could you do the following steps and attach your screenshots?
1. Turn on debug logging on your browser with an additional query parameter (?log_level=debug) follows the url:
e.g. http://xxx.xxx.xxx.xxx:8888/?log_level=debug
2. Open Developer Tools and go to Console, then check if there are any errors reported.
3. Take screenshots for the browser's window and the output in the Console

ENGR-mike, 2017-10-25:
Cheton,

See below as you requested regarding my visualizer  not working in Chromium on my Raspberry Pi 3.

From: Michael Dwyer
Sent: Monday, October 23, 2017 9:54 PM
To: cncjs/cncjs
Subject: RE: [cncjs/cncjs] cncjs won't run on new Raspberry Pi 3 Model B(#223)

I’ll try to do this tomorrow evening.  It’s hard to accomplish anything during the work week.

Thanks,
Michael

From: Cheton Wu
Sent: Saturday, October 21, 2017 8:08 PM
To: cncjs/cncjs
Cc: ENGR-mike; Mention
Subject: Re: [cncjs/cncjs] cncjs won't run on new Raspberry Pi 3 Model B(#223)

Hi @ENGR-mike
Could you do the following steps and attach your screenshots?
1. Turn on debug logging on your browser with an additional query parameter (?log_level=debug) follows the url:
e.g. http://xxx.xxx.xxx.xxx:8888/?log_level=debug
2. Open Developer Tools and go to Console, then check if there are any errors reported.
3. Take screenshots for the browser's window and the output in the Console

ENGR-mike, 2017-10-26:
[image]

cheton, 2017-10-27:
@ENGR-mike 

According to your description, you can run the loaded G-code program, but you cannot see the toolpath in 3D view with a Chromium browser running on a Raspberry Pi 3.

What's your result when connecting to CNCjs with a remote desktop computer or laptop? Have you ever tried installing different browsers like Google Chrome or Firefox on Raspberry Pi 3?

> Everything seems to be working fine except for the visualizer widget.  I can see code, but if I turn on 3D view, I cannot see a tool path.  Is there a setting I should change on the Raspberry Pi?

ENGR-mike, 2017-10-28:
I also cannot see the visualizer in 3D view via remote connection from my PC.  Chromium is Chrome, but for the Raspberry Pi.  I will have to install Firefox to try it.

Any other solutions.

Thanks,
Michael

From: Cheton Wu
Sent: Friday, October 27, 2017 8:42 AM
To: cncjs/cncjs
Cc: ENGR-mike; Mention
Subject: Re: [cncjs/cncjs] cncjs won't run on new Raspberry Pi 3 Model B(#223)

@ENGR-mike
According to your description, you can run the loaded G-code program, but you cannot see the toolpath in 3D view with a Chromium browser running on a Raspberry Pi 3.
What's your result when connecting to CNCjs with a remote desktop computer or laptop? Have you ever tried installing different browsers like Google Chrome or Firefox on Raspberry Pi 3?
Everything seems to be working fine except for the visualizer widget. I can see code, but if I turn on 3D view, I cannot see a tool path. Is there a setting I should change on the Raspberry Pi?

ENGR-mike, 2017-11-06:
I installed Firefox and I can now see the 3D visualizer.

Sent from Michael's iPhone

ENGR-mike, 2017-11-23:
I was unable to work with my mill and CNCjs for several weeks, but I’m now back at it.

CNCjs works in Firefox on the Raspberry Pi and I can see a toolpath, but it does not seem to run reliably in Firefox.  When starting in Firefox, I don’t always get a connection to the port.  It sometimes takes closing and reopening Firefox several times, or rebooting the Raspberry Pi to get a connection.  I’m also experiencing some communications lag in Firefox where it can take a second or two before jogging results in movement of my stepper motors.

I do not have connection problems with Chrome, nor do I have any communications lags in Chrome.  However, I still cannot see a toolpath in Chrome.  I wish you could figure out how to get the toolpath to be visible using Chrome on the Raspberry Pi.  I’m going to continue using Chrome and ignore that I cannot see a toolpath.

Thanks,
Michael Dwyer

From: Michael & Denise
Sent: Monday, November 6, 2017 3:09 PM
To: Michael Dwyer
Cc: cncjs/cncjs
Subject: Re: [cncjs/cncjs] cncjs won't run on new Raspberry Pi 3 Model B(#223)

I installed Firefox and I can now see the 3D visualizer.

Sent from Michael's iPhone

On Oct 28, 2017, at 4:47 PM, Michael Dwyer <mikeanddenise521@gmail.com> wrote:
I also cannot see the visualizer in 3D view via remote connection from my PC.  Chromium is Chrome, but for the Raspberry Pi.  I will have to install Firefox to try it.
 
Any other solutions.
 
Thanks,
Michael
 
 
From: Cheton Wu
Sent: Friday, October 27, 2017 8:42 AM
To: cncjs/cncjs
Cc: ENGR-mike; Mention
Subject: Re: [cncjs/cncjs] cncjs won't run on new Raspberry Pi 3 Model B(#223)
 
@ENGR-mike
According to your description, you can run the loaded G-code program, but you cannot see the toolpath in 3D view with a Chromium browser running on a Raspberry Pi 3.
What's your result when connecting to CNCjs with a remote desktop computer or laptop? Have you ever tried installing different browsers like Google Chrome or Firefox on Raspberry Pi 3?
Everything seems to be working fine except for the visualizer widget. I can see code, but if I turn on 3D view, I cannot see a tool path. Is there a setting I should change on the Raspberry Pi?

MitchBradley, 2017-11-23:
I just did a fresh install of CNCjs on a Pi 2.  When I used Chromium to access CNCjs, the 3DViewer window was zoomed out so far that the axes and their numbers were very small - barely visible.  At that zoom factor, I imaging that some tool paths would be essentially invisible.  Have you tried using the controls at the bottom right of the 3d viewer to zoom in?

Just a thought.

MitchBradley, 2017-11-24:
On my current install on a Pi 2, Chromium displays toolpaths in the 3d viewer; see the attached screenshot and the test GCode file.
[image]

[TestSquares.txt](https://github.com/cncjs/cncjs/files/1500455/TestSquares.txt)

Starting with a Raspbian wheezy image, I dist-upgraded to Raspbian jessie, then followed the Raspberry Pi setup instructions on cncjs's wiki to install node v8, npm, and cncjs.

I don't have a Pi 3 to test.

I can't say I'm particularly surprised that you are having problems.  Linux in general has been in dependency hell essentially forever.  It is even worse on Raspberry Pi due to its resource limitations.

MitchBradley, 2017-11-24:
Which operating system version are you using?  Raspbian wheezy, Raspbian jessie, Raspbian stretch, or something else?

It turns out that the versions of Chromium in wheezy and jessie are fairly old and do not support a Javascript feature called "arrow functions".  Various cncjs modules use arrow functions so things sort of randomly break.  I am currently upgrading to Raspbian stretch which should be somewhat better - but it will no doubt break some other things.

MitchBradley, 2017-11-24:
Okay, it is possible that I duplicated your problem.  After upgrading my pi2 to Raspbian stretch, uninstalling the old "chromium" (v22) package, installing the new "chromium-browser" (v60) package, and enabling the OpenGL driver with raspi-config , I saw the same problem as you report.  I.e. the 3dviewer area was blank - no axes, no toolpath when GCode was loaded.

At the time, the "Memory Split" parameter in raspi-config was set to 64, i.e. only 64MB of memory available to the GPU.  When I set that to 256 and rebooted, 3dviewer worked!

I eventually settled on the "Fake KMS" version of the OpenGL driver, as the system seems to be a bit faster than with the real-KMS version.

So, now that all the system software is up to date, everything seems to be working.  It is much too sluggish to be pleasant to use, but it works.  Some operations require a great deal of patience; for example it can take many seconds just to establish a serial connection and get the output into the Console window.

I'm going to stick with my cncjs-shopfloor-tablet workspace because it runs fast and is easy to control, but at least the full cncjs is functional.

ENGR-mike, 2017-11-24:
I'm also using Raspbian Stretch.  I increased the memory allocated to the GPU to 256mb and rebooted, but it did not solve the problem.  Keep in mind I'm using a Raspberry Pi 3 if that makes any difference.

Sent from Michael's iPhone

MitchBradley, 2017-11-25:
See https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Adventures-in-PiLand for a long story.

I don't have a Pi 3 so I cannot exactly reproduce your problem.  You don't mention which Chromium version you have.  I am using v60.

At this point I have pretty much given up on running the main CNCjs UI on a Pi.  I am going to refocus my efforts on the cncjs-shopfloor-tablet work.  In that model, I run the cncjs server program on a headless Pi ZeroW, running the cncjs-shopfloor-tablet UI on a $100 Android tablet over WiFi.  That gives me a stable, responsive UI with which I can control my milling machine safely.  There are no keyboards or mice in the shop area to get all dusty and oily.

ENGR-mike, 2017-11-25:
Mitch / Cheton,

Thanks for all of your help.  After making the purchase of the Raspberry Pi, I am sort of committed to using it.  Not having use of the 3D toolpath visualizer should not affect my work.  Although I am a licensed Professional Engineer in mechanical engineering, I tend to be a very simple minded, nuts and bolts kind of guy (i.e a gearhead).  I am reasonably good at troubleshooting problems with Windows based machines using the trial and error approach.  I am not a programmer, coder or hacker by any stretch of the imagination.  So, could please help me understand some things:

1. If the Pi is running headless without a monitor connected to it, are the computing resources dedicated to all other processes, etc. other than graphics?
2. Assuming the answer to the above question is yes, would interfacing with the Pi and CNCjs via VNC viewer also allow the computing resources of the Pi to be dedicated to everything but the graphics?
3. If the answer to question 1 is no, does it mean the gpu functions have to be intentionally disabled?  If so, how?

I want to understand what makes the headless option better and give the Pi a fair chance before I use it as a rifle target.  If the Pi and CNCjs become too frustrating or troublesome, I may revert back to a Windows laptop and something like GRBL Panel for interfacing with my CNC mill controller.  I’d rather focus my efforts on designing and machining parts than troubleshooting software and communications issues.  I want to make chips, not play around with electrons.

Mitch, if I feel like expending the energy on your cncjs-shopfloor-tablet idea, I may contact you with more questions.

Thanks,
Michael

From: Mitch Bradley
Sent: Friday, November 24, 2017 6:23 PM
To: cncjs/cncjs
Cc: ENGR-mike; Mention
Subject: Re: [cncjs/cncjs] cncjs won't run on new Raspberry Pi 3 Model B(#223)

See https://github.com/cncjs/cncjs/wiki/Setup-Guide:-Raspberry-Pi-%7C-Adventures-in-PiLand for a long story.
I don't have a Pi 3 so I cannot exactly reproduce your problem. You don't mention which Chromium version you have. I am using v60.
At this point I have pretty much given up on running the main CNCjs UI on a Pi. I am going to refocus my efforts on the work. In that model, I run the cncjs server program on a headless Pi ZeroW, running the cncjs-shopfloor-tablet UI on a $100 Android tablet over WiFi. That gives me a stable, responsive UI with which I can control my milling machine safely. There are no keyboards or mice in the shop area to get all dusty and oily.

cheton, 2017-11-25:
You can run CNCjs server on your Raspberry Pi, and operate [cncjs-shopfloor-tablet](https://github.com/cncjs/cncjs-shopfloor-tablet#usage) on the connected monitor for simple operation. If you also want to use advanced UI features like 3D view, you can open another browser on your Windows laptop, and connect to the CNCjs server over WiFi or an ethernet cable. I assume your Windows laptop should have more powerful GPU, that will be more suitable for rendering 3D visualization.

In summary, the CNCjs server can serve multiple connected clients over the same serial connection, so you can open cncjs-shopfloor-tablet on a connected monitor, and open the main UI on another computer that has more powerful GPU resource.

> 1. If the Pi is running headless without a monitor connected to it, are the computing resources dedicated to all other processes, etc. other than graphics?
> 2. Assuming the answer to the above [...]

MitchBradley, 2017-11-26:
To add to what @cheton said:

> 1. If the Pi is running headless without a monitor connected to it, are the computing resources dedicated to all other processes, etc. other than graphics?

Yes, especially if you have the Pi configured to boot to a console instead of to a graphical desktop.  You can use raspi-config to configure that option.  If the desktop is enabled, about 200MB of memory is consumed by the X window server (Xorg), the desktop management software and various ancillary programs.  That's 20% of the 1GB memory on a Pi3 - or 26% of the 750MB that is left after dedicating 256MB to the GPU.  Desktop CPU utilization is insignificant if you are not interacting with with, but the memory is still consumed even if you are not doing anything.

> 2. Assuming the answer to the above question is yes, would interfacing with the Pi and CNCjs via VNC viewer also allow the computing resources of the Pi to be dedicated to everything but the graphics?

VNC would make it worse!  With VNC, all the graphical stuff still happens on the Pi.  VNC watches what is on the screen and sends the information to duplicate that image over the network.  All the display work to create the visuals still happens on the Pi, and then the result has to be sent over the net, consuming additional resources.  You can think of VNC as a "virtual camera" that is watching what is on the screen.

The way to run the Pi efficiently, as @cheton said, is to run just the cncjs server on the Pi, and let all the browser stuff happen on a more capable machine.  For the cncjs full UI, "more capable machine" boils down to "Mac, PC or PC laptop", because only those machines have the combination of a big enough screen, enough memory, and a powerful enough GPU to do a satisfactory job with the full cncjs UI.  A high-end tablet might barely work, but, in my experience, touchscreens are not well suited for that UI because the tiny controls that are difficult to hit reliably.  The last thing you want when controlling a machine is to hit the wrong control.  Similarly, smartphones these days have enough "compute and GPU grunt", but the screen is too small for the controls.

cncjs-shopfloor-tablet is specifically designed for use on a touchscreen - it has big buttons that are easy to hit, a layout that avoids putting dangerous buttons close to other things, and a fixed layout that does not scroll.  That latter factor is important because you really don't want the panic (stop) button to be off the screen just when you need it to avoid breaking a bit or gouging your mill tablet.  As a bonus, it doesn't stress the browser much, so it works well even on low-end tablets (I got one for $100 at WalMart) and lightweight browsers without 3D support.

> 3. If the answer to question 1 is no, does it mean the gpu functions have to be intentionally disabled? If so, how?

I recommend disabling the GPU on a Pi .  In my experiments, you can coax a Pi into displaying a 3D toolpath by turning on the GPU and using a suitable browser (Chromium or Firefox), but the cost is too high.  The GPU takes away 25% of the total system memory, and the amount that is left is not enough for the browser to perform well enough to control a machine reliably and safely.  Consequently, there is no reason to leave the GPU on.  If you turn it off (using raspi-config > Advanced Options > GL Driver > Legacy) and lower the memory split (raspi-config > Advanced Options > Memory Split > 16), then you avoid any bugs that might be in the still-experimental GPU driver and release 25% of the system memory for other uses.

With the GPU disabled, the full cncjs UI will work on a Pi, but the 3D viewer will not work.  The UI will fall back to displaying the GCode as text.  It is functional in that mode, but still sluggish.  For example, when I click on the jog buttons, there is a noticeable delay on the order of half a second before the machine moves, and another similar delay before the DRO numbers update.  That's not acceptable for my needs.

With the simpler UIs, cncjs-shopfloor-tablet and cncjs-pendant-tinyweb, all operations are instantaneous by human perception standards.

A later post will show just how much memory is needed to run the various UI options.

MitchBradley, 2017-11-26:
**Analysis of CNCjs Memory Usage on a Raspberry Pi**

Pi2 and Pi3 have 1GB of main memory - other Pis have half that - 512MB.  To display 3D graphics using the GPU, you have to dedicate 256MB of that memory to the GPU.

The Linux "top" command can be used to see how much memory is being used by various processes.  The following data was collected on a Pi2 running Raspbian Stretch, updated as of 11/24/2017.

**Desktop + cncjs server**

We will start from the situation where the Pi has just been turned on and has autobooted into the graphical desktop environment.

[image]

The picture above shows the state of memory when the graphical desktop is running but neither the CNCjs server nor any browser is running.  The highlights, from left to right, show.

1. 750 MB (750144 KB) total memory is accessible by the CPU.  That reflects the 1G of actual memory minus the 256 MB dedicated to the GPU.
2. Of that, 361 MB is currently free - not used for anything.
3. No ("0 used") swap space is being used (more on that later)
4. About 25% of the memory is being used by processes in the square yellow box, most of which are related to the graphical desktop / window system.  5.8% is the window system itself - Xorg.

At this point there is still a reasonable amount of memory left to do stuff.

Next we will run the "cncjs" server process.

[image]

1. Free memory has dropped from 361 MB to 289 MB - so starting the cncjs server used up about 70 MB, but there is still enough memory to do useful stuff.
2. Swap space is still unused (0 used)
3. The "node" process, which is the "NodeJS" Javascript environment that hosts the cncjs server, is using 7.4% of the system memory.

The memory situation is still just fine.

**Chromium Browser + full CNCjs UI**

Now we will start the Chromium brower.

[image]

1. Free memory has dropped a bit more (not highlighted), but now we are starting to use swap space (9912 used).  This is a big deal!  It means that Chromium needed, at least temporarily, more memory than the system actually has.  Swapping is a technique used to let a computer pretend that it has more memory, by shuffling memory contents back and forth onto the "disk drive" - which, in the case of a Pi, is a micro-SD card.  Swapping can let you do things that you could not otherwise do, but once a computer starts swapping, the performance falls off a cliff.
2. The square yellow boxes show that the Chromium browser is using 4 separate processes consuming a total of about 37% of the system memory - and it isn't displaying anything interesting yet.

Next we will point Chromium at cncjs so it displays the CNCjs UI.

[image]

1. Free memory has dropped some more, and swap space usage has increased to 14.9 MB - so the system is deeper into the "oh no, we are swapping" zone.
2. The memory usage by the 4 Chromium processes is now approaching 45% of the system memory.

Finally, we will load a GCode file and let the 3D Viewer display it.

[image]

1. Even deeper into swap territory - 25.9 MB used
2. There are now 5 Chromium processes, using a total of 49% of system memory.

At this point, the performance is quite bad.  It often takes several seconds for the display to update after you click on a control.

**Epiphany Browser + cncjs pendant UI**

Now we go back to the state where the desktop and cncjs server is running, but the browser has not been started.  To do that we have to reboot to clear out the swap space that Chromium was using; just stopping Chromium isn't enough to release it.  Then we start the Epiphany browser instead of Chromium.
[image]

1. Unlike with Chromium, no swap space is being used, so the system is not in a performance problem danger zone.
2. There is only one Epiphany process, consuming 9.5% of memory, compared to 4 Chromium processes using 37% of memory.

Now we point Epiphany at the cncjs-pendant-tinyweb page.

[image]

1. Still no swap space used.
2. Epiphany is using a bit more memory - 11.7% - but less than 1/4 of what Chromium was using with the full cncjs UI.

The performance here is just fine - no noticeable delays on any of the controls.

ENGR-mike, 2017-11-27:
Mitch,
From what I can tell, cncjs-shopfloor-tablet requires installation of additional files on the Pi, but nothing on the tablet itself.  It appears access to the tablet UI is by means of entering the unique url  http://host:8000/tablet/' in a browser on the tablet.  Am I correct?  If so, it seems any table would work; even an iPad.  Are there minimum requirements for the tablet?  I'd expect it should work fine on a 16GB iPad 2.

cheton, 2017-11-27:
@ENGR-mike 
From 1.9.10, CNCjs can support adding multiple mount points, like below:

```sh
cnc -v -p 8000 \
    --mount /tablet:/home/pi/cncjs-shopfloor-tablet/src \
    --mount /pendant:/home/tinyweb
```

So you can access them via the mounted URLs: `http://your-host:8000/tablet/`, `http://your-host:8000/pendant/`

ENGR-mike, 2017-11-28:
OK guys, here is what I have been able to do so far:

1.  I've successfully connected to the full UI with my iPad and controlled my mill with it.
2.  I've successfully accessed the tablet UI on the Pi and on the iPad

What I can't do is connect using the tablet UI.  The problem as I see it is the port number.  When mounting the tablet UI, it reported it was starting the server at http://127.0.0.1:8000, but I'm running CNCjs on port 8888 because 8000 appeared to be occupied when I was originally installing and setting up CNCjs.

As I said, I could access the tablet UI using URL http://<hostname>:8000/tablet/ but I could not make a connection to my mill.  I tried URL http://<hostname>:8888/tablet/ but the browser reported back that the server did not exist.

What now?

Thanks,
Michael

MitchBradley, 2017-11-28:
Please be specific about the commands that you used to start the server.

The "EADDRINUSE" problem that you experienced is probably due to having succeeded in autostarting the server without realizing it.  So when you tried to start it manually, a copy of the server was already running, holding down the 8000 port.

Did you use the PM2 method of configuring for startup?

What is the result of issuing the commands:

```
pm2 list
ps ax | grep node
```

MitchBradley, 2017-11-28:
Oh, wait - reading more carefully, I see that you did get to the tablet UI.  I was confused by the word "connection", which has two meanings - either connecting from the UI to the server over the net, or to the machine controller from the server over serial.  It seems that you can do the former and not the latter.  If so, that is probably my problem and I will fix it.  I made some assumptions in the tablet code to make it faster to connect to my mill.  I will fix that.

Are you using GBRL on your controller?

MitchBradley, 2017-11-28:
I was wrong about tho aforementioned assumptions; I guess I did a better job than I was giving myself credit for.  I just now connected to the tablet UI from Chromium on Pi and thence connected to grbl via /dev/ttyUSB0, then jogged my laser engraver.  So all that is working.

Okay, back to the previous debugging approach.  One possibility is that there are two cncjs servers running, one on port 8888 and one on port 8000.  My hypothesis is that you did something like the following sequence:

Started a cncjs server on port 8888, possibly automatically via pm2
Accessed its full UI at <hostname>:8888/
Connected to your machine via some serial port
Without killing the first cncjs server, started another cncjs server on port 8000 via the command line, giving the arguments to mount the tablet workspace
Accessed the tablet UI at <hostname>:8000/tablet

If that is correct, the likely problem is that the first server on 8888 is holding open a connection to the machine on its serial port.  The second server on 8000 is unable to get that serial port because it is already in use.

This would be yet another manifestation of the "ADDRINUSE" situation, but this time instead of the port number being the conflicted resource, the conflict is over the serial port hardware interface.  It's like you can't make a phone call because someone else is already on the line.

MitchBradley, 2017-11-28:
And just to clear up a point that I expect you might be confused about (and I totally understand why you might be confused; this is complicated stuff that non-programmers can't be expected to just know - and even programmers can get confused because the landscape is so varied.)

It seems from one of your earlier messages that you are thinking the port number 8000 is necessary for using the tablet workspace.  That's not the case.  The port number, whether it is 8000 or 8888 or anything else, is the cncjs server's "phone number extension".  The <hostname> part is like the phone number that gets you to the office building, and then you need the extension (port number) to get to a specific program.

Assuming that you are autostarting cncjs via pm2, the right way to enable the tablet workspace is to change that autostart setup.  The Setup Guide instructions say to do

```
pm2 start $(which cncjs) -- --port 8000 -m /tinyweb:/home/pi/tinyweb
pm2 save
```

I guess, based on your earlier messages, that you changed 8000 to 8888 in that command.  If you also want the tablet workspace, you need to say:

```
pm2 start $(which cncjs) -- --port 8888 -m /tinyweb:/home/pi/tinyweb -m /tablet:/home/pi/cncjs-shopfloor-tablet/src
pm2 save
```

But we still may not be out of the woods, because the pm2 configuration may still have the old start command active.  I will try to figure out how to sort that out.  I'm not a pm2 expert; instead I have been using "old school" methods to auto-start cncjs, instead of this newfangled pm2 thing.  I will try to learn enough about pm2 to work out how to wrangle it.

ENGR-mike, 2017-11-28:
> Accessed its full UI at :8888/
> Accessed the tablet UI at :8000/tablet

Sorry I wasn’t more specific in my email.  I was pecking it out on a tablet and didn’t have the energy to peck all the details.  Now that I’m sitting at a desk with a keyboard in front of me, let me elaborate:
 
Yes, my CNCjs server is starting automatically on port 8888 using PM2 and I’m using GRBL.  I had followed the original Raspberry Pi setup guide and followed the PM2 method of configuring the startup.  However, I had problems (self-created) using port 8000.  When originally following the Raspberry Pi setup guide and installing CNCjs via command line, it was not clear to me I needed to switch back to the GUI and type the URL into a browser to access CNCjs.  I think I was expecting a browser to pop up after typing “cnc” on the command line (that’s what happens when a newbie tries to do this kind of stuff at night after a full day at a day job).  So, with CNCjs already started using PM2, I was typing “cnc –v” (or something like that) to figure out what was happening.  However, all I was doing was trying to start CNCjs a second time, which explains why I was getting the “EADDRINUSE” error regarding port 8000.  It was at that point Cheton suggested using port 8888, which really was not the solution to the problem.  At some point during my troubleshooting dialogue with Cheton, he stated the server was probably already started.  So, with a clear head and sleep, it occurred to me to switch to the GUI and type the URL “http://<hostname>:8000/” in a browser.   
 
Keep in mind this was like trying to type a letter in a foreign language and I was only typing what I was being instructed to type.  All of the command strings I was typing at the command line prompt were like jibberish to me.  So, now that we are probably on the same page, I would appreciate your direction to achieve the following:
 
1.       Automatically start the CNCjs server upon booting up the Pi.
2.       Have access to multiple versions of the UI, so I can load a G-code file and start a job from my desktop, and run the job from a tablet in the shop.
 
Please give me an explicit list of commands in the proper order.
 
P.S.  There were also some commands I originally typed during the installation/setup to redirect port 80 to 8000 or something like that.

Sent from Michael's iPhone

MitchBradley, 2017-11-28:
The challenge of giving you an explicit list of commands is that I don't know what state your computer is presently in.  What is already running?  Has pm2 already been configured to autostart something?  If so that might, depending on what it is, need to be undone.  Where, if anywhere, have you unpacked the ancillary UIs like pendant and shopfloor?  How are your tablet and desktop networked to your Pi?

Since the commands are like gibberish, will you be able to adapt if I make a guess that is wrong, and thus mislead you?

This is the sort of thing that does not lend itself to a set of exact instructions because there are so many variables.  It would be better to work through it interactively, perhaps via texting or Google Hangouts or Skype.  Give me contact info for one of those channels and I will help.

MitchBradley, 2017-11-28:
As a starting point, let's turn off the current pm2 autostart so we can start fresh.  Type
```
pm2 list
```
Look in the "App name" column to find the ID of the cncjs process.  It will probably be either "cnc" or "cncjs".  Depending on what it is, type either
```
pm2 delete cnc
```
or
```
pm2 delete cncjs
```
Do you seem my dilemma about giving explicit commands?  Virtually every command I would give has possible variations that need to be correct for your setup.  That's why we need to go interactive.  Otherwise it will take forever and we will both go crazy.

ENGR-mike, 2017-11-28:
No, I do realize what you mean.  Actually, those look familiar.  I believe I did something like that when I changed the port from 8000 to 8888.  

I'm actually smarter than I may sound.  Yes, it's all gibberish at first.  I catch on after a while, but forget it again if I'm not using it regularly (I suffer from the use it or lose it syndrome).  The logic and order of things does not completely elude me.  I'd probably spend more time trying to learn some of these techniques and code if I thought I would use it in the future.  Unfortunately, I only have so much time and I'd rather spend it designing 3D models, setting up the CAM operations and making chips.

Your help has been invaluable and you are to be commended for your willingness to provide me with so much assistance.  Let me try a few things tonight and get back with you.  I have to do that day job thing for another 4 1/2 hours before I go home and have access to the Pi, etc.

MitchBradley, 2017-11-28:
I understand.  I have been using Linux or its predecessor for almost 40 years, and I forget stuff - plus there is always some new wrinkle to worry about, or something that changed to invalidate what you used to know.

ENGR-mike, 2017-11-28:
Well, I'm a Mechanical Engineer working as a Sr. Facilities Engineer in the process industries (i.e. oil & gas, chemical, power, etc.).  Designing and maintaining oil & gas platforms, refineries, chemical plants and power plants makes up 90% of the engineering jobs in my area.  It's not what I wanted to do, but I didn't have many other choices locally.  After 25 years doing this *%^!, I still don't like it.  My true love is machine design and building my CNC mill has been an effort to get back to what I love about mechanical engineering.  CAD/CAM and fabrication is what I love and do in my spare time.  I've been building/fabricating things since I was a kid.  Some are born with silver spoons in their mouths, but I was born with tools in my hands.

Thanks again and I will get back with you tonight.

MitchBradley, 2017-11-29:
If you want to work together it will have to be soon because I am about to go out for several hours.  I am in Hawaii.

ENGR-mike, 2017-11-29:
Well, I was reasonably successful.

Regarding PM2, I did the following:

"pm2 list" to discover if anything else was being managed by PM2, but CNCjs was the only app listed
"pm2 stop all" to stop cncjs
"pm2 start $(which cncjs) -- --port 8000 -m /tinyweb:/home/pi/tinyweb -m /tablet:/home/pi/cncjs-shopfloor-tablet/src" to restart cncjs with both UI's mounted
"pm2 save"

Following the above, I first tested the UI's using Chrome on the Pi.  I successfully accessed both UI's after being prompted to login to CNCjs with my account information.  I was able to connect and control my mill with both UI's, although the full UI was not always agreeable in making a connection.  I counted it as a success.

I then switched to the iPad 2 and typed in the URL for the tablet UI, which directed me to the CNCjs login screen.  Following my login, it strangely went to the full UI.  I tried typing in the URL for the tablet UI again, but it would not access the tablet UI page.  It would only bring up a blank page.  I tried clearing Safari's history, etc. and even hard booting the iPad, but nothing allowed me to access the tablet UI.  I could only access the full UI from the iPad, but connections were spotting.  I was also able to control the mill from the full UI on the iPad.  I also made sure I closed the connection each time I tried to switch from the full UI to the tablet UI, but it did not help.

I decided to switch to my iPhone and was able to access the tablet UI after being prompted to login to my CNCjs account.  However, it first went to the full UI following the login.  I retyped the URL for the tablet UI and it opened without a problem.  I also connected and was able to control my mill from the tablet UI on the iPhone.

I'm baffled why I type the tablet URL, get the login screen and the full UI opens after logging into my CNCjs account on both the iPad and iPhone.  Since I tried the full UI first on the Pi, I cannot say what would have happened if I tried the tablet URL first.  I hope all of the above information helps.

I was initially concerned about whether there should be some sort of operator/divider between the two pm2 start -m options/arguments for mounting both UI's.  My concern was alleviated when I was able to access both.  I also want to point out that when downloading the ZIP file with the tablet UI files, the main  folder within the ZIP file was named "cncjs-shopfloor-tablet-master" and that's how the files were extracted to my Pi.  I manually changed the folder name by deleting the "-master" part at the end.  Finally, the main menu on the top right of the tablet UI expands down and to the right so that it is cut off by the edge of the screen.  You might want to change how it expands, if that is possible.

I would really like to get the tablet UI to work on my iPad, since that is the purpose of that UI.  I suspect this is some kind of weird Apple *&?# related to the age of the iPad (it's an iPad 2 with 16 GB memory).  My iPhone is a 6S, which is a much newer generation Apple device.

PS  No problems with either UI from my Windows 10 PC.

ENGR-mike, 2017-11-29:
It just occurred to me as I'm heading to bed that I can try to use Chrome on the iPad to access the tablet UI.  I'll try tomorrow.

cheton, 2017-11-29:
This is a bug that previous URL was not remembered after login. I will try to fix it in a later release.
> I decided to switch to my iPhone and was able to access the tablet UI after being prompted to login to my CNCjs account. However, it first went to the full UI following the login. I retyped the URL for the tablet UI and it opened without a problem.

ENGR-mike, 2017-11-30:
Apparently, the blank page thing is a known problem with Safari.  Unfortunately, I also experienced it with Chrome on my iPad 2.  So, as it stands I can access both UI's on all devices accept the iPad 2.  In Chrome, it appears as if the page is fully loaded, but it is completely blank.  As I said before, I cleared the browser history and even rebooted the iPad without any luck.  I'm researching possible solutions now.

MitchBradley, 2017-11-30:
Can you access the pendant workspace on your iPad2?  The URL would be <hostname>:8000/pendant instead of .../tablet.  It pendant works and tablet does not that would be very interesting because tablet is derived from pendant.

MitchBradley, 2017-11-30:
Sorry, I meant <hostname>:8000/tinyweb, not .../pendant

ENGR-mike, 2017-11-30:
Found this
http://bytes.schibsted.com/ios7-bug-shows-white-page-when-getting-304-not-modified-from-server/

ENGR-mike, 2017-11-30:
I get page not found, probably because I don't have it mounted.  Am I correct?

ENGR-mike, 2017-11-30:
Well, according to the pm2 commands I typed, I do have it mounted.  I guess???

MitchBradley, 2017-11-30:
Depends on what "it" is.  Did you try .../pendant or .../tinyweb ?  Or both.

Specific : good.  Vague: wastes everyone's time.

Please please please be explicit.

ENGR-mike, 2017-11-30:
I get a page not found with both /pendant and /tinyweb.

cheton, 2017-11-30:
Try adding a backslash to the end: `/pendant/` and `/tinyweb/`

ENGR-mike, 2017-11-30:
I get a page not found with both /pendant/ and /tinyweb/.

MitchBradley, 2017-11-30:
One thing that can cause that is if the wrong directory is specified in the mount part of the cncjs command line.  In a previous message, you said that you did

"pm2 start $(which cncjs) -- --port 8000 -m /tinyweb:/home/pi/tinyweb -m /tablet:/home/pi/cncjs-shopfloor-tablet/src"

Try this from the command line:

```
ls -l /home/pi/tinyweb/index.html
```
If it says "No such file or directory", you need to change the directory in the mount part to a directory that does contain "index.html" - and also "app.js" and "controller.js".  It might be the src/ subdirectory .  If so, the command to test it would be:

```
pm2 stop cncjs
cncjs -vv --port 8000 -m /tinyweb:/home/pi/tinyweb/src -m /tablet:/home/pi/cncjs-shopfloor-tablet/src
```
That sequence does not affect the autostart setup; it simply stops the cncjs process for the current session and restarts a new one with debugging turned on.  To change the autostart, you would need to reissue the pm2 start command with the correct arguments, after deleting the old recipe.
```
pm2 delete cncjs
pm2 start ...
pm2 save
```

ENGR-mike, 2017-11-30:
I'll have to try that later after eating dinner, but I think it is irrelevant.

I googled the blank page in Safari issue and obtained plenty of hits documenting the problem.  I even found that it can happen in Chrome on and iPad.  Apparently, it is a known issue that persisted for several versions of iOS since about iOS 7.  Unfortunately, the iPad 2 is limited to running iOS versions prior to 10 only and Apple no longer supports the iPad 2.  Since my iPhone 6S is a newer generation Apple device it is capable of running the latest iOS version.  It would seem Apple corrected the blank page problem in later iOS versions beyond iOS 9 and that is probably why I do not experience the blank page problem on my iPhone.  I was hoping my wife had her newer iPad from work, but she did not bring it home today.

MitchBradley, 2017-11-30:
But if the problem is a bad cached page, as the link you gave says, then clearing the cache should make the page come up.

MitchBradley, 2017-11-30:
Best Buy has a 10" Android 7 tablet on sale for $70

ENGR-mike, 2017-11-30:
It appears the tinyweb folder is not on my Pi.  I originally installed everything following the Pi Set-up Guide, so I don't know what to say about the folder being missing.

ENGR-mike, 2017-11-30:
I now have tinyweb installed on the pi.

Accessing/loading the tinyweb or tablet UI in a browser on the iPhone 6S is completely successful.

Accessing/loading the tinyweb or tablet UI in a browser on the iPad 2 results in the following:

tinyweb - page loads and displays, but says there are no options in the port drop down box to select.
tablet - page appears to load but displays as a blank page.

My daughter is a sophomore in college and my son is a junior in a private high school.  I work in the oil & gas industry, which has been in a slump for about 3 years now.  I have not had a raise in almost 4 years and I took an 8% pay cut last year.  Consequently, I have limited funds.  $70 for a new Android tablet does not seem like much, but I'm going to have to wait and see before I make another purchase towards t his mill.  This diy CNC mill project is turning in to a money pit.

MitchBradley, 2017-11-30:
What happens if, on the iPad 2, you access the URL:

<hostname>:8000/tablet/#/axes

or

<hostname>:8000/tablet/#/connection

MitchBradley, 2017-11-30:
I meant, of course, _hostname_:8000

[21 more replies]