cncjs/cncjs #225: Zeroing issue

SteveHogg, 2017-11-01:
#### Description

I'm completely new to this but thank you for this software. I have CNCjs installed on windows (v1.9.11 installed using the windows installer) and my machine is an MPCNC with a Smoothieboard. My machine has no endstops and I'm just trying to find out the correct way to zero the machine so I know where it will start. The current best method I've found is to use the homing button which has the strange result that it moves about +6mm in X and Y and about 1-2mm in Z. I understand I don't have any endstops installed nor configured, but it is odd that there is this little movement, if anything I would have expected it to move up to the limits of the machine (crashing if not starting from the max) and then stopping due to a safety restriction.

I've tried using the temporary zero and the other zero button but once I start a cut it still moves to machine zero first. I'm sure I'm just doing something wrong, but I don't know what. 

The gcode I'm using is coming from the default smoothieware post in Fusion 360. 

Perhaps the best way would be to use the temp zero workspace and have the gcode in relative mode, but I haven't figured out how to do that from the Fusion 360 post, nor whether it would even help. Any suggestions?

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [x] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2017-11-01:
The generic smoothie post, smoothie.cps, generates a "G28 G91 Z0" command. 

[image]

 That is what is causing the unwanted motion in machine coordinates.  Your idea of using only work coordinates is a very good idea for a machine without homing switches; it is the way I have worked for years on desktop mills and laser engravers.

Here's how to modify the Fusion smoothie post to do that:

(You can do all of the following steps using ordinary Windows operations and an editor like Notepad, but it is probably easier to do it from within Fusion, as shown below).

**First copy the generic smoothie file from Fusion's Generic Posts directory:**

[image]

---

[image]

---

[image]

---

**Paste that file into Fusion's Personal Posts directory:**

---

[image]

---

[image]

---

[image]

---

**Rename the copy to "mpcnc.cps"**

---

[image]

---

[image]

---

**Start editing the new file:**

[image]

---
 
**First change the description** 

[image]

**Change those lines to whatever you want, e.g. "MPCNC Smoothie"**

**Then comment out, by inserting // at the beginnings, the lines that issue G28 commands.  (The easy way to reach those lines is by using the Find menu to search for 28.**

[image]

---

[image]

---

**Save the file, then run the new post from the Personal Posts directory.**

[image]

**Verify that the generated GCode no longer has the offending G28 command:**

[image]

**Win!**

MitchBradley, 2017-11-01:
Just for completeness:  If you would rather not modify the generic post, it is probably possible to manually set the machine coordinate home positions.  First you jog to the position you want to be 0,0,0, then you issue an appropriate command.  The smoothieware documentation is somewhat unclear on the correct command to use, but it claims to support G28.3, so you might be able to use "G28.3 X0 Y0 Z0" after manually jogging home.  CNCjs has that command on a dropdown menu in the Axes widget:

[image]

GRBL, if homing switches are not configured, will set machine 0,0,0 to wherever the machine is at power up.  Smoothieware might do that to if it is in GRBL mode.  It might do it in non-GRBL mode.  If so, that might be another way to establish the machine coordinate system.

But, all things considered, if you don't have homing switches, I recommend avoiding G28 entirely.  Zero the G54 work coordinate system by jogging to the origin relative to your stock and using the "Zero Out Work Offsets" item in them menu shown above.

Oh, and to answer your implied question about why the machine only moved a short distance during homing, it is probably because of some soft limit value that is configured somewhere.  I find smoothieware configuration somewhat bewildering (pretty much par for the coarse considering how adaptable it is), so I can't point you directly to where the setting might be.  It doesn't really matter anyway; since you don't have switches, you can't auto-home the machine regardless of the soft limits.

SteveHogg, 2017-11-02:
Wow, thanks for the detailed response. I've updated the post script as suggested and will give it a go maybe tonight after work. Also I don't recall seeing "Set Machine Zero" before but I could have missed it. I have a look for that as well since it seems the combination of the two is exactly what I'm looking for.

MitchBradley, 2017-11-02:
Actually, combining the two approaches is pointless.  If you avoid G28 and always use work coordinates, then you do not care what the machine coordinates are.  I have used that approach since forever.

Machine coordinates start to become interesting in professional/production setups, where you have multiple fixtures that can be set precisely relative to the machine.  In that environment you establish different work coordinate offsets (G54, G55, G56, etc) for the various fixtures, so you can automatically switch between fixtures.  The work coordinate offsets are in machine coordinate space.

For low volume or hobbyist use, it is typically more convenient to manually jog to the part origin then set the work coordinates to 0 there.  Never even think about machine coordinates.

SteveHogg, 2017-11-03:
Actually in my case the Set Machine Zero was what I really needed. Tried it out and its exactly what I was looking for. I don't know why I hadn't seen it there before. Thanks, makes it so much easier to set it up each time!

closed 2017-11-03 by SteveHogg