cncjs/cncjs #227: USB numeric keypad as pendant

goloplo, 2017-11-14:
How can I map the keys on an USB numeric keypad to convert it in CNCJS pendant?

cheton, 2017-11-15:
@goloplo 

Do you want to connect the USB numeric keypad to a Raspberry Pi?

[node-hid](https://github.com/node-hid/node-hid) is a solution to access USB HID devices. You can find a pendant example at https://github.com/cncjs/cncjs-pendant-keyboard/blob/master/bin/cncjs-pendant-keyboard

GorillaMachines, 2017-12-08:
I was thinking about this as well and found that it would be fantastic to have some pendant where I could use a wireless usb numpad to control the machine. Wireless is key for me though. Is it possible to have the pendant run on a desktop, possibly as an addition to the cncjs software? The ability to connect external control interfaces would be amazing.

cheton, 2017-12-09:
@GorillaMachines 

Do you want to run CNCjs desktop application and a pendant client on the same Windows device? The pendant client is possible to run from the Windows command prompt, but I haven't tested the pendant program on Windows. It may require a little modification.

closed 2020-03-04 by cheton