cncjs/cncjs #229: Z axis sent to TinyG doesnt match position in the gcode.

mcentyre, 2017-11-22:
#### Description

Trying to move to cncjs from ChiliPeppr. I'm having an issue with my initial test job. I created a simple circular pocket toolpath in Aspire. Used the TinyG mm post processor. Pocket should be 10mm.  The Z axis commands in the gcode appear to have the correct coordinate. Watching the console in cncjs, it appears to be doubling the Z axis position so its going to twice the expected depth. X and Y do not have the same problem. Moving the Z axis manually seems to work correctly. 

#### Versions

- CNCjs: 1.9.11 Windows x64 running on a WinXP desktop

#### How Do You Install CNCjs?

- [ ] NPM
- [ X] Download the CNCjs Desktop 

#### CNC Controller
TinyG
- [ ] Grbl
- [ ] Smoothieware
- [ X] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [X ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [X ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2017-11-22:
Please post the GCode and I will try it with my TinyG setup.

phillips82, 2017-11-25:
I am now having an issue with CNCJS. It did not do this the first couple of weeks and now it is. It is randomly stopping and giving me this error. It happens dozens of times per file. Anyone help fix this?

[image]

cheton, 2017-11-25:
@mcentyre @phillips82 

Please post your firmware build and firmware version, and the G-code file that would cause this error, so I can check if there is something went wrong. 

Also see https://github.com/cncjs/cncjs/issues/190, if you're trying to build a G2 firmware by your own, you have to be careful to take the right configuration and the right motate branch, or it may randomly cause errors with an incompatible firmware.

mcentyre, 2017-12-04:
Sorry, I haven't been able to get back to this.  Attached is the gcode file and the Aspire post processing file. 
[TinyG_mm.zip](https://github.com/cncjs/cncjs/files/1528847/TinyG_mm.zip)

[simple_test.zip](https://github.com/cncjs/cncjs/files/1528841/simple_test.zip)

cheton, 2017-12-05:
Your G-code seems to be working fine, I don't see any errors appear to the console.

**-2.500mm**
[image]

**-5.000mm**
[image]

**Program End**
[image]

---

What's about your Z work offset value? Have you ever perform a Z-probe before doing this job?

mcentyre, 2018-01-10:
Tried it again. The Gcode looks okay for the depth. There is and ARC error appearing in the console.  I stopped it after the first pass but that pass is at about 7mm deep. Can you tell me something about the ARC error?
[image]

MitchBradley, 2018-01-11:
The console output does not match the GCode you posted earlier (simple_test.zip).  The console output shows the following sequence:
```
G91
G0 Z-0.1
G90
G10 L20 P1 Y0
G92 Z0
G28.3 Y0
G3X137.867Y88.500*0.000J-0.632
<followed by the Arc specification error message>
```
The simple_test.tap file that you posted instead contains this sequence:
```
G90
G17
G21
G0Z20.320
G0X0.000Y0.000
M3 S12000
G0X138.500Y88.815Z5.080
G1Z-1.250F360.0
G3X138.185Y88.500I0.000J-0.315F900.0
G3X138.500Y88.185I0.315J0.000
G3X138.815Y88.500I0.000J0.315
G3X138.500Y88.815I-0.315J0.000
G1Y89.132
G3X137.867Y88.500I0.000J-0.632
```
The last line is the same, but the preceeding lines are different.

"Arc specification error" means that  the current starting position and the ending position given in the command are not part of a circle with the given center position.

In the simple_test.tap program, it is possible to determine the current starting position - Y is 89.132 as a result of the line just before the final G3 line, i.e. "G1Y89.132", and X is 138.500 as a result of the G3 line third from last.  That's a proper quarter-circle arc - it starts at (138.5, 89.132) and ends at (139.133,88.5), with the center at (138.5, 88.5).

In the console output, it's impossible to determine the starting X value, because there are no "posx" values in the "sr" position reports, and no GCode commands that set the X value.  It is, however, possible to determine that starting Y=0.  Since we know that the final G3 line in both cases is "G3X137.867Y88.500I0.000J-0.632", and that is a good ending point for a starting point of (138.5, 89.132), it seems highly unlikely that said G3 line is consistent with a starting point of (something, 0).

So we are left with the question: How did those other lines get into the GCode stream before the final G3 line in the console output.  It turns out that the commands in the console log prior to the offending G3 are all used by User Interface position setting buttons.

* "G91", "G0 Z-0.1", "G90" is a jog in the -Z direction for a distance of 0.1 units
* "G10 L20 P1 Y0" is "Zero Out Work Y Axis"
* "G92 Z0" is "Zero Out Temporary Z Axis"
* "G28.3 Y0" is "Zero Out Machine Y Axis".

It appears that a GCode program may have been paused, then some jogging and resetting of coordinates occurred, the the GCode program was resumed.  But after resuming, TinyG could not proceed because the new Y position was not a possible starting position for an arc that ends in the specified position.

I assume that you stopped the program because it was running too deep, then repositioned, and somehow resumed the program from where it left off.  In that case, the Arc specification error is not directly a problem.  Rather, the problem is the resumption after having invalidated the positioning.

In simple_test.tap, the fourth line is `G0Z20.320` and the seventh line is `G1Z-1.250F360`.  Your problem would be explained if your machine has less than 20 mm of "flying height" above the work, so that the first G0 is not actually taking the tool to 20.32 mm above, but instead "topping out" somewhere lower than that.  The program, however, thinks the tool really is at 20.32, so when it tries to plunge to Z-1.250, it goes too deep.

mcentyre, 2018-01-11:
I'll run it again. It could be a problem with grabbing the wrong part of the console with the screen shot. Is there a way to copy/export the actual console data as a whole?  I couldn't figure it out and went with the screen shot. 
The sequence of events is this:
-Load the gcode
-Position the bit in the home position (I don't have a z probe)
-Set the current position to be zero for all axis. 
- Send the gcode. 
- Bit moves to the center of the circle and plunges to a depth of approx -7mm. 
- Once it plunges, an Arc error appears in the console and movement stops. 
- I click resume. Twice, as the error repeats. 
- The job continued to run at the same depth. 
- Before it completes the first pass, I click Pause and stop the job. 

I'll rerun the job tomorrow and be very careful about the steps and the gcode file used. I believe that I used the same one, but I could have made a mistake. Thanks for taking your time to look at it.

mcentyre, 2018-01-11:
Also, I have plenty of movement above 20.32,. Or at least I should. I'll check to be sure I don't have an obstruction causing the scenario you described.  Thanks again.

cheton, 2018-01-11:
Hi @mcentyre 

@amx1 and I had fixed the G2 arc code error around one year ago (see https://github.com/cncjs/cncjs/issues/106). May I know your TinyG firmware build version? You can type `{fv:n}` and `{fb:n}` in the console to display the firmware information.

cheton, 2018-01-11:
Hi @MitchBradley,

Running a G-code program will only respond sent/received line number and the occurrence line of error to the UI to reduce unnecessary network traffic and improve the UI performance. I think he just did some UI operations and then ran a G-code program without clearing the console first, so we may get confused about the output. Displaying errors with a different color or use a prefix string to indicate different sources (e.g. sender, feeder, etc.) might be good for UX improvement.

MitchBradley, 2018-01-11:
Hi @mcentyre,

I thought of another possibility.  If the motor tuning on your  Z axis were to aggressive, then it could be missing steps during the rapid move to 20.32, making the program think it's higher than it is.  One way to test that would be to manually jog to 20.32 or higher after setting Z0  and before starting the program. That way the z motion during the run would be downward only, which, obviously, takes less motor torque than upward.

mcentyre, 2018-01-11:
[image]

mcentyre, 2018-01-12:
Well, it looks like you were on to something with the Z axis motion being the problem. This was suppose to be moving 5mm with each movement.  Looks like I have some tuning to do.
https://streamable.com/ds2um

MitchBradley, 2018-01-12:
After tuning my motors, I did a stress test by making a GCode file with repeated rapid motions over the full travel of each axis.  I did something like 50 back and forth motions on each axis, and also some diagonal movements to test simultaneous motion.  I found that my initial settings, which seemed to work on single tests, were too aggressive.  Sometimes it would go back and forth dozens of times then lose steps or stall on the next one.  You can check the result of repeated moves by checking that the end position is correct after the complete sequence.  Often I could just tell from the sound.  I would be doing something else in the shop while the test was running.  The sound would suddenly change and I would see that something had gone wrong.

mcentyre, 2018-01-12:
I've been through all the tuning before and had everything working great at one point.  Unfortunately, it's been a while since then.  I'll have to go back and re-learn where to make adjustments (other than the pot itself). I'm not sure why anything would have changed so I guess I'll start with wiring checks and move forward from there. Thanks again for your help.

mcentyre, 2018-01-12:
Unrelated to this issue, but is there documentation on the UI somewhere? Such as explanations of the axis settings (the different ways to "zero" out each axis)?

MitchBradley, 2018-01-12:
https://github.com/synthetos/TinyG/wiki/TinyG-Configuration-for-Firmware-Version-0.97

If your Z axis moves the right distance for G1 commands at low feedrate, then the sa, tr, mi, and po parameters in the Motor Groups section are correct.  Then there are the Axis Group parameters, particularly $zvm, $zfr, and $zjm.

With TinyG, the long-term way to establish those settings is to recompile after editing the file firmware/tinyg/settings/settings_YOURMACHINE.h .  The relevant names in that file are -
 to control the distances per unit - M*_STEP_ANGLE, M*_TRAVEL_PER_REV, M*_MICROSTEPS, and M*_POLARITY.  For speed, Z_VELOCITY_MAX, Z_FEEDRATE_MAX, and Z_JERK_MAX.

For testing, you can set the $ parameters temporarily from the console, e.g.
```
{"$zvm": 300}
```
Then, at the console, you can type GCode motion commands to check the motion:
```
G0 Z20
G0 Z0
```
There is an alternative syntax that avoids the JSON boilerplate ({""}), but I don't recommend it because it can put TinyG into a state that confuses CNCjs.  The alternative syntax is fine if you are using a terminal program to talk to TinyG, but it's best to use the full JSON form from CNCjs.

MitchBradley, 2018-01-12:
Regarding the different ways to zero the axis, the domain of that topic is actually GCode and CNC in general, not so much the CNCjs UI.  Look up "G Code coordinate system" on the Internet.

Briefly, there are several coordinate systems:

### Machine Coordinates:

These are relative to the machine itself.  You establish machine coordinates during homing operations.  For machines without limit switches, or in cases where you prefer to establish zero positions manually, you can pretty much ignore Machine Coordinates and just use Work Coordinates for everything.

### Work Coordinates

These are the coordinates that you see in UI readouts (the jargon for such a readout is DRO - Digital Read Out - coming from the time when manual mills with dial knobs began to be retrofitted with electronic sensors).  The translation between Machine Coordinates and Work Coordinates is called Work Offsets.  If Work Offsets is (0,0,0), Machine Coordinates and Work Coordinates are the same.

Typically, to set up a program run manually, you insert the stock, jog to the stock-relative origin, and zero the work axes.  That has the effect of setting the Work Offsets to the correct values so that the Work Coordinates are now (0,0,0).

The GCode command for setting Work Offsets, i.e. for establishing the Work Coordinate origin, is "G10 L20 P1 ..."

There are some additional wrinkles that you can ignore for most hobby/DIY usage. I will summarize them just so you will know enough to safely ignore them:

### Multiple Work Coordinates

On big machines used for commercial jobs, it's common to have multiple fixtures for different parts of a job.  You could, for example, have 3 different vises semi-permanently clamped onto the machine at different places.  For such setups, it's convenient to have a different Work Coordinate system pre-set for each fixture, so it's easy to run the subjobs.  For this reason, most CNC controllers support at least 6 different sets of Work Offsets.  You can select the work offset set with G54 for set 1, G55 for set 2, ... G59 for set 6.  In the previous section, the GCode command for setting the Work Offsets was given as "G10 L20 P1 ...".  Actually, that command sets just the first one, the G54 system.  That's the "P1" part of the command.  To set the second set of Work Offsets (the G55 system), you would use "G10 L20 P2 ..."

By default, CNC controllers power up in G54 mode, i.e. set 1 is selected.  Unless you have a multiple-fixtures setup, you can just ignore all the alternate sets and pretend that G54 mode is all there is.

### Temporary Coordinates

Sometimes you want to "step and repeat" a GCode program to make multiple copies of the same part.  For example, with a single clamping setup, I can mill 3 copies of my product.  One way to do that would be to put three copies in the CAD drawing at appropriate places, the let the CAM program work out the coordinates for each copy.

Another way, which was more common back in the days before CAD/CAM got so good, is to have just one version of the GCode for the part, then to offset the coordinate system between runs of that same GCode.  That is "Temporary Coordinates".  It is set with the "G92" command, and it is applied "on top of" the Work Coordinate system.

My advice is to avoid G92 temporary coordinates like the plague.  I personally use them, with great care, but there are all sorts of ways they can bite you, especially if you have to interrupt a job in the middle and restart.  In that case, it is all too easy to have the temporary coordinate offset set to something strange without you realizing it, and the machine will go off into the weeds.

closed 2019-03-07 by cheton