cncjs/cncjs #230: Allow CNCjs Desktop version to remember the Apps Screen position and size when CNCjs size and position are changed.

edalger, 2017-11-24:
#### Description

For the Wish List: I wish CNCjs would remember it's size and screen position when changed. It would save some mouse work when adding other programs to the display.

#### Versions

x- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ Raspberry Pi
- [x ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [] Not Applicable
- [x ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2017-11-25:
Just found a useful code snippet:

https://github.com/electron/electron/issues/526#issuecomment-245821655

```js
const {app, BrowserWindow} = require('electron')
const Config = require('electron-config')
const config = new Config()
let win

app.on('ready', () => {
  let opts = {show: false}
  Object.assign(opts, config.get('winBounds'))
  win = new BrowserWindow(opts)
  win.loadURL(`file://${__dirname}/app/index.html`)

  win.once('ready-to-show', win.show)

  // save window size and position
  win.on('close', () => {
    config.set('winBounds', win.getBounds())
  })
})
```

edalger, 2017-11-25:
I suppose that the above code should be placed somewhere in the electron js lib?

cheton, 2018-02-03:
Fixed in 1.9.13

closed 2018-02-03 by cheton

edalger, 2018-02-03:
Thank you Cheton for adding this!
Ed...