cncjs/cncjs #231: Error Code 202 Issue. Pausing during file.

phillips82, 2017-11-25:
#### Description

Issue started appearing last week. CNCJS pauses and shows Code: 202, "msg": "move < min time", "~

Does this with all files and happens dozens of time while running a file. 
[image]

#### Versions

- CNCjs: 1.9.11

#### How Do You Install CNCjs?

- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] TinyG/g2core

#### Hardware

- [x] Desktop or Laptop

#### Operating System

- [x] Windows

phillips82, 2017-11-25:
Part of one file that it happens during.
G90
G17
G20
G0Z0.8000
G0X0.0000Y0.0000
M3 S22000
G0X-0.5874Y1.5788Z0.2000
G1Z-0.0650F30.0
G1X-0.5952Y1.5820F100.0
X-0.6041Y1.5863
X-0.6130Y1.5915
X-0.6218Y1.5974
X-0.6307Y1.6042
X-0.6397Y1.6119
X-0.6487Y1.6206
X-0.6576Y1.6302
X-0.6666Y1.6409
X-0.6756Y1.6526
X-0.6843Y1.6651
X-0.7026Y1.6942
X-0.7200Y1.7255
X-0.7366Y1.7590
X-0.7524Y1.7949
X-0.7673Y1.8331
X-0.7813Y1.8737
X-0.7944Y1.9165
X-0.8065Y1.9619
X-0.8178Y2.0087
X-0.8281Y2.0559
X-0.8373Y2.1031
X-0.8456Y2.1504
X-0.8528Y2.1978
X-0.8590Y2.2453
X-0.8641Y2.2930
X-0.8682Y2.3408
X-0.8747Y2.4345
X-0.8793Y2.5211
X-0.8820Y2.6005
X-0.8829Y2.6727
X-0.8825Y2.7184
X-0.8812Y2.7700
X-0.8791Y2.8273
X-0.8761Y2.8902
X-0.8741Y2.9225
X-0.8714Y2.9550
X-0.8680Y2.9878
X-0.8639Y3.0210
X-0.8591Y3.0546
X-0.8536Y3.0886
X-0.8474Y3.1229
X-0.8405Y3.1575
X-0.8331Y3.1913
X-0.8253Y3.2231
X-0.8171Y3.2531
X-0.8086Y3.2812
X-0.7997Y3.3074
X-0.7904Y3.3317
X-0.7809Y3.3541
X-0.7713Y3.3743
X-0.7620Y3.3912
X-0.7532Y3.4044
X-0.7444Y3.4151
X-0.7412Y3.4182
X-0.7378Y3.4213
X-0.7347Y3.4236
X-0.7320Y3.4254
X-0.7295Y3.4268
X-0.7292Y3.4269
X-0.7284Y3.4260
[... 1403 lines cut ...]
X-3.8293Y0.4521
X-3.8423Y0.5038
X-3.8544Y0.5583
X-3.8655Y0.6142
X-3.8756Y0.6701
X-3.8848Y0.7256
X-3.8930Y0.7808
X-3.9001Y0.8357
X-3.9063Y0.8902
X-3.9115Y0.9445
X-3.9158Y0.9986
X-3.9223Y1.1017
X-3.9270Y1.1930
X-3.9298Y1.2722
X-3.9308Y1.3396
X-3.9302Y1.4124
X-3.9286Y1.4860
X-3.9260Y1.5603
X-3.9222Y1.6353
X-3.9174Y1.7110
X-3.9115Y1.7874
X-3.9046Y1.8644
X-3.8966Y1.9421
X-3.8874Y2.0197
X-3.8769Y2.0969

MitchBradley, 2017-11-25:
That's almost 1500 lines.  Saying that it happens during that file doesn't narrow it down much.  It would be more helpful if you showed just a few lines of context around an exact place where it occurs.

The original bug report showed it happening near "G1X7.3...".  Neither that nor the preceding G28.3 appear in the posted file.

That said, here's what I think is happening:

- It's not really a CNCjs problem, but rather a problem with tinyg rejecting some moves as being too short to handle at the requested feedrate.  The 202 error code is generated from inside tinyg and CNCjs is just passing it on.
- In the latest version of the tinyg source, MIN_SEGMENT_USEC is 2500, i.e. 2.5 milliseconds.  If a segment would take less time than that, tinyg will issue the error.  
- Just eyeballing the gcode above, I see some moves where the total distance is less than 0.01 inches.  At 100 ipm feedrate, a 0.01" move would take about 6 milliseconds.  If there are any distances less than 0.004", they will trigger the errors.  I could write a program to search for the short moves in the 1500 lines above, but I'd rather not :-)
- One easy way to check would be to decrease the feedrate and see if the errors disappear or become less frequent.  If so, and you want to keep the high feedrate, you might need to generate GCode with longer segments.  Most CAM posts have a way to configure that.
- In principle it would be possible to have CNCjs ignore such errors instead of pausing, but that seems like a bad idea.  If tinyg is rejecting the moves, you could end up with part integrity problems.
- g2core, which is the successor to tinyg, does not generate that particular error.  The infrastructure for it is still in the code, but the lines of code for that error are commented out.

phillips82, 2017-11-26:
Thank you for the insight on this. It raised a few more questions.  

I am using V-carve Desktop. To get rid of the small moves; do I delete points in the file that a very close together? Also, you mentioned the CAM ignoring the short segments. Can V-carve be set up this way?

swpaxton66, 2017-11-26:
Seems like I have exactly the same issue.  I am also running a tinyG controller.  I have run may mammoth GCode files for very detailed 3D cutting and not run across this issue before.  I have started to have this issue now that I have started to engrave photos.  I've reduced the file and cutting paths as much as I can in software but I still have this issue.  I used MeshCAM to generate my file from a photo.  After this job finishes I am going to use PhotoVCarve to also test the result.  And exactly as MitchBradley said there are very small move segments.  Another thing I am going to try is 1) smooth out the file in Blender and 2) reduce the file with MeshLab.  I'm not sure how the results will be because I don't want to reduce the detail of the cut too much.  I'll post again if I find a solution and would love to hear other ideas/comments as well.  Thanks.

swpaxton66, 2017-11-26:
This might be premature because I just started the job again but I chose a different post processor that uses arcs.  This reduced the file size but I haven't had an error since I started the job with this new post processor.  Usually I would have to restart at least 10x by now due to the min time error.  I checked PhotoVCarve (I assume the software suite uses the same post processors) and didn't see the same option as the one I am using in MeshCAM.  However you might want to try (I will try this later) to output the same simple file using all of the post processors and then review the file sizes.  My money is on the post processor with the smallest file size.

MitchBradley, 2017-11-26:
This sounds like the kind of question that should be asked on the Vectric forum, not here.

The arc approach sounds good to me, as a relatively long curve segment can often be approximated by a single gcode arc (G2 or G3 gcode command), whereas if you approximate it by a series of line segments you end up with a lot of really tiny line segments (G1 gcode commands).  Using a single arc is beneficial in many ways.  It greatly reduces the amount of data that goes over the serial line to tinyg so the planner buffering works better, and it lets the machine controller (tinyg) work out the best stepper control for the arc, taking into account its knowledge of the machine's mechanical characteristics.

The metric of smallest file size is a reasonably proxy for what you want.  Smaller files generally correspond to fewer small segments.

phillips82, 2017-11-26:
Thanks for all the input guys. I will try this in my shop tonight.

MitchBradley, 2017-11-27:
I downloaded the VCarve Desktop Trial Edition and searched for the option to set the segment tolerance.

I found it at Edit>Options>2D Toolpath Tolerance  - and possibly 3D Toolpath Tolerance and VCarve Toolpath Tolerance.  The default value for those is 0.0004 - 4 tenths of a mil - which is quite tight.  I changed them to 0.01, and later to 0.005 to see what would happen.   You have to make a new toolpath operation for the change to take effect; editing an existing toolpath won't do it.

Anyway, with the original 0.0004 setting and the TinyG_inch post, the output file for the AvalonNameplate sample was 1086 lines long, and the first segment length that I looked at was 0.012 inches.  With the 0.01 setting, the output file was 402 lines and the first segment was 0.037 inches.  So clearly this setting affects the GCode in the right way.  With the 0.005 setting, the file was only slightly larger - 459 lines - and the first segment was still about 0.037 inches.

You will have to decide for yourself what setting is appropriate for your machine and work. 0.01 is probably too coarse.

phillips82, 2017-11-27:
Ok. I will definitely play with these setting; as I was not able to change the post or delete nodes that were close to corners etc. and make the error go away. 

The same files run perfectly fine on chilipeppr, for reference. Not sure what the difference is. Chilipeppr’s instability and running in the browser is what drove me to CNCJS. just wish I could fix this one issue so far.

MitchBradley, 2017-11-27:
I can explain why they "run perfectly fine on chilipeppr" but pause on CNCjs.  chilipeppr completely ignores the tinyg error messages.  The relevant chilipeppr source code is https://github.com/chilipeppr/widget-tinyg/blob/master/widget.js lines 755..796.  That code processes the data received from the tinyg serial port.  It responds to messages like "qr", "sr", etc, but completely ignores "err".

The relevant cncjs code, in which tinyg "err" reports are handled, is https://github.com/cncjs/cncjs/blob/master/src/app/controllers/TinyG/TinyGController.js lines 467..499.  In particular, line 496 says
```
        this.workflow.pause({ err: err.msg });
```
So when cncjs sees such an error message, it pauses the workflow.

One on could ignore that error by changing that line to:
```
    if (code != 202) {
        this.workflow.pause({ err: err.msg });
    }
```
Or simply delete the line entirely to avoid pausing on all errors.

The practical problem is that one would then need to recompile cncjs, which is a PITA and can take a long time.  If you already have a cncjs compilation setup, it would not be too bad.

swpaxton66, 2017-11-27:
Thanks Mitch. That’s valuable info. This was my suspicion. Thanks for confirming it. 

This may also explain why I have had issues I can’t quite explain using Chilipeppr.

phillips82, 2017-11-27:
Thank you for the explanation.  I am less than an amateur at the programming side. 

Changing my 2D tolerance to 0.001 got rid of the 202 err code for the trial files. Though I am getting a 155 arc error code. I will play with the setting in V- carve more. Thanks again.

cheton, 2017-11-27:
@MitchBradley Thanks for sharing with us your findings.

You can see below reason why a G-code sender should pause/stop program execution when an error is occurred:

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#g-code-error-handling
> It's highly recommended to do what all professional CNC controllers do when they detect an error in the G-code program, halt. Don't do anything further until the user has modified the G-code and fixed the error in their program. Otherwise, bad things could happen.

I'm not sure what error codes can be completely ignored, actually it may depend on user's needs. If some errors are necessary to be excluded, I can add an option to ignore certain error codes in the whitelist with  comma-separated list of strings or patterns.

phillips82, 2017-11-28:
For my purposes both the 202 and 155 errors only resulted in a pause. Neither had an affect on the final product, just delayed the end resulted and frustrated me. I for one would love a desktop version without all the error codes.  Thank you.

swpaxton66, 2017-11-28:
Thanks Cheton. I would also like to see some sort of config option that would allow bypassing errors, or ideally bypass individual errors (pick and choose which errors to bypass)

I totally agree with your error approach however. But it would be ideal to bypass.

swpaxton66, 2017-11-28:
Forgot to say I use the desktop version also.

MitchBradley, 2017-11-28:
The other way to do it would be to change tinyg not to issue those errors.  g2core, which is based on tinyg, already omits them.

swpaxton66, 2017-11-28:
I see in the tinyg config settings you can change the minimum segment time but it’s strongly recommended not to change the default value. I don’t see any setting to disable errors. I guess the developer would need to reflash the board or if you had the source code I guess you could do it yourself.

swpaxton66, 2017-11-28:
I tried to clone the tinyG github code and received an error: Cannot checkout to invalid path 'firmware/tinyg/default/NUL'.  Although the message is clear, at the moment I'm not sure how to fix it.

cheton, 2017-11-28:
@phillips82 @swpaxton66 

This change was made in 1.9.11: https://github.com/cncjs/cncjs/commit/d16264a4f53b7b2d9cf19441cea44176ddfbe134

I will plan to add the "ignoreErrors" option in 1.9.12. As a temporary solution, you can roll back to 1.9.10, which will not pause program execution when an error is occurred.

phillips82, 2017-11-28:
Forgive my ignorance. How do I roll back to 1.9.10 on the desktop app? Within the app or a new download from github? Thank you for all the responses to this.

cheton, 2017-11-28:
@phillips82 You don't need to uninstall previous program, just download the file from the releases page and run installer.

phillips82, 2017-11-28:
Thank you. I guess this will close this issue then. Thanks again.

cheton, 2017-11-28:
Just keep this issue open for a while. I will close this issue once 1.9.12 is released.

cheton, 2017-12-03:
Added in #233 

[image]

cheton, 2017-12-06:
Added in [1.9.12](https://github.com/cncjs/cncjs/releases/tag/v1.9.12)

I'd consider adding a whitelist option in a future release that allows users to ignore non-critical error codes.

closed 2017-12-06 by cheton

phillips82, 2017-12-06:
Awesome. Thank you.