cncjs/cncjs #232: Axes movement is not executed. Changes in widget is not saved

sailor1958, 2017-11-28:
#### Description
# **Axes moment or other gcode**
Connection works well. 
Homing execute okay.
When try to use ex. Z+ no gcode is send.
Same problem with all function, there was expected to send gcode to controller.
If I use the Console window, and type in ex. G91 X10 Y10 this is execute as expected, and same for all other commands I type in at the console.

Has also created a macro. When execute, this not sent to the controller.

# **Widget**
If I changes in the widgets setting ex. disable TinyG, it is enablednext time I access the workspace again. 

Guest it is a install problem, have used this guide : https://cnc.js.org/docs/rpi-setup-guide/

**When running CNC -P 8000 from the CLI, I get this when hit the Y+ buttom :** 
_2017-11-28T10:18:36.176Z - error: uncaughtException: words.includes is not a function date=Tue Nov 28 2017 11:18:36 GMT+0100 (CET), pid=2140, uid=1000, gid=1000, cwd=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app, execPath=/home/pi/.nvm/versions/node/v4.8.6/bin/node, version=v4.8.6, argv=[/home/pi/.nvm/versions/node/v4.8.6/bin/node, /home/pi/.nvm/versions/node/v4.8.6/bin/cnc, -p, 8000], rss=66727936, heapTotal=48627584, heapUsed=44448140, external=983501, loadavg=[0.359375, 0.89794921875, 1.44287109375], uptime=13371, trace=[column=27, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=Feeder.dataFilter, line=15097, method=dataFilter, native=false, column=32, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=Feeder.next, line=2710, method=next, native=false, column=35, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=gcode, line=16179, method=null, native=false, column=9, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=GrblController.command, line=16254, method=command, native=false, column=36, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=null, line=14803, method=null, native=false, column=13, file=events.js, function=emitThree, line=97, method=null, native=false, column=7, file=events.js, function=Socket.emit, line=175, method=emit, native=false, column=12, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js, function=null, line=503, method=null, native=false, column=9, file=node.js, function=nextTickCallbackWith0Args, line=489, method=null, native=false, column=13, file=node.js, function=process._tickCallback, line=418, method=_tickCallback, native=false], stack=[TypeError: words.includes is not a function,     at Feeder.dataFilter (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:15097:27),     at Feeder.next (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:2710:32),     at gcode (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:16179:35),     at GrblController.command (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:16254:9),     at Socket.<anonymous> (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:14803:36),     at emitThree (events.js:97:13),     at Socket.emit (events.js:175:7),     at /home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js:503:12,     at nextTickCallbackWith0Args (node.js:489:9),     at process._tickCallback (node.js:418:13)]
2017-11-28T10:18:36.217Z - error: uncaughtException: words.includes is not a function date=Tue Nov 28 2017 11:18:36 GMT+0100 (CET), pid=2140, uid=1000, gid=1000, cwd=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app, execPath=/home/pi/.nvm/versions/node/v4.8.6/bin/node, version=v4.8.6, argv=[/home/pi/.nvm/versions/node/v4.8.6/bin/node, /home/pi/.nvm/versions/node/v4.8.6/bin/cnc, -p, 8000], rss=66822144, heapTotal=48627584, heapUsed=44542092, external=983501, loadavg=[0.359375, 0.89794921875, 1.44287109375], uptime=13371, trace=[column=27, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=Feeder.dataFilter, line=15097, method=dataFilter, native=false, column=32, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=Feeder.next, line=2710, method=next, native=false, column=35, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=gcode, line=16179, method=null, native=false, column=9, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=GrblController.command, line=16254, method=command, native=false, column=36, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=null, line=14803, method=null, native=false, column=13, file=events.js, function=emitThree, line=97, method=null, native=false, column=7, file=events.js, function=Socket.emit, line=175, method=emit, native=false, column=12, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js, function=null, line=503, method=null, native=false, column=9, file=node.js, function=nextTickCallbackWith0Args, line=489, method=null, native=false, column=13, file=node.js, function=process._tickCallback, line=418, method=_tickCallback, native=false], stack=[TypeError: words.includes is not a function,     at Feeder.dataFilter (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:15097:27),     at Feeder.next (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:2710:32),     at gcode (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:16179:35),     at GrblController.command (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:16254:9),     at Socket.<anonymous> (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:14803:36),     at emitThree (events.js:97:13),     at Socket.emit (events.js:175:7),     at /home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js:503:12,     at nextTickCallbackWith0Args (node.js:489:9),     at process._tickCallback (node.js:418:13)]
2017-11-28T10:18:36.268Z - error: uncaughtException: words.includes is not a function date=Tue Nov 28 2017 11:18:36 GMT+0100 (CET), pid=2140, uid=1000, gid=1000, cwd=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app, execPath=/home/pi/.nvm/versions/node/v4.8.6/bin/node, version=v4.8.6, argv=[/home/pi/.nvm/versions/node/v4.8.6/bin/node, /home/pi/.nvm/versions/node/v4.8.6/bin/cnc, -p, 8000], rss=66916352, heapTotal=48627584, heapUsed=44637200, external=983501, loadavg=[0.359375, 0.89794921875, 1.44287109375], uptime=13371, trace=[column=27, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=Feeder.dataFilter, line=15097, method=dataFilter, native=false, column=32, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=Feeder.next, line=2710, method=next, native=false, column=35, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=gcode, line=16179, method=null, native=false, column=9, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=GrblController.command, line=16254, method=command, native=false, column=36, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js, function=null, line=14803, method=null, native=false, column=13, file=events.js, function=emitThree, line=97, method=null, native=false, column=7, file=events.js, function=Socket.emit, line=175, method=emit, native=false, column=12, file=/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js, function=null, line=503, method=null, native=false, column=9, file=node.js, function=nextTickCallbackWith0Args, line=489, method=null, native=false, column=13, file=node.js, function=process._tickCallback, line=418, method=_tickCallback, native=false], stack=[TypeError: words.includes is not a function,     at Feeder.dataFilter (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:15097:27),     at Feeder.next (/home/pi/.nvm/versions/node/v4.8.6/lib/node_modules/cncjs/dist/cnc/app/index.js:2710:32),     at gcod [... cut ...] b/socket.js:503:12,     at nextTickCallbackWith0Args (node.js:489:9),     at process._tickCallback (node.js:418:13)]_

#### Versions

- CNCjs: 1.9.11
- Node.js: 4.8.6
- NPM: 2.15.11

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

cheton, 2017-11-28:
This issue was introduced in 1.9.11, I did notice this problem and had fixed it in https://github.com/cncjs/cncjs/commit/88d4dec4a48787af32b7307f952e107fc87d72b7. This update will be included in the next release of CNCjs 1.9.12.

So far you have two workarounds:

1. Upgrade Node.js to v6, which provides support for `Array.includes()`

2. Downgrade CNCjs to 1.9.10

Sorry for any inconvenience.

sailor1958, 2017-11-28:
Thanks for the responce.
I have just upgraded Node.js to v6 and found it works now.

closed 2017-11-28 by sailor1958