cncjs/cncjs #236: Visualizer not recognised g-code from T2Laser

danowar, 2017-12-03:
#### Description

T2Laser is a laser cam software designed for use with low-power diode lager engravers.  The software uses an M3 command in each of the G0 movements, for controlling laser power (see g-code snippet below).  This M3 command is preventing the visualizer for displaying toolpaths.  I verified this was the cause by removing all M3 commands from the file, and reloading.  The toolpaths are then displayed, but the laser doesn't turn on without them.

Is it possible to have the visualizer ignore the M3 commands?

#### Versions

- CNCjs: 1.9.11
- Node.js: 7.10.1
- NPM: 5.5.1

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux

MitchBradley, 2017-12-03:
Did you mean to attach a gcode sample?  I don't see one.

cheton, 2017-12-03:
@danowar 

Please drag and drop your G-code file into the comment field, so we can help troubleshoot this issue.

MitchBradley, 2017-12-03:
@cheton  - I realized that I had an evaluation copy of T2Laser on my computer, so I used it to generate some sample GCode.  It had a line like this:

G0 X1.1 M03 S0

The wikipedia article on gcodes and mcodes includes an example of a line with both G and M codes, so the formulation is at lease plausible.

If you would like, I can look into this issue.  I have some experience with GCode parsing.

cheton, 2017-12-03:
The toolpath generated by [gcode-toolpath](https://github.com/cncjs/gcode-toolpath/) is preprocessed by [gcode-interpreter](https://github.com/cncjs/gcode-interpreter/). Within the interpreter, it separate words into groups with **G** word and **M** word.

https://github.com/cncjs/gcode-interpreter/blob/master/src/Interpreter.js#L36
```js
const partitionWordsByGroup = (words = []) => {
    const groups = [];

    for (let i = 0; i < words.length; ++i) {
        const word = words[i];
        const letter = word[0];

        if ((letter === 'G') || (letter === 'M')) {
            groups.push([word]);
            continue;
        }

        if (groups.length > 0) {
            groups[groups.length - 1].push(word);
        } else {
            groups.push([word]);
        }
    }

    return groups;
};
```

@MitchBradley 
As your example, the result is supposed to be `[['G0', 'X1.1'], ['M03', 'S0']]`. I may need a complete example to know how I can fine tune the interpreter.

MitchBradley, 2017-12-03:
```
( Generated by T2Laser ) 
( Image2Gcode for Grbl )
( Start Point: LL      )
( Frame Mode : Abs.    )
( X Maximum  : 5.9     )
( Y Maximum  : 5.9     )
( Laser Max  : 255     )
( Feed Rate  : 1000    )
( Resolution : 0.1     )
( Image Name : Digit2.bmp )
G21
G90
F1000
M05
M03 S0
G01 X4.9 Y5.6 M03 S0
X4.8 M03 S255
X1 Y5.6 M03 S255
X0.9 M03 S0
X0.7 Y5.5 M03 S0
... many repetitions of similar lines with different coordinates
M03 S0
M05
G00 X0 Y0
```

Comments:
M03 S0   leaves the laser on but sets the power to 0
M03 S255  is laser on at full power

The range of S values differs for different laser controllers.

Some laser posts do not use M03 S0 but instead use M5 for laser off.
The laser posts that I am most familiar with put the M03 and M05 commands on separate lines from the G moves, but obviously this one does not.

cheton, 2017-12-03:
Given a G-code like below:
```
G01 X4.9 Y5.6 M03 S0
X4.8 M03 S255
X1 Y5.6 M03 S255
```

will be partitioned into the following groups by the gcode-interpreter:
```
cmd = G01, words = ['X4.9', 'Y5.6']
cmd = M03, words = ['S0']
cmd = M03, words = ['X4.8']
cmd = M03, words = ['S255']
cmd = M03, words = ['Y5.6']
cmd = M03, words = ['S255']
```

That will explain why no toolpaths are generated.

I will fix the code in gcode-interpreter by not replacing `self.cmd` with the **M** word.

https://github.com/cncjs/gcode-interpreter/blob/master/src/Interpreter.js#L62-L77
```js
if ((letter === 'G') || (letter === 'M')) {
    self.cmd = cmd;
    args = fromPairs(words.slice(1));
} else {
    // Use the same command if the line does not start with Gxx or Mxx.
    // For example:
    // G0 Z0.25
    //  X-0.5 Y0.
    //  Z0.1
    // G01 Z0. F5.
    //  X0.5 Y0. I0. J-0.5
    //  X0. Y-0.5 I-0.5 J0.
    //  X-0.5 Y0. I0. J0.5
    cmd = self.cmd;
    args = fromPairs(words);
}
```

MitchBradley, 2017-12-03:
That seems like a good fix.  I'm pretty sure that M commands are not supposed to be inferred on subsequent lines.

MitchBradley, 2017-12-03:
In fact, it might be better to infer the command only for G0 and G1 motion modes.

cheton, 2017-12-03:
Will all motion commands (e.g. G0, G1, G2, G3, G38.2, G38.3, G38.4, G38.5, G80) apply the same rule?

danowar, 2017-12-03:
My bad, I forgot the code.  Here is the first 50 lines of a job I just ran.  It is worth mentioning, the job runs fine, just doesn't visualize properly.

`( Generated by T2Laser ) 
( Image2Gcode for Grbl )
( Start Point: Center  )
( Frame Mode : Abs.    )
( X Maximum  : 84.4    )
( Y Maximum  : 99.9    )
( Laser Max  : 1000    )
( Feed Rate  : 750     )
( Resolution : 0.1     )
( Image Name : text4272.png )
G21
G90
F750
M05
G00X0Y0F750
G92X42.2Y49.95
G90
G00 X0 Y0
G01
M03 S0
X5.2 Y0.2 M03 S0
X5.3 Y0.1 M03 S1000
X5.4 Y0 M03 S0
X5.5 Y0 M03 S0
X5.4 Y0.1 M03 S1000
X5.2 Y0.3 M03 S1000
X5.1 Y0.4 M03 S0
X5.2 Y0.4 M03 S0
X5.3 Y0.3 M03 S1000
X5.5 Y0.1 M03 S1000
X5.6 Y0 M03 S0
X5.7 Y0 M03 S0
X5.6 Y0.1 M03 S1000
X5.2 Y0.5 M03 S1000
X5.1 Y0.6 M03 S0
X5.2 Y0.6 M03 S0
X5.3 Y0.5 M03 S1000
X5.7 Y0.1 M03 S1000
X5.8 Y0 M03 S0
X5.9 Y0 M03 S0
X5.8 Y0.1 M03 S1000
X5.2 Y0.7 M03 S1000
X5.1 Y0.8 M03 S0
X5.2 Y0.8 M03 S0
X5.3 Y0.7 M03 S1000
X5.9 Y0.1 M03 S1000
X6 Y0 M03 S0
X6.1 Y0 M03 S0
X6 Y0.1 M03 S1000
X5.2 Y0.9 M03 S1000`

neilferreri, 2017-12-03:
All of those M03s, but the first one, are unnecessary.  The spindle speed (or laser power) should change with just the Sxxx.
@cheton All of the modal commands should persist until another in the same group is issued. A motion mode (G0) will replace a previous motion mode (G3), but it shouldn't be replaced by a command from a different modal group or an m-code.

cheton, 2017-12-03:
Fixed in [gcode-interpreter@2.0.1](https://github.com/cncjs/gcode-interpreter/releases/tag/v2.0.1)

G0, G1, G2, G3, G38.2, G38.3, G38.4, G38.5 will update the motion mode, and G80 will clear the mode mode.
```js
if (letter === 'G') {
    cmd = (letter + code);
    args = fromPairs(words.slice(1));

    // Motion Mode
    if (code === 0 || code === 1 || code === 2 || code === 3 || code === 38.2 || code === 38.3 || code === 38.4 || code === 38.5) {
        self.motionMode = cmd;
    } else if (code === 80) {
        self.motionMode = '';
    }
} else if (letter === 'M') {
    cmd = (letter + code);
    args = fromPairs(words.slice(1));
} else if (letter === 'T') { // T1 ; w/o M6
    cmd = letter;
    args = code;
} else if (letter === 'F') { // F750 ; w/o motion command
    cmd = letter;
    args = code;
} else if (letter === 'X' || letter === 'Y' || letter === 'Z' || letter === 'A' || letter === 'B' || letter === 'C' || letter === 'I' || letter === 'J' || letter === 'K') {
    // Use previous motion command if the line does not start with G-code or M-code.
    // @example
    //   G0 Z0.25
    //   X-0.5 Y0.
    //   Z0.1
    //   G01 Z0. F5.
    //   G2 X0.5 Y0. I0. J-0.5
    //   X0. Y-0.5 I-0.5 J0.
    //   X-0.5 Y0. I0. J0.5
    // @example
    //   G01
    //   M03 S0
    //   X5.2 Y0.2 M03 S0
    //   X5.3 Y0.1 M03 S1000
    //   X5.4 Y0 M03 S0
    //   X5.5 Y0 M03 S0
    cmd = self.motionMode;
    args = fromPairs(words);
}
```

cheton, 2017-12-03:
@danowar 

The bug fix will be included in CNCjs 1.9.12. For a temporary workaround, you can follow @neilferreri's suggestion to remove all **M03**s except the first one, as shown below:
```
M03 S0
X5.2 Y0.2 S0
X5.3 Y0.1 S1000
X5.4 Y0 S0
X5.5 Y0 S0
```

If you want a hot fix in short, I can give you a patched JavaScript bundle to replace the file in CNCjs 1.9.11.

danowar, 2017-12-03:
Wow, that was quick work.  Thanks for your help.  I had done as you suggested (removing redundant M3s) during my testing and it worked well, but of course it is an extra step.

If the hot fix is simply enough to apply, I would take it.  Otherwise I'll wait for the next version.

cheton, 2017-12-06:
Fixed in [1.9.12](https://github.com/cncjs/cncjs/releases/tag/v1.9.12)

closed 2017-12-06 by cheton