cncjs/cncjs #237: Initial position of toolpath

MitchBradley, 2017-12-07:
@cheton -

A design question: 

I have added to cncjs-shopfloor-tablet a lightweight 2D  gcode previewer.  It is built on gcode-interpreter / gcode-toolpath / gcode-parser , with addLine and addArcCurve methods for determining the bounding box and rendering onto a 2dcanvas.

The problem is that I would like to set the starting position for gcode-toolpath to the current machine position, instead of having it start at 0,0,0 on the first invocation and on the last position on subsequent invocations.  As a first try, I changed gcode-toolpath so that, if its options argument contains a "position" property, the position starts from there.

That works well the first time, but in order for it to work when you later load a second gcode file, it is necessary to make a new instance of gcode toolpath, letting the garbage collector worry about the old no-longer-referenced instance.  That's workable, but it seems inefficient.  Is there a better design pattern?

I thought of exporting a setPosition method from Toolpath, but I think that doing so would require a change in Toolpath's constructor to return "this" instead of "new Interpreter(...)".

What is your recommended solution?

cheton, 2017-12-07:
Sorry, I misunderstood your meaning.

Given a G-code program like below:
```
N1 G21 G90
N2 G0 Z0.25
N3 X-0.5 Y0.
N4 Z0.1
```

will generate such tool path in mm units

```
[ { motion: 'G0',
    v1: { x: 0, y: 0, z: 0 },
    v2: { x: 0, y: 0, z: 0.25 } },
  { motion: 'G0',
    v1: { x: 0, y: 0, z: 0.25 },
    v2: { x: -0.5, y: 0, z: 0.25 } },
  { motion: 'G0',
    v1: { x: -0.5, y: 0, z: 0.25 },
    v2: { x: -0.5, y: 0, z: 0.1 } }
]
```

To determine the bounding box, we can exclude the initial position by ignoring the first line of the array. 
That said, the start position should be (0, 0, 0.25) (i.e. `G0 Z0.25`) instead of (0, 0, 0).

I can make a fix in [gcode-toolpath](https://github.com/cncjs/gcode-toolpath) to bypass the first call to `addLine` and `addArcCurve`.

cheton, 2017-12-07:
I recall that why a initial position was specified in the toolpath, the first line of G0/G1/G2/G3 command may move on only one axis, but it is required to have two 3D points (x,y,z) to define a line in 3D space.

I agreed that using current work position is much better than (0, 0, 0). How about only filtering G1/G2/G3 lines to determine the bounding box of a workpiece?

MitchBradley, 2017-12-07:
Your first answer matches my question.  I am already using the techniques you first suggested - creating new Toolpath instances and translating the canvas.  I was hoping to make things more efficient by avoiding the overhead of "new" and garbage collection on every invocation, but I suppose that is a small factor compared to the objects that parsing creates.

It is important to see the initial move because it tells you whether your machine starting position is correct relative to the workpiece.

The patch to toolpath is simply:

```
diff --git a/src/Toolpath.js b/src/Toolpath.js
index 4c39a7b..cbab48f 100644
--- a/src/Toolpath.js
+++ b/src/Toolpath.js
@@ -563,6 +563,9 @@ class Toolpath {
         });
         this.setModal(nextModal);
         this.fn = { addLine, addArcCurve };
+       if (options.position !== undefined) {
+            this.setPosition(options.position.x, options.position.y, options.position.z);
+       }

         return new Interpreter({ handlers: this.handlers });
     }
```

cheton, 2017-12-08:
Added in [gcode-toolpath@2.1.0](https://github.com/cncjs/gcode-toolpath/releases/tag/v2.1.0)

You can set current position with `toolpath.setPosition({ x, y, z })` or `toolpath.setPosition(x, y, z)` in additional to the constructor options.

MitchBradley, 2017-12-08:
Nice solution. Thanks!

closed 2017-12-08 by MitchBradley