cncjs/cncjs #238: Grbl Status Updates not streaming

liqdfire, 2017-12-08:
#### Description

The UI is not receiving status updates from Grbl on an interval. The UI does not update coordinates or grbl state.

1) When I home my machine, the axis control widget is locked unless I issue a manual ? command
2) When a gcode job is running, the axis control widget does not update with the machine coordinates, not doess the grbl widget update

#### Versions

- CNCjs: 1.9.11

-- do not remember the exact version of these off the top of my head, they were updated to the latest on raspbian a couple of weeks ago
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [x] NPM
- [ ] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl
- [ ] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [x] Raspberry Pi
- [ ] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [x] Not Applicable
- [ ] Windows
- [ ] Mac
- [ ] Linux
[image]

cheton, 2017-12-09:
@liqdfire 

It appears that your G-code is quite simple with only straight lines. Did you see any errors on Node.js server output or browser's console (Press **F12** to open developer tools)?

I also upgraded the serialport library from v4 to v6 in CNCjs 1.9.12. It's worth a try if you cannot find any error messages related to this issue.

liqdfire, 2017-12-18:
1.9.12 does not appear to have the issue, thx!

closed 2017-12-19 by cheton

pbuyle, 2017-12-29:
I've the same issue with CNCjs 1.9.12 running on a Raspberry Pi 2 and a Grbl controller (0.9).

pbuyle, 2017-12-29:
I did a full system upgrade, updated to node 8.9.3, and did a `sudo npm -g uninstall cncjs` befire re-installing with `sudo npm -g install cncjs --usafe-perm`.
The UI does not sync the controller status. Nothing in the browser console, and nothing in the server logs. Sending a `?` command from the GRBL console does update the status in the UI but does not restore syncing.

pbuyle, 2017-12-30:
Installed a brand new Raspbian on a RPI3, then NodeJS 8.9.3 (using package manager) and cncjs (sudo npm install -g cncjs@latest --unsafe-perm) and it is still the same behavior and no status sync. in the UI.

pbuyle, 2017-12-30:
Here are the logs of running the server, connection to the UI and homing the CNC: https://pastebin.com/AwsK1DAK
And here are the Grbl settings of the CNC: https://pastebin.com/iX0Z3ZE4

cheton, 2017-12-31:
Do you have a Windows or Mac desktop or laptop that can reproduce this issue? Here is the nightly build (not the official release) of 1.9.13 desktop application:
https://github.com/cncjs/cncjs/releases/tag/master-latest

[image]

Please let me know the result if it will behave different when toggling the "Enable hardware flow control" switch.

pbuyle, 2017-12-31:
Unfortunately, I don't have a Windows or Mac computer. I was able to test the `master-latest`  from a Windows 10 VM (qemu-system-x86). It didn't change anything. With or without checking the ""Enable hardware flow control".

pbuyle, 2018-02-19:
Upgraded to 1.9.14 and I still have the issue. I still have no clue what's going on and how to with help debug. Other softwares on the same hardware have no issue (just tried ChiliPeppr without issue). I've also tried a Marlin (v1.0.2+) controller from the same Raspberry Pi without any issue.

Note: Should I open a new issue (since this one is closed)?