cncjs/cncjs #239: New feature for lathe

alkabal, 2017-12-11:
#### Description

Hi

This is not a real issue but ask for enhancement.

It is possible with your code design to create a branch for use with cnc lathe in place of cnc mill ?

I try to retrofit a lathe with smoothieboard and i think the best way for manage is to use a special version from your soft if possible.

Don't know the amount of job needed.

Please can you consider this new feature ?

Best regards, Aurélien

#### Versions

- CNCjs: 1.9.x
- Node.js: 6.x
- NPM: 5.x

#### How Do You Install CNCjs?

- [ ] NPM
- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [ ] Grbl
- [x] Smoothieware
- [ ] TinyG/g2core

#### Hardware

- [ ] Raspberry Pi
- [x] Desktop or Laptop
- [ ] Mobile Device

#### Operating System

- [ ] Not Applicable
- [x] Windows
- [ ] Mac
- [ ] Linux

saxa, 2017-12-11:
It would be good to understand what exactly you need. If this is a normal lathe you should have basically
3 axis, but sine one axis is just rotating the piece you end up using basically 2. This means that using a 2D view should be enough if i get it rght.

MitchBradley, 2017-12-11:
Here's a Mach3 screen for a lathe, for reference.  It appears to use X and Z coordinates, rather conventional feed, spindle, and gcode controls, and a 2D toolpath.

By the way, I am about to check in a 2D toolpath viewer for the cncjs-shopfloor-tablet workspace.

[image]

saxa, 2017-12-11:
I see, I think that using X and Z or X and Y should be trivial thing to change, probably just a configuration issue, but do not take my words as correct as I do not have much experience on this.

alkabal, 2017-12-11:
Yes this is for normal lathe but cnc off course.

Like to have pieces view X Z

customize some button and adding some gcode like G90.1 G91.1 in future threading if possible.

Many thanks for all reply

acwright, 2017-12-21:
I have also been investigating how to control my Sherline CNC lathe using a TinyG board. I would like to chime in and give this an upvote. 

I can confirm also that most CNC lathe software considers only the X and Z axis. For clarity, the Z axis is considered to be along the length of the lathe bed.

[image]

Therefore it does seem that a 2D view (it would be great if it was a specific lathe view!) and "limiting" the controls to X and Z would most likely suffice.

Thanks for considering this feature!

alkabal, 2017-12-22:
Hi

Thanks for better explain !

Really like to have the good 2D view direct.

Best regards

MitchBradley, 2017-12-22:
Cncjs-shopfloor-tablet now has a 2d viewer.  Currently it shows x and y but it would be trivial to make an x/z version

cheton, 2017-12-22:
Hi all,

Do you have some lathe G-code examples so that we can make further tests when it's implemented.

Three.js provides support for LatheGeometry, the geometry can be defined with an array of 2D points.
https://threejs.org/docs/#api/geometries/LatheGeometry
https://github.com/josdirksen/learning-threejs/blob/master/chapter-06/02-advanced-3d-geometries-lathe.html

alkabal, 2017-12-22:
`G21 G90 G61 G40 G8 G18
G0 Z8.0
G18
M3 S1000
G0 X7.05
G1 F300.0 Z16.0
G1 F800.0 Z1.0712
G0 Z8.0
G0 X6.35
G1 F300.0 Z16.0
G1 F800.0 Z1.6538
G0 Z8.0
G0 X5.65
G1 F300.0 Z16.0
G1 F800.0 Z2.3519
G0 Z8.0
G0 X4.95
G1 F300.0 Z16.0
G1 F800.0 Z2.5
G0 Z8.0
G0 X4.25
G1 F300.0 Z16.0
G1 F800.0 Z3.8292
G0 Z8.0
G0 X3.55
G1 F300.0 Z16.0
G1 F800.0 Z4.3955
G0 Z8.0
G0 X2.85
G1 F300.0 Z16.0
G1 F800.0 Z5.2713
G0 Z8.0
G0 X2.15
G1 F300.0 Z16.0
G1 F800.0 Z12.9631
G0 Z8.0
G0
G1 F300.0 Z11.0369
G1 F800.0 Z8.9631
G0 Z8.0
G0 X1.45
G1 F300.0 Z16.0
G1 F800.0 Z14.3841
G0 Z8.0
G0 X0.75
G1 F300.0 Z16.0
G1 F800.0 Z15.3713
G0 Z16.0607
G1 F300.0 X0.0607
G1 F800.0 X1.0607 Z15.0607
G3 X1.5 Z14.0 I-1.0607 K-1.0607
G1 Z13.6213
G1 X2.0607 Z13.0607
G3 Z10.9393 I-1.0607 K-1.0607
G1 X1.5 Z10.3787
G1 Z10.0
G1 Z9.6213
G1 X2.0607 Z9.0607
G3 X2.5 Z8.0 I-1.0607 K-1.0607
G1 Z6.0
G1 Z5.6213
G1 X3.0607 Z5.0607
G3 X3.4354 Z4.4354 I-1.0607 K-1.0607
G3 X4.5 Z3.0 I-0.4354 K-1.4354
G1 Z2.5
G1 X5.0
G3 X6.4354 Z1.4354 I0.0 K-1.5
G3 X7.5 Z0.0 I-0.4354 K-1.4354
G1 Z-1.0
G0 Z8.0
M5
$H
M30
`
generated with custom cambam smoothieboard postprocess.

runryder912, 2022-05-28:
It would be nice to have a lathe. How is this going?

MitchBradley, 2022-05-28:
I am not aware of anyone working on it.

runryder912, 2022-05-28:
> I am not aware of anyone working on it.

Thanks for the update