cncjs/cncjs #241: Can't paste into console anymore

myself248, 2017-12-14:
#### Description

In previous versions, where there was a text entry box above the console, I could copy some text from Notepad and paste it into the box, then hit enter to execute it. This was part of my single-step debugging of new g-code. 

Now that the cursor lands right in the console window, Ctrl-V doesn't seem to paste (does it get passed straight to grbl? It seems to hang the machine) and I can't right-click-paste either. While I like the feel of the interactive terminal window, feature-wise this seems like a regression.

#### Versions

- CNCjs: 1.9.12
- Node.js: 6.x?
- NPM: 5.x?

#### How Do You Install CNCjs?

- [x] Download the CNCjs Desktop Application

#### CNC Controller

- [x] Grbl

#### Hardware

- [x] Desktop or Laptop

#### Operating System

- [x ] Windows

silverfoxvt, 2017-12-23:
This is also an issue running CNC.js 1.9.12 on OSX High Sierra 10.13.2.  Cannot cut/copy/paste to or from any portion of the app.  Ctrl-C, Ctrl-X, Ctrl-V do not work nor is there an Edit menu in the CNC.js OSX menu bar. App completely useless for pasting large g-code code blocks to create macros or for backing up g-code settings from the CNC.js terminal.

AustinSaintAubin, 2017-12-23:
Confirmed that non of the keyboard macros work on OSX High Sierra 10.13.2. with CNC.js 1.9.12
Looks like the Mac Electron app is failing to load /api/macros and a few other.
[image]

cheton, 2017-12-26:
This will be fixed by the pull request https://github.com/cncjs/cncjs/pull/249

cheton, 2017-12-26:
Hi @whimsycwd 
I believe you may run into the same problem with the Electron app, you can take a look at these commits and fix the copy/paste issue on your side. Thanks. 😃

cheton, 2017-12-26:
Reopen this issue until 1.9.13 is released

cheton, 2018-01-15:
New improvements to the console widget will be available in 1.9.13

[image]
[image]

cheton, 2018-02-03:
Fixed in 1.9.13
https://github.com/cncjs/cncjs/releases/tag/v1.9.13

commit b11ac043 (2017-12-26): Merge pull request #249 from cncjs/electron-context-menu

Electron context menu (resolves #241)
closed 2018-02-03 by cheton